Koenkk/zigbee2mqtt #3428: Color temperature of Hue white ambiance GU10

ananchev, 2020-04-25:
# Bug Report
## What happened
Setting color temperature on a Philips Hue White Ambiance GU10 does not seem to work.
## What did you expect to happen
Device documentation [here](https://www.zigbee2mqtt.io/devices/8718696598283.html) says that the bulb colon temperature shall be supported.
## How to reproduce it (minimal and precise)
Run `mosquitto_pub -h <broker_ip> -t <zigbee2mqtt>/<hueGU10>/set -m '{"color_temp": <value_mired>}'`.
The <value_mired> is between 154 is the coolest, 455 is the warmest. 
It is equivilent to 1000000/T (where T is the temperature in Kelvin) corresponding for this bulb to  6500K (154 Mired) to 2200K (455 Mired). Supported color temperature ranges [here](ttps://community-openhab-org.s3.dualstack.eu-central-1.amazonaws.com/original/3X/3/5/3535fa795a4b712983b9777c42f86d0bafc09993.png).

## Debug Info
Zigbee2mqtt version: 1.12.2 (commit #7d27a54)
Adapter hardware: CC2531
Adapter firmware version: "type":"zStack12","meta":{"transportrev":2,"product":0,"majorrel":2,"minorrel":6,"maintrel":3,"revision":20190608}

Koenkk, 2020-04-26:
Can you share your `database.db`?

ananchev, 2020-04-26:
Attached. Think I know my answer already - device is listed as LWG001, which is Hue White GU10 Bluetooth bulb....
[database.db.txt](https://github.com/Koenkk/zigbee2mqtt/files/4535056/database.db.txt)

Koenkk, 2020-04-28:
This indeed seems to be a hue white bulb (which doesn't support color temperature). It also does not have the cluster needed for color temperature support (so I actually think this is a hue white gu10).

ananchev, 2020-05-01:
All clear. Thanks Koenkk for the great work!

closed 2020-05-01 by ananchev