Koenkk/zigbee2mqtt #3432: ikea bulb not working / CC1352P-2

NicolasBoulanger, 2020-04-26:
hello,
i have an issue with my only one ikea bulb

`zigbee2mqtt:error 2020-04-26 16:14:09: Publish 'set' 'state' to 'buanderie' failed: 'Error: Command 0x000d6ffffe62cbae/1 genOnOff.off({}, {"timeout":6000,"manufacturerCode":null,"disableDefaultResponse":false}) failed (Error: Data request failed with error: 'No network route' (205))'`

im running CC1352P-2 flashed with the latest firmware since 2 weeks
restart of ha of zigbee2mqtt and not fix this issue
could you take a look please !?
thanks !

NicolasBoulanger, 2020-04-26:
and i have :
zigbee2mqtt:error 2020-04-26 19:38:21: Failed to execute LQI for '#device#'
for 2-3 devices

NicolasBoulanger, 2020-04-26:
👎 my new wifi ap make zigbee network crash !
No the same Chanel , 1 for wifi and 25 for zigbee.. :/

dreimer1986, 2020-04-29:
Reset the bulb and re-pair it. No clue what the WIFI AP does. Worksforme

closed 2020-06-28 by NicolasBoulanger