Koenkk/zigbee2mqtt #3435: Gledopto GL-C-008 2ID version white temperature not controllable

jussih, 2020-04-27:
# Bug Report

## Context information
There a are 2 different versions of the GL-C-008 LED strip controller: 1ID version and 2ID version. This is specifically about the 2ID version which has 2 endpoints, one for controlling the CW/WW leds and one for controlling the RGB leds.

The 2ID version introduces itself as GL-C-007 when pairing for some reason. GL-C-007 is apparently a very similar model, but only has one white channel, not warm white and cold white like 008.

## What happened
GL-C-008 2ID version was paired with zigbee2mqtt and Home Assistant. The device was recognized as model GL-C-007.

The device was automatically discovered in Home Assistant. Only the warm white channel is controllable via the "White value" slider, white channel color temperature can't be changed. RGB channel can be controlled normally.

Additionally ephemeral scenes in Home Assistant (https://www.home-assistant.io/integrations/scene/#creating-scenes-on-the-fly) do not work, since the white_value state attribute is set to -1 by zigbee-herdsman-converters when color modes are used and Home Assistant considers negative values incorrect.

## What did you expect to happen
Expected to be able to control white channel color temperature.

## How to reproduce it (minimal and precise)
- Pair GL-C-008 2ID version
- Observe there is no color temperature slider available in Home Assistant

## Debug Info
Zigbee2mqtt version: 1.12.2
Adapter hardware: CC2530
Adapter firmware version: 20190608

## Analysis
The Home Assistant integration already contains a fitting configuration with a temperature slider in addition to the color wheel: https://github.com/Koenkk/zigbee2mqtt/compare/master...jussih:feature/gl-c-008-2id?expand=1

Just enabling the temperature slider in Home Assistant is not enough, as currently the temperature slider will control the RGB channel color. The color temperature change needs to be targeted to the white endpoint: https://github.com/Koenkk/zigbee-herdsman-converters/compare/master...jussih:feature/gl-c-008-2id?expand=1#diff-3e0c275bd41af699153bf59d8dda4a2aR1389

The big problem here is that the device is detected as the 007 model which has no white color temperature. My examples currently work when you change the mode to `GL-C-008-2ID` in the database by hand. Is it possible to use an option in `devices.yaml` instead of that if we can't detect the model id automatically?

Note that the code samples linked contain also changes to control both white and rgb channels simultaneously which is not part of this issue per se.

jussih, 2020-05-10:
Fix for issue #3436 allows the white channel to be controlled separately, however it still does not allow temperature control in Home Assistant.

The Home Assistant configuration for the device should be:
```js
    'GL-C-007': [lightEndpoint('light_brightness_colorxy', 'rgb'),
                 lightEndpoint('light_brightness_colortemp', 'white')],`
```

Koenkk, 2020-05-11:
Assuming this can be closed as #3526 is merged. Thanks!

PR #3526 (2020-05-11): GL-C-008-2ID: White endpoint color temp control in Home Assistant
White endpoint color temperature control for Home Assistant integration.
Fixes #3435
closed 2020-05-11 by Koenkk