Koenkk/zigbee2mqtt #3436: Gledopto GL-C-008 2ID version not able to control both white and color leds simultaneously

jussih, 2020-04-27:
# Bug Report

## Context information
There a are 2 different versions of the GL-C-008 LED strip controller: 1ID version and 2ID version. This is specifically about the 2ID version which has 2 endpoints, one for controlling the CW/WW leds and one for controlling the RGB leds.

The 2ID version introduces itself as GL-C-007 when pairing for some reason. GL-C-007 is apparently a very similar model, but only has one white channel, not warm white and cold white like 008.

## What happened
GL-C-008 2ID version was paired with zigbee2mqtt and Home Assistant. The device was recognized as model GL-C-007.

The device was automatically discovered in Home Assistant.

When setting white channel brightness, rgb leds are turned off. When setting rgb color white leds are turned off.

Additionally ephemeral scenes in Home Assistant (https://www.home-assistant.io/integrations/scene/#creating-scenes-on-the-fly) do not work, since the white_value state attribute is set to -1 by zigbee-herdsman-converters when color modes are used and Home Assistant considers negative values incorrect.

## What did you expect to happen
Expected to be able to use both RGB leds and white leds simultaneously. This is the key selling point of the 2ID version of the device.

## How to reproduce it (minimal and precise)
- Pair GL-C-008 2ID version
- Observe that setting one channel causes the other channel to be turned off

## Debug Info
Zigbee2mqtt version: 1.12.2
Adapter hardware: CC2530
Adapter firmware version: 20190608

## Analysis
There is an old issue about controlling the channels separately with the Gledopto family of devices (#2335) but I would argue the 2ID version of the device should be handled differently as anyone who chooses to buy it is reasonably expected to want separate control out of the box. Also Home Assistant scenes should be able to control both white and color leds, which is currently not possible due to the -1 value in the `white_value` attribute set by zigbee-herdsman-converters.

I experimented on controlling the 2 channels using one Home Assistant device in the following way:
- Brightness slider controls the brightness of the RGB leds
- White value slider controls the brightness of the white leds
- Moving either slider to the minimum causes the channel to turn off the leds entirely
https://github.com/Koenkk/zigbee-herdsman-converters/compare/master...jussih:feature/gl-c-008-2id?expand=1

This kinda sorta works but seems to require some questionable hackiness and still has many quirks with Home Assistant. That leads me to think having 2 separate actual devices in Home Assistant would work better. Is that feasible?

Koenkk, 2020-05-04:
I'm not sure if it works for your device (as gledopto uses many different firmware), but it should be possible via https://github.com/Koenkk/zigbee2mqtt/issues/2335#issuecomment-566701129

jussih, 2020-05-04:
I forgot to mention this in the issue description, but I tried it. It doesn't seem to work correctly with the 2ID version.

As discovered (model id GL-C-007):
- Both brightness and white value sliders control the brightness of the white leds. There is no way to control the brightness of the RGB leds (changing their color with the xy color wheel works however)
- The on/off switch in home assistant controls only the white leds. RGB leds can't be turned on or off.

Manually set to a model id GL-C-008 from database.db:
- There is no white value slider in Home Assistant
- Color temperature slider changes the color of the RGB leds
- XY color wheel also changes the color of the RGB leds
- There is no way to control the brightness of the white leds
- There is no way to control the temperature of the white leds
- On/off switch in Home Assistant controls only the RGB leds

After these experiences I tried to fix the problems by sending the commands to the correct endpoints, code diff in the analysis section of the issue. I couldn't get that working quite as well as I liked, which is why I think the device would be more usable as 2 different virtual devices in Home Assistant.

Koenkk, 2020-05-05:
Can you try with the following definition:

```js
{
    zigbeeModel: ['GL-C-007'],
    model: 'GL-C-007',
    vendor: 'Gledopto',
    description: 'Zigbee LED controller RGBW',
    extend: generic.light_onoff_brightness_colortemp_colorxy,
    endpoint: (device) => {
        return {rgb: 10, white: 11};
    },
},
```

and the following in `homeasisstant.js`: 

```
'GL-C-007': [lightEndpoint('light_brightness_colortemp_colorxy', 'rgb'),
        lightEndpoint('light_brightness', 'white')],
```

jussih, 2020-05-05:
I had time for quick test today.

Updated to a clean 1.13.0 zigbee container in home assistant to get rid of any previous customizations I made.

Tested the modifications:
- RGB and white light entities are now discovered in Home Assistant
- The white entity can control the RGB led brightness in the strip.
- The RGB entity does not do anything.

I will have time for further testing tomorrow. I will try the following:
- Change the RGB endpoint to 11
- Change the white endpoint to 15
- Add color temp controls in the home assistant configuration for the white entity

Fabiancrg, 2020-05-06:
@jussih @Koenkk I am not using home assistant but tried the fix using mqtt topics on my GL-C-008 - 2ID, it's also detected as GL-C-007.

If I target the two endpoint 11 and 15 (with separate_control: true), I can:
- Adjust brightness of RGB  (zigbee2mqtt/LED/11/set)
- Adjust color of RGB
- Set RGB ON or OFF
- Adjust brightness of white  (zigbee2mqtt/LED/15/set)
- adjust color temp of white
- Set white ON or OFF 

so for me it looks good.

Was just wondering something, the brighness_move does not seem to work, is it normal ?
Not too sure if something is not handled by the device FW or in Z2M ?

jussih, 2020-05-06:
Did some testing with the following config:

```js
    'GL-C-007': [lightEndpoint('light_brightness_colorxy', 'rgb'),
                 lightEndpoint('light_brightness_colortemp', 'white')],`
```
```js
    {
        zigbeeModel: ['GL-C-007'],
        model: 'GL-C-007',                              
        vendor: 'Gledopto',
        description: 'Zigbee LED controller RGBW',     
        extend: generic.light_onoff_brightness_colortemp_colorxy,
        endpoint: (device) => {                    
            return {rgb: 11, white: 15};
        }, 
    },   
```

Results are that both the warm and cold white and rgb leds are controllable nicely with their separate entities in Home Assistant.

However I still had problems with the Home Assistant UI that I couldn't yet get to the bottom of. Controlling the lights works as expected, but the Home Assistant state doesn't update for color temp or xy color:

- Changing brightness works, entity brighness state updates
- Changing on/off state works, entity state updates
- Changing color temperature in the white entity does not work. The leds change color temperature, but the state in Home Assistant stays at the default level of 150
- Changing color in the rgb entity does not work. The leds change color, but the state in Home Assistant stays at the default of rgb [255,255,255]

The log looks like this when I click the xy color wheel in the rgb entity:
```
zigbee2mqtt:debug 2020-05-06 22:45:51: Received MQTT message on 'zigbee2mqtt/0x00124b001f79d7ea/rgb/set' with data '{"state": "ON", "color": {"x": 0.172, "y": 0.747}}'
zigbee2mqtt:debug 2020-05-06 22:45:51: Publishing 'set' 'state' to '0x00124b001f79d7ea'
zigbee2mqtt:info  2020-05-06 22:45:51: MQTT publish: topic 'zigbee2mqtt/0x00124b001f79d7ea', payload '{"state":"OFF","color":{"x":0.701,"y":0.299},"color_temp":37,"brightness":155,"white_value":187,"state_rgb":"ON","state_white":"OFF","brightness_white":60,"brightness_rgb":47}'
zigbee2mqtt:debug 2020-05-06 22:45:51: Publishing 'set' 'color' to '0x00124b001f79d7ea'
zigbee2mqtt:info  2020-05-06 22:45:51: MQTT publish: topic 'zigbee2mqtt/0x00124b001f79d7ea/rgb', payload '{"state":"ON","brightness":47}'
zigbee2mqtt:info  2020-05-06 22:45:51: MQTT publish: topic 'zigbee2mqtt/0x00124b001f79d7ea/white', payload '{"state":"OFF","brightness":60}'
zigbee2mqtt:info  2020-05-06 22:45:51: MQTT publish: topic 'zigbee2mqtt/0x00124b001f79d7ea', payload '{"state":"OFF","color":{"x":0.172,"y":0.747},"color_temp":154,"brightness":155,"white_value":187,"state_rgb":"ON","state_white":"OFF","brightness_white":60,"brightness_rgb":47}'
zigbee2mqtt:info  2020-05-06 22:45:51: MQTT publish: topic 'zigbee2mqtt/0x00124b001f79d7ea/rgb', payload '{"state":"ON","brightness":47}'
zigbee2mqtt:info  2020-05-06 22:45:51: MQTT publish: topic 'zigbee2mqtt/0x00124b001f79d7ea/white', payload '{"state":"OFF","brightness":60}'
```
Only the `state` and `brightness` values are published to the rgb and white endpoint topics, which is probably the reason the Home Assistant state is missing that data as well.

Fabiancrg, 2020-05-07:
@Koenkk 

I alo updated the code 

`return {rgb: 10, white: 11};`

and replaced it with

`return {rgb: 11, white: 15};`

in order to use rgb or white in the MQTT topic

So now it's working with topic zigbee2mqtt/LED/rgb/set or zigbee2mqtt/LED/white/set

Only problem I still see is the brightness_move; when I target the white it's not doing anything and when I target RGB it works but the "brightness_rgb" is not updated in new messages payload and I see a new field "brightness" that was not present before using the brightness_move.

Koenkk, 2020-05-07:
Can you try changing https://github.com/Koenkk/zigbee2mqtt/blob/master/lib/extension/publish.js#L194 to `for (const key of ['state', 'brightness', 'color', 'color_temp']) {`, does it fix the issue?

jussih, 2020-05-08:
With this change everything is working perfectly for me. Home Assistant state keeps up and setting lighting scenes work. Even a script which captures the current state of the strip, then flashes red leds and finally returns the previous state works fine.

A side effect was that `color` state appears in the white led entity and `color_temp` state in the rgb entity, but that didn't seem to cause issues.

@Koenkk Would you consider accepting a PR so that I create a new model id which behaves like this? We can't probably change the GL-C-007 config without causing regression and I don't have the device to test that anyway. With a new model id this functionality could be used by manually changing the device model after pairing. Then later we could try to solve how to detect this 2ID device correctly while pairing.

Koenkk, 2020-05-08:
Can you share your `database.db` entry after pairing this device? (Without modifications)

jussih, 2020-05-08:
https://gist.github.com/jussih/238dfbfb6e722f1995f336c0f23e20fd

Koenkk, 2020-05-09:
With the new fingerprint identification it's possible to identify a 1ID/2ID now. This is done now in the latest zigbee2mqtt dev branch. Turned out I had both the 1 and 2 ID versions of this controller (which Gledopto send me sometime ago).

Assuming this can be closed now.

closed 2020-05-09 by Koenkk

dakatso, 2020-05-24:
Thanks, it's working

joshuapfl, 2023-12-19:
Hello Folkls

Any chance to get this working on ZHA too? I dont want to change to 2mqtt for just this?

Any help would be much appreciated. :)