Koenkk/zigbee2mqtt #3437: What is needed, that Z2M generates a HA config?

swissbyte, 2020-04-27:
Hi

I am trying to add my own device to Z2M which supports on and off. In fact it is a lamp. 
Therefore i have added: 

```
	{
        zigbeeModel: ['DTB-ED2004-012'],
        model: 'DTB-ED2004-012',
        vendor: 'databyte.ch',
        description: 'Panda 1',
        extend: generic.light_onoff_brightness,
    },
```

The device pairs successfull and i am able to receive its status logs. 

```
zigbee2mqtt:debug 2020-04-27 21:21:05: Received Zigbee message from '0x00124b001201d226', type 'attributeReport', cluster 'genOnOff', data '{"onOff":0}' from endpoint 8 with groupID 0
zigbee2mqtt:info  2020-04-27 21:21:05: MQTT publish: topic 'zigbee2mqtt/0x00124b001201d226', payload '{"state":"OFF","linkquality":105}'
zigbee2mqtt:debug 2020-04-27 21:22:05: Saving state to file /share/zigbee2mqtt/state.json
zigbee2mqtt:debug 2020-04-27 21:27:05: Saving state to file /share/zigbee2mqtt/state.json
zigbee2mqtt:debug 2020-04-27 21:32:05: Saving state to file /share/zigbee2mqtt/state.json
zigbee2mqtt:warn  2020-04-27 21:34:19: Device '0x00124b001201d226' left the network
zigbee2mqtt:info  2020-04-27 21:34:19: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"device_removed","message":"left_network","meta":{"friendly_name":"0x00124b001201d226"}}'
zigbee2mqtt:info  2020-04-27 21:34:20: Device '0x00124b001201d226' joined
zigbee2mqtt:info  2020-04-27 21:34:20: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"device_connected","message":{"friendly_name":"0x00124b001201d226"}}'
zigbee2mqtt:info  2020-04-27 21:34:20: Starting interview of '0x00124b001201d226'
zigbee2mqtt:info  2020-04-27 21:34:20: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"pairing","message":"interview_started","meta":{"friendly_name":"0x00124b001201d226"}}'
zigbee2mqtt:debug 2020-04-27 21:34:23: Device '0x00124b001201d226' announced itself
zigbee2mqtt:info  2020-04-27 21:34:23: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"device_announced","message":"announce","meta":{"friendly_name":"0x00124b001201d226"}}'
zigbee2mqtt:debug 2020-04-27 21:34:28: Device '0x00124b001201d226' announced itself
zigbee2mqtt:info  2020-04-27 21:34:28: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"device_announced","message":"announce","meta":{"friendly_name":"0x00124b001201d226"}}'
zigbee2mqtt:debug 2020-04-27 21:34:31: Received Zigbee message from '0x00124b001201d226', type 'readResponse', cluster 'genBasic', data '{"modelId":"DTB-ED2004-012  ","manufacturerName":"databyte.ch     "}' from endpoint 8 with groupID 0
zigbee2mqtt:debug 2020-04-27 21:34:31: Received Zigbee message from '0x00124b001201d226', type 'readResponse', cluster 'genBasic', data '{"powerSource":1,"zclVersion":1}' from endpoint 8 with groupID 0
zigbee2mqtt:debug 2020-04-27 21:34:31: Received Zigbee message from '0x00124b001201d226', type 'readResponse', cluster 'genBasic', data '{}' from endpoint 8 with groupID 0
zigbee2mqtt:debug 2020-04-27 21:34:31: Received Zigbee message from '0x00124b001201d226', type 'readResponse', cluster 'genBasic', data '{"hwVersion":1,"dateCode":"20060831        "}' from endpoint 8 with groupID 0
zigbee2mqtt:debug 2020-04-27 21:34:31: Received Zigbee message from '0x00124b001201d226', type 'readResponse', cluster 'genBasic', data '{"swBuildId":"2004012-001     "}' from endpoint 8 with groupID 0
zigbee2mqtt:info  2020-04-27 21:34:31: Successfully interviewed '0x00124b001201d226', device has successfully been paired
zigbee2mqtt:info  2020-04-27 21:34:31: Device '0x00124b001201d226' is supported, identified as: databyte.ch Panda 1 (DTB-ED2004-012)
zigbee2mqtt:info  2020-04-27 21:34:31: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"pairing","message":"interview_successful","meta":{"friendly_name":"0x00124b001201d226","model":"DTB-ED2004-012","vendor":"databyte.ch","description":"Panda 1","supported":true}}'
```
I also receive some status updates: 

```
zigbee2mqtt:debug 2020-04-27 21:17:48: Received Zigbee message from '0x00124b001201d226', type 'attributeReport', cluster 'genOnOff', data '{"onOff":1}' from endpoint 8 with groupID 0
zigbee2mqtt:info  2020-04-27 21:17:48: MQTT publish: topic 'zigbee2mqtt/0x00124b001201d226', payload '{"state":"ON","linkquality":52}'
zigbee2mqtt:debug 2020-04-27 21:17:51: Received Zigbee message from '0x00124b001201d226', type 'attributeReport', cluster 'genOnOff', data '{"onOff":0}' from endpoint 8 with groupID 0
zigbee2mqtt:info  2020-04-27 21:17:51: MQTT publish: topic 'zigbee2mqtt/0x00124b001201d226', payload '{"state":"OFF","linkquality":52}'

```

But Z2M does not generate a config for HA like the one below: 

```
{
    "brightness": true,
    "schema": "json",
    "command_topic": "zigbee2mqtt/0xd0cf5efffebdd1c0/set",
    "state_topic": "zigbee2mqtt/0xd0cf5efffebdd1c0",
    "json_attributes_topic": "zigbee2mqtt/0xd0cf5efffebdd1c0",
    "name": "0xd0cf5efffebdd1c0_light",
    "unique_id": "0xd0cf5efffebdd1c0_light_zigbee2mqtt",
    "device": {
        "identifiers": [
            "zigbee2mqtt_0xd0cf5efffebdd1c0"
        ],
        "name": "0xd0cf5efffebdd1c0",
        "sw_version": "Zigbee2mqtt 1.12.0",
        "model": "TRADFRI LED bulb E27 1000 lumen, dimmable, opal white (LED1623G12)",
        "manufacturer": "IKEA"
    },
    "availability_topic": "zigbee2mqtt/bridge/state"
}
```

What is needed, that Z2M will generate such a config like the one above for my device?
Thanks for your answer.

matchett808-gh, 2020-04-27:
have a look at lib/extensions/homeassistant.js - you need to add your model and the Hass config in there - something along the lines of:
```
    'DTB-ED2004-012': [cfg.light_brightness],
```

swissbyte, 2020-04-28:
perfect. Thank you! :)

closed 2020-04-28 by swissbyte