Koenkk/zigbee2mqtt #3440: Supported SmartThings Moisture STS-WTR-250 fails to pair

knkaiser, 2020-04-28:
I have a  SmartThings Moisture STS-WTR-250 that I try to pair. The sensor works flawlessly under Smartthings.

However when I try to pair it to Z2M I get an interview error:

  Add support for SmartThings Moisture STS-WTR-250
 
 https://www.gitmemory.com/issue/Koenkk/zigbee-herdsman-converters/922/575931067
 
 
 zigbee2mqtt:info  2020-04-27 22:40:39: Device '0x24fd--------------' joined
Apr 27 22:40:39 espressobin npm[19718]: zigbee2mqtt:info  2020-04-27 22:40:39: Starting interview of '0x24fd--------------'
Apr 27 22:40:39 espressobin npm[19718]: zigbee2mqtt:info  2020-04-27 22:40:39: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"device_connected","message":{"friendly_name":"0x24fd--------------"}}'
Apr 27 22:40:39 espressobin npm[19718]: zigbee2mqtt:info  2020-04-27 22:40:39: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"pairing","message":"interview_started","meta":{"friendly_name":"0x24fd--------------"}}'
Apr 27 22:42:46 espressobin npm[19718]: zigbee2mqtt:info  2020-04-27 22:42:46: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"device_announced","message":"announce","meta":{"friendly_name":"0x24fd--------------"}}'
Apr 27 22:42:52 espressobin npm[19718]: zigbee2mqtt:info  2020-04-27 22:42:52: MQTT publish: topic 'homeassistant/binary_sensor/0x24fd--------------/water_leak/config', payload '{"payload_on":true,"payload_off":false,"value_template":"{{ value_json.water_leak }}","device_class":"moisture","state_topic":"zigbee2mqtt/0x24fd--------------","json_attributes_topic":"zigbee2mqtt/0x24fd--------------","name":"0x24fd--------------_water_leak","unique_id":"0x24fd--------------_water_leak_zigbee2mqtt","device":{"identifiers":["zigbee2mqtt_0x24fd--------------"],"name":"0x24fd--------------","sw_version":"Zigbee2mqtt 1.12.2","model":"Water leak sensor (2016 model) (STS-WTR-250)","manufacturer":"SmartThings"},"availability_topic":"zigbee2mqtt/bridge/state"}'
Apr 27 22:42:52 espressobin npm[19718]: zigbee2mqtt:info  2020-04-27 22:42:52: MQTT publish: topic 'homeassistant/sensor/0x24fd--------------/battery/config', payload '{"unit_of_measurement":"%","device_class":"battery","value_template":"{{ value_json.battery }}","state_topic":"zigbee2mqtt/0x24fd--------------","json_attributes_topic":"zigbee2mqtt/0x24fd--------------","name":"0x24fd--------------_battery","unique_id":"0x24fd--------------_battery_zigbee2mqtt","device":{"identifiers":["zigbee2mqtt_0x24fd--------------"],"name":"0x24fd--------------","sw_version":"Zigbee2mqtt 1.12.2","model":"Water leak sensor (2016 model) (STS-WTR-250)","manufacturer":"SmartThings"},"availability_topic":"zigbee2mqtt/bridge/state"}'
Apr 27 22:42:52 espressobin npm[19718]: zigbee2mqtt:info  2020-04-27 22:42:52: MQTT publish: topic 'homeassistant/sensor/0x24fd--------------/temperature/config', payload '{"unit_of_measurement":"‹C","device_class":"temperature","value_template":"{{ value_json.temperature }}","state_topic":"zigbee2mqtt/0x24fd--------------","json_attributes_topic":"zigbee2mqtt/0x24fd--------------","name":"0x24fd--------------_temperature","unique_id":"0x24fd--------------_temperature_zigbee2mqtt","device":{"identifiers":["zigbee2mqtt_0x24fd--------------"],"name":"0x24fd--------------","sw_version":"Zigbee2mqtt 1.12.2","model":"Water leak sensor (2016 model) (STS-WTR-250)","manufacturer":"SmartThings"},"availability_topic":"zigbee2mqtt/bridge/state"}'
Apr 27 22:42:52 espressobin npm[19718]: zigbee2mqtt:info  2020-04-27 22:42:52: MQTT publish: topic 'homeassistant/sensor/0x24fd--------------/linkquality/config', payload '{"icon":"mdi:signal","unit_of_measurement":"lqi","value_template":"{{ value_json.linkquality }}","state_topic":"zigbee2mqtt/0x24fd--------------","json_attributes_topic":"zigbee2mqtt/0x24fd--------------","name":"0x24fd--------------_linkquality","unique_id":"0x24fd--------------_linkquality_zigbee2mqtt","device":{"identifiers":["zigbee2mqtt_0x24fd--------------"],"name":"0x24fd--------------","sw_version":"Zigbee2mqtt 1.12.2","model":"Water leak sensor (2016 model) (STS-WTR-250)","manufacturer":"SmartThings"},"availability_topic":"zigbee2mqtt/bridge/state"}'
Apr 27 22:42:56 espressobin npm[19718]: zigbee2mqtt:info  2020-04-27 22:42:56: MQTT publish: topic 'zigbee2mqtt/Smart Switch', payload '{"battery":74,"voltage":2955,"linkquality":65}'
Apr 27 22:43:11 espressobin npm[19718]: zigbee2mqtt:error 2020-04-27 22:43:11: Failed to interview '0x24fd--------------', device has not successfully been paired
Apr 27 22:43:11 espressobin npm[19718]: zigbee2mqtt:info  2020-04-27 22:43:11: Configuring '0x24fd--------------'
Apr 27 22:43:11 espressobin npm[19718]: zigbee2mqtt:info  2020-04-27 22:43:11: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"pairing","message":"interview_failed","meta":{"friendly_name":"0x24fd--------------"}}'
Apr 27 22:43:21 espressobin npm[19718]: zigbee2mqtt:error 2020-04-27 22:43:21: Failed to configure '0x24fd--------------', attempt 1 (Error: Bind 0x24fd--------------/1 genPowerCfg from '0x00124b000e0ea8bf/1' failed (Error: AREQ - ZDO - bindRsp after 10000ms)
Apr 27 22:43:21 espressobin npm[19718]:     at Endpoint.<anonymous> (/srv/zigbee2mqtt/node_modules/zigbee-herdsman/dist/controller/model/endpoint.js:244:23)
Apr 27 22:43:21 espressobin npm[19718]:     at Generator.throw (<anonymous>)
Apr 27 22:43:21 espressobin npm[19718]:     at rejected (/srv/zigbee2mqtt/node_modules/zigbee-herdsman/dist/controller/model/endpoint.js:6:65))

I tried repairing many times but without luck. 

A details herdsman log is available here: https://www.dropbox.com/s/ciqy2wzqaros9v3/Smartthings%20Leak.txt?dl=0

knkaiser, 2020-05-10:
After removing the device from the device.js file and trying to pair it, it started to work, then I just added device back to the device,js file and it successfully paired. I suspect the device was stuck in a bad state. Case is closed.

closed 2020-05-10 by knkaiser

zijings, 2020-06-17:
> After removing the device from the device.js file and trying to pair it, it started to work, then I just added device back to the device,js file and it successfully paired. I suspect the device was stuck in a bad state. Case is closed.

hello, I have met the same problem, however, removed the device and added it back to the device.js file didn't work, the new device still can't be paired successfully. by the way, the other new device I try to support didn't meet the problem and has been paired successfully, it is confusing, and I dk where the problem is. Do you have any idea or do you meet the same problem again?