Koenkk/zigbee2mqtt #3442: CC2530 / CC2652R Hard reset

liparrok, 2020-04-28:
Hey guys,

this is not an issue its more or less a feature request.  I did not know where to place them but here :)

Since I have all USB ports full and its SPI/I2C to UART IC driver already there, on raspberry pi, working well, I decided to develop shield for raspberry pi. 
Here are some specs:
- SC16IS752 SPI/I2C to UART converter (both options available, SPI and I2C)
- CC2530 (E18-MS1PA1-PCB or E18-MS1-PCB) module on board, same config as CC2531 - switches, leds,....
- CC2652R chip solution with PCB antenna / U.FL connector
- Additional serial indication leds on board (TX and RX) driven with SLG46826
- HC-12 module for raspberry pi Serial (not sure yet)
At this point I decided rather not use raspberry default serial, which can be very usefull for debugging, SC16IS752 works well at my table :)

Also I am going to connect CC2530/CC2652 reset pins to raspberry pi GPIO pins. Is possible to add control to zigbee2mqtt application. Lets say in config file and drive them if coordinator freezes?

Well if its anyone else interested in that kind of pcb with/or without components let me know, so I order enough pcbs.

Cheers, Rok

sjorge, 2020-04-28:
Not sure how hard resetting but you can soft reset it via mqtt if z2m is running: 
https://www.zigbee2mqtt.io/information/mqtt_topics_and_message_structure.html#zigbee2mqttbridgeconfigreset

Also have you looked at https://electrolama.com/projects/zoe/

liparrok, 2020-04-28:
Yes, you can do soft-reset by sending reset command over usb, serial,.... to target device, CC2531 or CC2530 for example. But what do you do if your device freezes, there is no other way to reset it but power-cycle or reset with dedicated pin on procesor. 
What should do app then, that is what I think, I am hw guy, is constant pinging zigbee unit ( once per 10sec) and if there is not response you do reset with either power cycle or toggling reset pin.

I check electrolama zoe. I just dont link using raspberry pi serial. In many cases its too useful for debugging. I rather add SPI to UART converter and keep raspberry's serial untouched.

Hopefully I made my thinking clear enough

closed 2020-07-05 by stale[bot]