Koenkk/zigbee2mqtt #3443: Philips HUe Go - problem information

ludoviclt, 2020-04-28:
hi sorry for my english
i try to had my philips hue go but it seem not working

when i reset my philips hue go, the logs in zigbee2mqtt are empty

i have this one : Philips 7146060PH
Adapter hardware: CC2531
zigbee2mqtt@1.12.2

<
zigbee2mqtt:info  2020-04-28 16:54:00: 0x00178801060c52c6 (0x00178801060c52c6): 7146060PH - Philips Hue Go (Router)
.
.
.
zigbee2mqtt:debug 2020-04-28 16:54:01: Received MQTT message on 'zigbee2mqtt/friendly_name/set' with data '["color":["hue":100]]'
.
.
.
zigbee2mqtt:info  2020-04-28 16:54:01: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"entity_not_found","message":{"friendly_name":"friendly_name"}}'
zigbee2mqtt:error 2020-04-28 16:54:01: Entity 'friendly_name' is unknown
zigbee2mqtt:info  2020-04-28 16:54:01: MQTT publish: topic 'zigbee2mqtt/bridge/config', payload '{"version":"1.12.2","commit":"911cd58","coordinator":{"type":"zStack12","meta":{"transportrev":2,"product":0,"majorrel":2,"minorrel":6,"maintrel":3,"revision":20190608}},"log_level":"debug","permit_join":true}'

  zigbee-herdsman:adapter:zStack:znp:AREQ <-- ZDO - endDeviceAnnceInd - {"srcaddr":32144,"nwkaddr":32144,"ieeeaddr":"0x00178801060c52c6","capabilities":142} +3m
  zigbee-herdsman:controller:log Device announce '0x00178801060c52c6' +3m
zigbee2mqtt:debug 2020-04-28 16:56:38: Device '0x00178801060c52c6' announced itself
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [] +58ms
zigbee2mqtt:info  2020-04-28 16:56:38: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"device_announced","message":"announce","meta":{"friendly_name":"0x00178801060c52c6"}}'
  zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,12,69,202,144,125,198,82,12,6,1,136,23,0,0,0,110] +7s
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,12,69,202,144,125,198,82,12,6,1,136,23,0,0,0,110] +5ms
  zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 12 - 2 - 5 - 202 - [144,125,198,82,12,6,1,136,23,0,0,0] - 110 +26ms
  zigbee-herdsman:adapter:zStack:znp:AREQ <-- ZDO - tcDeviceInd - {"nwkaddr":32144,"extaddr":"0x00178801060c52c6","parentaddr":0} +8s
  zigbee-herdsman:controller:log Device '0x00178801060c52c6' joined +8s
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [] +44ms
  zigbee-herdsman:controller:log Device '0x00178801060c52c6' accepted by handler +22ms
  zigbee-herdsman:controller:log Not interviewing '0x00178801060c52c6', completed 'true', in progress 'false' +9ms
> 

thank for your help

Koenkk, 2020-04-29:
Try resetting via https://www.zigbee2mqtt.io/information/touchlink.html

ludoviclt, 2020-04-30:
thank i will try this

ludoviclt, 2020-04-30:
this is the result

nothing on the mqttclient of jeedom

Apr 30 20:33:05 Jeedom npm[1273]: zigbee2mqtt:info  2020-04-30 20:33:05: Starting touchlink factory reset...
Apr 30 20:33:05 Jeedom npm[1273]: zigbee2mqtt:info  2020-04-30 20:33:05: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"touchlink","message":"reset_started","meta":{"status":"started"}}'
Apr 30 20:33:08 Jeedom npm[1273]: zigbee2mqtt:info  2020-04-30 20:33:08: Successfully factory reset device through Touchlink
Apr 30 20:33:08 Jeedom npm[1273]: zigbee2mqtt:info  2020-04-30 20:33:08: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"touchlink","message":"reset_success","meta":{"status":"success"}}'
Apr 30 20:33:15 Jeedom npm[1273]: zigbee2mqtt:info  2020-04-30 20:33:15: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"device_announced","message":"announce","meta":{"friendly_name":"Philips_Hue_Go"}}'

ludoviclt, 2020-04-30:
when i run this

zigbee2mqtt/Philips_Hue_Go/set

my philips hue go switch off the light

i have the reponse 

Apr 30 21:55:31 Jeedom npm[1273]: zigbee2mqtt:info  2020-04-30 21:55:31: MQTT publish: topic 'zigbee2mqtt/Philips_Hue_Go', payload '{"state":"OFF","brightness":0}'

Koenkk, 2020-05-02:
If you can control it from MQTT now it means the zigbee2mqtt part works now.

ludoviclt, 2020-05-02:
do you the the order to switch on the light
where i can find all the action and info possible with philips hue go

thanks

Koenkk, 2020-05-02:
You can find the documentation here; https://www.zigbee2mqtt.io/information/mqtt_topics_and_message_structure.html#zigbee2mqttfriendly_nameset

I will close this as the original issue has been solved.

closed 2020-05-02 by Koenkk