Koenkk/zigbee2mqtt #3444: IKEA 1743 - Switch - Sudden drop/drain of battery

tsprivate, 2020-04-28:
I just invested in a few of the IKEA 1743 - Switches. The switches were easily detected by Z2M and worked like a charm besides a binding issue (other thread).

As part of device management I record the battery state of my devices and realized below behavior on all 3 Ikea switches

1) I started with the original battery for the setup and they state was showing 100%
2) After 2-3days the battery state dropped from 100% to 87% (no value change in between)
3) After another 2 days the battery was fully drained (Last msg from Z2m still showed 87 for the device). 
4) I replaced the battery with new Maxell batteries (as I had bad experience with original IKEA before)
5) the same scenario (1-3) happened again (as said on all 3 devices)

`{"update_available":false,"last_seen":"2020-04-27T20:52:46+10:00","linkquality":54,"battery":87,"device":{"friendlyName":"Switch.New","model":"E1743","ieeeAddr":"0xec1bbdfffe9105ad","networkAddress":9755,"type":"EndDevice","manufacturerID":4476,"manufacturerName":"IKEA of Sweden","powerSource":"Battery","applicationVersion":34,"stackVersion":98,"zclVersion":3,"hardwareVersion":1,"dateCode":"20190410","softwareBuildID":"2.2.008"}}`
Have you experienced similar?  Any advice?

## Debug Info
Zigbee2mqtt version: 1.12.2
Adapter hardware: CC26X2R1
Adapter firmware version: 3.x.0
Device count: 36

highlandr, 2020-04-30:
Hello. Had a battery drain yesterday as well on a motion sensor from Ikea, could be related or not... At the moment all ikea has no battery status.

[image]

Thanks.

nikdoof, 2020-05-25:
I know this comment will be unhelpful to your situations, but In terms of actually tracking the battery % I wouldn't take too much notice of the IKEA device reports. The values seem arbitrary at best based on the voltage of the battery, which varies with temperature. I've seen the same model of switch with a new battery go from 100% to 11% within a minute and sit there for 3 months. 

Here is one of our least uses switches (probably one press per week):

[image]

I found that IKEA switches are generally more battery hungry than a equivalent Hue switch, but i've just kept to having a stack of CR2032/16 in the house for when needed.

tsprivate, 2020-05-26:
I don't rely on the battery status .. but I had it now several timed (different devices) that I put in a new battery and it was drained within days (sometimes even on same day!)

I saw that behavior only with the IKEA switches and remotes

MattWestb, 2020-07-19:
I was buying 3 more combi pack (E1743 and the cheapest bulb for 10€) before corona shutdown.
All 3 bulbs is working as expected and 2 E1743. The 3:d was eating the first battery before the the firmware was updated. In 3 weeks it was "eating" 4 batterys without being active used only being pared and laying on the table.
Ended up standing 2 hours in que with 2 meters distance for giving it back (the combi pack) and and then going back in and grabbing one new one that its still working normal.
Cost in € = 0  in time = 4 hours without meatballs then the bistro was still not open :-(

tsprivate, 2020-07-20:
as mentioned earlier I had the same issue, the batteries were fully drained (i do not trust the battery status but did measure them with a voltmeter.. it looks like the battery status was correct after all).... I now did a full factory reset of the devices and replaced them with fresh batteries ... so far the have lasted 3 weeks .. and even the batter status is showing correctly..so far

closed 2020-08-28 by github-actions[bot]

boojew, 2021-03-10:
Just experienced this myself.. I wonder if its a z2m issue or Tradfri issue.

FrancYescO, 2021-05-29:
Same here.. from 100% to 10% in 3weeks (with the button mainly unused)
Also the E1810 seems dropping battery too fast...

MattWestb, 2021-05-29:
Not very likely (but possible) the IKEA controller.
Very fast battery draining is very common with Sonoff ZBB (48 hours on on battery) and is coming from bad deigned hardware and was also problem with the firmware that shall being fixed but not the bad hardware.

What coordinator hardware is you using @FrancYescO  ?

ZHA have fixing some problems with then long pull time was very tight and was reducing the battery life time and is now using the same as IKEA GW is using for getting better battery life time.

One thing with IKEA controlling devices is only updating the battery status then being triggered (button pressed) and is between reporting the "last read". So is you is not  using  it for 2 weeks its reporting  the same value to the coordinator until you is pressing on button then its updating the attribute and reporting the new value until next button being pressed.

I think my returned remote was not bad, i was using it under tasmota that was having some bugs that was draining the battery in some days but that is fixed for some mouths ago with updated firmware in the NCP.

FrancYescO, 2021-05-29:
CC2531

Zigbee2MQTT version
    1.18.3 commit: 1.18.3

Coordinator type
    zStack12

Coordinator revision
    20201127

Frontend version
    0.4.0

MattWestb, 2021-05-29:
The CC2531 is not known having problems with sleeping end device but it can being possible.
Do you have good routers (=no OSRAM routers) around the remotes or is all connected direct to the coordinator ?

I think (= not knowing) that Z2M is not having one big problem with IKEA remotes then it shall being more complains about it.

FrancYescO, 2021-05-29:
[image]

I bought the ikea buttons in pack with  LED1836G9 and LED1732G11 but during the discharging weeks there were unpaired, there was just 1 Xiaomi ZNLDP12LM as a router and the button was mainly connected to the coordinator.

Going to replace the battery an try to updated the firmare with ikea test servers (if a new one is available..) hopefully to not waste another battery

ZNLDP12LM

MattWestb, 2021-05-29:
The E1743 dont  have getting any new firmware for around one year but most lights have getting updates and some is only working on the test server.

The "letto" is having very good signal to its parent and the parent is having good signal to the coordinator so shall no having problems. The "cucina" is having very bad signal and must using much more power for "talking" with is parent (coordinator) and is very likely eating batteries.

Also its better having end device (sleeping or not) to routers and not the coordinator its possible so they is not getting problem then you is restarting the host system.

If being not paired battery powered device can still burning batteries and not being in deep sleep but its on other story only good to knowing.

MattWestb, 2021-05-29:
Both is working OK but i think you very soon need one better coordinator then the CC-2531 is little week then adding little more devices and its having bad radio performance (you can see the  LIQ from / to it) but if its working OK its not needed :-)))

FrancYescO, 2021-05-29:
Yep i've also modded the 2531 but seems the external antenna lowered LQI :D, more than this i was hoping some router around will help me to have a good LQI everywhere (unfortunately the xiaomi contact sensors does not migrate to router, so i'll have to repair some that actually are very faraway..)

BTW, bad to say in the z2m repo, but fully migrated to ZHA (i was having some issue only with these buttons and blueprint https://community.home-assistant.io/t/zha-deconz-zigbee2mqtt-ikea-e1743-on-off-switch-dimmer-universal-blueprint-all-actions-double-click-events-control-lights-media-players-and-more-with-hooks/277861/50?u=francyesco)

The repairing took very short, with z2m in the past i needed to deatach CC2531 turn off power to routers and some other pain, another 30min to rename all entities, and now i'll see how its going for the next weeks i just miss out the z2m frontend :D

MattWestb, 2021-05-29:
The TI zStack12 is only HA1.2 and all devices is using the same link key so its fast and easy "moving around" but i can getting corrupted NVM.
Also the PCB and external antenna is classical with all CC-253X chips but its one cheep and old chip but good  for starting playing with Zigbee.

Xiaomi sleeping end devices (no zigbee 3 ones) is never changing its parent and if its getting problems its leaving the network and is being lost and must being repaired.

In ZHA i using "adding device thru this device" for getting Xiaomi devices having good router but some device dont like all routers so it can being little tricky to getting them having the router you like.

That also the reason not having the coordinator as parent if the Xiaomi device is trying talking to you coordinator and its not online (you is restarting the host system for software update) then its very likely leaving the network but if its have one router as parent its still online and all is working OK also then the host system is offline.

MattWestb, 2021-05-29:
By the way you shall binding the IKEA remotes to groups of light then they is also working then the host system is offline and you dont need doing automatons in HA that can stop working if soft or hardware is crashing. (both in Z2M and ZHA)

FrancYescO, 2021-05-29:
just explored the cluster/binding possibility, i cannot get it to work (probably because is conflicting with the automation, or cuz i'm trying to bing a xiaomi light to ikea remote idk need to study about :D) and i'll surely use the _add thru this device_ func to bind the contact sensor (the one that i want to be connected to the coordinator, was paired damn fast with ZHA, with Z2M was pretty challenging)

BTW, i haven't changed the battery to the IKEA 1743 remote (was at 15% with Z2M), and ZHA is reporting is at 100% ....

MattWestb, 2021-05-29:
IKEA controller can only being bonded to groups not direct to devices so first making on group and putting the light i it and then binding the remote to the group.
I can also being needed unbinding the coordinator from the remote before its accepting on new binding (the coordinator is added then pairing the device).

ZHA is using "fast pairing" for Xiaomi devices and dont do it the "normal way" then they is not doing all things by Zigbee standard but its making it very fast!!

They is also refactoring the normal pairing then many devices is having problem  and leaving before finishing the pairing (Zigbee 3 devices that need getting on new TC link key) but its not finished.

IKEA controller is "jumping" very much up and down in the % reported its the "nature" of them and Xiaomi is more one line then looking in the historical data from them but Xiaomi is not reporting % they is only reporting V and the % is calculated in the host system.

foorschtbar, 2021-12-13:
Has anyone here been able to solve the problem? My IKEA 1743 remote control also only lasts 24h and the battery is then discharged to 2.2V and stops working. I use a Silverest (LIDL) Gateway with the EZSP Protocol.

MattWestb, 2021-12-13:
Wot version is you running on the ZBGW the original EZSP 6.5.0.1 or the updated 6.7.8.0 ?
Also have you updated the switch (27 October released) firmware in your E1743 ?
Do you have route device around that it can using or must it  using  the coordinator ?

I was missing one in some hours for sniffing what is doing then is in draining mode:
[image]
I was doing wrong then updating the firmware in living room and putting it back in bedroom and not repairing it so it was using on bad router or the coordinator and was jumping to one new parent  and was hitting the bug.

foorschtbar, 2021-12-13:
> Wot version is you running on the ZBGW the original EZSP 6.5.0.1 or the updated 6.7.8.0 ? Also have you updated the switch (27 October released) firmware in your E1743 ? Do you have route device around that it can using or must it using the coordinator ?

ZBGW was updated to 6.7.8.0
Switch Firmware: The dashboards shows: 20190723/2.2.010

> I was missing one in some hours for sniffing what is doing then is in draining mode: [image] I was doing wrong then updating the firmware in living room and putting it back in bedroom and not [...]

I have only one ZBGW and one Switch. I added them to Z2M an configure nothing there. All the other stuff in Node-Red

rdorys, 2022-01-16:
Hi, maybe some hints : 
I paired 2 brand new E1473 to Z2M (with ikea battery, same firmware  2.2.010, same route on the network and located at the same place, LQI > 60). In the default configuration, each one of them had default battery report enabled. 

[image]

I deactivated this report for the second one (test 2) and I clicked them each 10 times along the day. The results are quite interesting...

[image]

But this is to early to conclude anything by now...

### What's Next :
I will wait until test 1 is fully discharged, and I will put 2 brand new batteries (Energyser or any battery I can trust) and I will configure the report only for test 2 and see what happens. 

- If test 1 discharges quickly again (and not test 2), it will point a hardware issue for test 1 and we will exclude the "report" issue.

- If test 2 discharges quickly and not test 1, it will indicate that this report configuration may be the source of the issue.

There are many other cases which won't allow to conclude anything...

I will keep you in touch. Feel free to deactivate this report to see if there is any improvement. It's a bit of work to test all cases alone :)

Regards

foorschtbar, 2022-01-17:
> Hi, maybe some hints : I paired 2 brand new E1473 to Z2M (with ikea battery, same firmware 2.2.010, same route on the network and located at the same place, LQI > 60). In the default configuration, each one of them had default battery report enabled.

This is very interesting. I don't know much about Zigbee yet. Can I assume that the battery report is not stored in the remote control from the factory, but is set by the gateway?

I noticed that the problem "must" be with Z2M. I have the same configuration (remote + gateway) in a different location, but integrated with Home Assistant and it has been working for a few months without any problems.

rdorys, 2022-01-17:
@foorschtbar I'm not convinced it is totally a Z2M problem.

The fact is that before switching to zigstar (based on cc2652p), I used a cc2531 without such battery issue on Z2M.

If these default reports are the key of the battery issue (that's what I'm trying to proof with good methods) , it may be a configuration problem during the pairing of the device.

If someone on this thread tried with ZHA, I'm curious to know if these reports are configured by default.

Wait and see... I need time to get a solid conviction.

foorschtbar, 2022-01-17:
Do u know how i could read out the settings from a device with Zigbee Home Assistent? 

I could test also one of my E1473. I would gladly donate a battery for this test, but i cant disable the reporting. When i click on disable, i got this:

```
Received MQTT message on 'zigbee2mqtt/bridge/request/device/configure_reporting' with data '{"attribute":"batteryPercentageRemaining","cluster":"genPowerCfg","id":"ikea_test_btn/1","maximum_report_interval":65535,"minimum_report_interval":3600,"reportable_change":0,"transaction":"fdvj0-10"}'

Request 'zigbee2mqtt/bridge/request/device/configure_reporting' failed with error: 'Bind XXX/1 genPowerCfg from 'XXX/1' failed ({"address":6957,"clusterId":32801,"sequence":48} after 30000ms)'

Error: Bind 0xcc86ecfffe88f57a/1 genPowerCfg from 'XXX/1' failed ({"address":6957,"clusterId":32801,"sequence":48} after 30000ms) at Timeout._onTimeout (/app/node_modules/zigbee-herdsman/src/utils/waitress.ts:64:35) at listOnTimeout (internal/timers.js:557:17) at processTimers (internal/timers.js:500:7)

MQTT publish: topic 'zigbee2mqtt/bridge/response/device/configure_reporting', payload '{"data":{},"error":"Bind XXX/1 genPowerCfg from 'XXX/1' failed ({\"address\":6957,\"clusterId\":32801,\"sequence\":48} after 30000ms)","status":"error","transaction":"fdvj0-10"}'
```

rdorys, 2022-01-17:
Do you wake up the E1473 just before publishing the packet ? The device needs to be in a listenning state.
[image]
I only did it from the GUI clicking "Disable" and clicking the E1473.

Good Luck !

I don't know if the this reports is visible from HA. I only see it in Z2M frontend.

Keep us in touch if you manage to disable the reports and if battery performance gets better. 🤞

FrancYescO, 2022-01-17:
if you are interesed in: my E1743 on Z2M is reporting 10% of battery from ~9months, 5% maybe from the previous 2 week, never replaced and it still works, i assume the battery reporting it's pretty unrealiable, i'll wait until the remote is fully dead...

foorschtbar, 2022-01-17:
@rdorys I tried again, now I have no error in the user interface (and in the log), but also no success message. How can I see that the battery report is disabled?

@FrancYescO problem is not the wrong reporting. My (and some others) batteries drain completely after a few hours without using the switch

rdorys, 2022-01-17:
> @rdorys I tried again, now I have no error in the user interface (and in the log), but also no success message. How can I see that the battery report is disabled?

Sorry, I don't have this level in Zigbee to tell you .

Maybe you could try to reset your E1473 and pair it again, and then disable the report. It may be good to restart Z2M before and unplug/plug the coordinator.

@FrancYescO could you tell us what is the firmware of your E1473 and your setup ? Is your E1473 has anything in the report tab ?

I seems I didn't have the glitch before swithing to a CC2652P stick.

FrancYescO, 2022-01-17:
@rdorys hmm report tab.. where?

rdorys, 2022-01-17:
@FrancYescO in Z2M, when you click on the device name, you can see a "reporting" tab:
[image]
I never used ZHA by the way.

FrancYescO, 2022-01-17:
Ah, no sorry as i wrongly wrote Z3M in previous comments i migrated to ZHA months ago..

foorschtbar, 2022-01-17:
I have a special device for measuring and recording extremely low currents. Next weekend I will replace the battery of a remote control and test the power consumption with default settings and disabled reporting.

foorschtbar, 2022-01-22:
I have not so mutch time at the moment, but i did a short test: There is something very very wrong with the IKEA button. I re-paired the button and disabled the battery report (click disable and pressed the physical button several times. after a few moments the entry goes away). 

What i see is, that the button wakes up every 10 minutes for a short time (without pressing it). After 50 minutes it doesn't seem to go back to deep sleep and the power consumption is stable at 5 mA! That drains any button battery in a few hours.

[image]

For comparison, I also quickly tested an Aquara Button. The spikes are from pressing the button. This button works fine:

[image]

rdorys, 2022-01-22:
@foorschtbar great job, this is super interesting. I could be nice to see if activating battery report changes anything.

On my side, work is still in progress :

[image]

Test 1 & 3 have the reports, and only 2 have no report, but it is too soon to conclude anything (and voltage measurement is so unreliable on ikea remotes...).

In my case, the battery never dies within a day but in few weeks. But if I had a 5 mA leak, my 3 remotes would have died in a day for sure ! -> I'm not convinced we have the same issue !

Just for information, what devices did you used for these tests ? I wish I had this stuff at home !

MattWestb, 2022-01-22:
Most (= all but not the old motion sensor) is Zigbee 3 controller and is pulling there parent if they have some "news" for them every some minutes (its depend how the coordinator / system have configuring them). And is the coordinator and its parent is supporting it  they is also doing checkins every 55 minutes to the coordinator so there you have the paten.

All Silabs based device that is not have one very nes Zigbee stack have one (or more) that can making the device is cant going in sleep and it normally triggered i then its changing its parent or the parent is not behaving well.

Best workaround is forcing them using one good router (= no OSRAM plug) then paring them and also blocking the coordinator for having direct children then all coordinators is very bad routers then they is doing much "other" work.

Mos user with rapidly (48 hours) is having the controllers connected to coordinator and not to routers. The same have being seen on some smatrthongs controllers and Philips HUE so its not only IKEA but  they is well user = more exposed of user.

foorschtbar, 2022-01-22:
@rdorys i use a [Power Profiler Kit II](https://www.nordicsemi.com/Products/Development-hardware/Power-Profiler-Kit-2) from Nordic Semiconductor. 

@MattWestb thats very interesting. thats my setup. a coordinator and only childrens (buttons). no plugs or something.

MattWestb, 2022-01-22:
Most problem in ZHA was user with Somoff ZBB without routers but have being seen on all EZSP, deCONZ and TI coordinators. EZSP have getting firmware update that making it better but but not 100% if having the controllers connected to the coordinator.

I have one E1743 that have working well for 7 months and  i was upgrading the firmware and charging battery (after the upgrade it as very low) and then moving it back to my sleeping room and was killed in some days = it was jumping to one new prent and the bug was triggered.

I think your device is the "medium" draining and is normally the device can shutting down the external EEPROM its using for storing OTA files.
Thee 48 hours draining is like doing OTA updating all the time and your patent is not then is not using the radio all the time.

foorschtbar, 2022-01-26:
So, i added a Router (IKEA E1708). Sometime the Switch connected to it, sometimes not. The only change is, that the 10min-spikes go away (middle of picture), but the 5mA current keeps.

|  ------ connect to coordinators ----- |  re-pair and auto-connect to router - |  --------- re-pain and no connect to router --------- |
[image]

MattWestb, 2022-01-26:
I have looking for the reference design for Sirlabs EFR32MG1 with power consumption for the external flash chip but was not finding it but the ship manufacture is writing 5 mA read mode and some nanoo A the power save mode.
Sirlab was having one bug that was making the external flash not being powered down then the zigbee ship was going in sleep mode and was still awake and using power (5mA) and its fixed in the SDK but i have not finding in which version it was fixed but i think is newer then EZSP 6.3 so very likely not implanted in this firmware.

Also the first gen 5 button remote was having the first gen Zigbee module that was having software power saving of the external flash and the new one is cutting the power to it for saving power and not using the internal stack for doing it with software commands.

Your paten looks very much being that the external flash is still awake then the device is sleeping but it around 10 more bugs that  can being trigeried and doing the same.
If like reading the fixed bugs in Silabs Zigbee stack then one good source is here https://community.silabs.com/s/article/reference-collection-of-ember-znet-and-afv2-release-notes?language=en_US.
The new Zigbee stack version used in all upgraded 1 and 2 gen devices is EZSP 6.3.1 so much to reading then the newest is EZSP 7.0

MattWestb, 2022-01-26:
One more thing is then your remote is connected to the outlet is still doing pulling of its parent every 5 minutes but is using much less power (the very small picks) and then being connected to your coordinator its using higher power (higher sikes) every 5 minutes.
And also it can being that its using pull control then being connected to the outlet but not to the coordinator and the coordinator is not so good answering its children so it can needing more time = the child must using more radio on time.

syyyr, 2022-01-28:
Hi, I have a similar issue with my E1524/E1810 remote:
```
Jan 18 13:10:08 raspberrypi npm[30681]: Zigbee2MQTT:info  2022-01-18 13:10:08: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"announce","meta":{"friendly_name":"remote"},"type":"device_announced"}'
Jan 18 18:24:14 raspberrypi npm[30681]: Zigbee2MQTT:info  2022-01-18 18:24:14: MQTT publish: topic 'zigbee2mqtt/remote', payload '{"battery":100,"linkquality":84,"update":{"state":"idle"},"update_available":false}'
Jan 18 21:32:23 raspberrypi npm[30681]: Zigbee2MQTT:info  2022-01-18 21:32:23: MQTT publish: topic 'zigbee2mqtt/remote', payload '{"battery":11,"linkquality":75,"update":{"state":"idle"},"update_available":false}'
```
At 13:00 I put a new battery in, at 18:00, it says that it's still 100 % and at 21:00 it's already at 11 %. I am not completely sure what happened, but I did update the devices firmware before it started happening and also changed my Zigbee adapter to "SMARTLIGHT CC2652P Zigbee USB Adapter SLZB-02".

West408, 2022-02-04:
I've gotten the same problem, it has happened twice since I set up my Hoobs Zigbee2MQTT network, and weirdly only with this one Trådfri Switch, the other 3 haven't had high drain. 
Both times is says low battery and in the network map it disconnected. Changed batteries last time and it worked again. Seems odd that the CR3032 would die within a week..

[image]

suprimex, 2022-09-05:
> Could [#9117 (comment)](https://github.com/Koenkk/zigbee2mqtt/issues/9117#issuecomment-1017479712) be related?
> 
> I followed those steps after migrating from CC2531 to a Sonoff Zigbee3.0 USB Dongle. I was not able to pair new devices, but also the IKEA battery powered devices battery drained in just 1-3 days. After the mentioned steps and adjusting extended_pan_id I could pair again new devices, but also the IKEA devices have reported stable battery level since 1 week.
> 
> Edit: to be more complete: This was happening with IKEA E1524/E1810 remote controls. Several of them I re-paired after replacing drained batteries. They are still up now, so now 2 weeks. 2 others I did not repair after the extended_pan_id change, and they drained again in a few days. So I re-paired them too and they have reported stable battery level for 1 week too now

Hello, how is it doing now? is it still on the same battery (as I understand correctly, 3 months already passed.)

mjanssens, 2022-09-05:
> Hello, how is it doing now? is it still on the same battery (as I understand correctly, 3 months already passed.)

Nothing really changed. All devices kept on working, no need to replace batteries, so I also didn't pay attention to this issue anymore.
Just looked and all E1810 are on 87% battery level, as well as one E1743.
At this moment I run z2m version 1.27.0

hnykda, 2022-11-10:
I have seen this with my E1743, battery got drained twice in two days without even touching the thing. Trying on a different zigbee network now + reconfigured, so will see what happens. Latest available firmware, latest zigbee, Sonoff 3 dongle. Quite surprising.