Koenkk/zigbee2mqtt #3445: Xiaomi switch WXKG01LM pulsation detection failure

kitopopo, 2020-04-28:
Hi dear @Koenkk , I hope you are well with this epidemic. I send him a lot of strength,

I will explain my problem with three  xiamo switchs (WXKG01LM)

## What happened

I have problems with the xiaomi switch  WXKG01LM (pulsation detection failure).
The switch has always worked correctly but since the last update I noticed that it does not correctly detect the pulsations, single, double, triple, quadruple,long..etc .
Doing a double click detects quadruple or triple or single. Making a triple click zigbe detects long click... it is random but and hardly ever matches .The problem is solved by stopping the zigbee2mqtt service and starting it again.

In the debug info i leave you a sample of the log, in this example ,In the last press was an individual click but zigbee2mqtt detects t other clicks. I wanted to turn on the light and the home theater came on and repeating a message through the speakers 3 times in a row. horrible. Single click should have turned on the light and double-clicking should have played the message through the speakers 
but not all at once. the light went out and the message was played many times.
I have 2 xiaomi switch more and all fail with this problem. I have tryied delete the three switches and pairing again but the problem persist.

Thanks in advanced and best regards.

## What did you expect to happen

the number of pulses emitted must coincide with the number of pulsations received

## How to reproduce it (minimal and precise)

testing single click, double click or triple click after some hours of start zigbee2mqtt service.

## Debug Info

``
abr 29 00:33:39 raspberrypi npm[416]: zigbee2mqtt:info  2020-04-29 00:33:39: MQTT publish: topic 'zigbee2mqtt/Denon plug', payload '{"state":"ON","power":7.79,"voltage":null,"consumption":68.39,"temperature":32,"linkquality":102}'
abr 29 00:34:06 raspberrypi npm[416]: zigbee2mqtt:info  2020-04-29 00:34:06: MQTT publish: topic 'zigbee2mqtt/Switch salon', payload '{"battery":100,"voltage":3042,"linkquality":44,"click":"quadruple"}'
abr 29 00:34:11 raspberrypi npm[416]: zigbee2mqtt:info  2020-04-29 00:34:11: MQTT publish: topic 'zigbee2mqtt/Switch salon', payload '{"battery":100,"voltage":3042,"linkquality":44,"click":"quadruple"}'
abr 29 00:34:15 raspberrypi npm[416]: zigbee2mqtt:info  2020-04-29 00:34:15: MQTT publish: topic 'zigbee2mqtt/Switch salon', payload '{"battery":100,"voltage":3042,"linkquality":47,"click":"triple"}'
abr 29 00:34:24 raspberrypi npm[416]: zigbee2mqtt:info  2020-04-29 00:34:24: MQTT publish: topic 'zigbee2mqtt/Hdd plug', payload '{"state":"ON","linkquality":113,"power":5.88,"voltage":null,"consumption":24.52,"temperature":37}'
abr 29 00:34:25 raspberrypi npm[416]: zigbee2mqtt:info  2020-04-29 00:34:25: MQTT publish: topic 'zigbee2mqtt/Switch salon', payload '{"battery":100,"voltage":3042,"linkquality":47,"click":"triple"}'
abr 29 00:34:29 raspberrypi npm[416]: zigbee2mqtt:info  2020-04-29 00:34:29: MQTT publish: topic 'zigbee2mqtt/Switch salon', payload '{"battery":100,"voltage":3042,"linkquality":47,"click":"long"}'
abr 29 00:34:29 raspberrypi npm[416]: zigbee2mqtt:info  2020-04-29 00:34:29: MQTT publish: topic 'zigbee2mqtt/Switch salon', payload '{"battery":100,"voltage":3042,"linkquality":44,"click":"long_release","duration":142}'
abr 29 00:34:29 raspberrypi npm[416]: zigbee2mqtt:info  2020-04-29 00:34:29: MQTT publish: topic 'zigbee2mqtt/Switch salon', payload '{"battery":100,"voltage":3042,"linkquality":47,"click":"long"}'
abr 29 00:34:30 raspberrypi npm[416]: zigbee2mqtt:info  2020-04-29 00:34:30: MQTT publish: topic 'zigbee2mqtt/Switch salon', payload '{"battery":100,"voltage":3042,"linkquality":47,"click":"long"}'
``

Zigbee2mqtt version:1.12.2-dev
Adapter hardware: CC2531
Adapter firmware version: DEFAULT_20190608

Koenkk, 2020-04-29:
Please check https://www.zigbee2mqtt.io/devices/WXKG01LM.html#device-type-specific-configuration

kitopopo, 2020-04-29:
Ok, thanks. I check this in the next hours and if it is the problem i will close the issue. Thanks very much. Best regards

kitopopo, 2020-04-30:
Dear @Koenkk ,

I have added in the configuration.yaml file `"long_timeout: 2000"` under my sensor but any result, same problem appear, i am not shure if the command is correctly added below my devices in configuration.yam (I am using domoticz)

```
my configuration.yaml.file

'0x00158d000418xxxx':
    friendly_name: Switch salon
    long_timeout: 2000
```
If the command is correctly added heare doesn't works. the problem persist.

Some minuts i am upgraded to 1.13 version, i will test the problem in the new version. i will inform you.

Best regards

kitopopo, 2020-05-01:
Hello dear @koenkk,

I have confirmed now again that adding the line you mentioned in configuration.yalml the problem persists. I try to make a long press and the long and short press occurs, I try to make 3 clicks and it produces a single click + double press, the problem persists. Any solution? I already told you that all the buttons until update 1.11 have always worked well for me, I think I started to notice the problem in update 1.12, but since I use betas versions I cannot assure you that dev branch started giving problems, in a raspberry pi 4 only with domoticz working, the system should not be overloaded. Thanks in advance. a cordial greeting

Koenkk, 2020-05-02:
With `long_timeout` 2000 you have to hold the button for 2 seconds to trigger a long. Can you check with what version it still works?

kitopopo, 2020-05-02:
Thanks for your answer @Koenkk ,

But the problem is not only in the long click or single click. 
For instance, i need dissable my alarm with three clicks, but the result is double click detected + single click. I would like to dissable my alarm but the result was  ( message of the temperature in home cinema speakers (double click) + lamp turned on (living room).
Yes i can check old versions . How i can downgrade to 1.11? Have you instructions?

Thank you very much.

Koenkk, 2020-05-02:
Detection of double/triple clicks is done by the button itself so I dont think downgrading will fix anything.

kitopopo, 2020-05-02:
Hi dear friend @Koenkk , 

Hello, I understand that the detection is done by the switch itself but since I installed zigbee2mqtt more than a year ago in my 3 switches, the clicks and the detection have never failed, there must be some other problem. I should not press 3 clicks and other actions occur such as the blinds being lowered or a message being heard through the speakers at night or lights are turned on. It is dangerous. each press should do its job as always, I do not know what has changed in the last month. Thank you very much in advance

kitopopo, 2020-05-04:
After testing the problem for two days, it seems that adding the 2 lines that @koenkk has commented, the problem has not occurred again only once, it seems that it has improved a lot. thank you

kitopopo, 2020-05-07:
Dear friend @koenkk, 

I apologize for bothering with the same thing but my xiaomi switches do not work well. I have been using them for a year and I have never had any problem, now for example I cannot deactivate the alarm with 3 presses. After making the 3 clicks zigbe2mqtt has detected the following:

```
may 07 23:56:03 raspberrypi npm[424]: zigbee2mqtt:info  2020-05-07 23:56:03: MQTT publish: topic 'zigbee2mqtt/0x00158d000418cd86', payload '{"battery":100,"voltage":3062,"linkquality":65,"click":"double"}'
may 07 23:56:03 raspberrypi npm[424]: zigbee2mqtt:info  2020-05-07 23:56:03: MQTT publish: topic 'zigbee2mqtt/0x00158d000418cd86', payload '{"battery":100,"voltage":3062,"linkquality":65,"click":"long_release","duration":63701}'
may 07 23:56:03 raspberrypi npm[424]: zigbee2mqtt:info  2020-05-07 23:56:03: MQTT publish: topic 'zigbee2mqtt/0x00158d000418cd86', payload '{"battery":100,"voltage":3062,"linkquality":65,"click":"single"}'
```
I have been turning the alarm on and off for a year and have never had these problems. I don't know what is happening but something happens. As you said I added the line to configuration.yaml but I don't get improvements.- 

```
my configuration.yaml.file

'0x00158d000418xxxx':
    friendly_name: Switch salon
    long_timeout: 2000
```

Every day the problem occurs, i can not use this switches, alarms fails, message in the speakers fails, Blinds go up when they shouldn't, lamp on/off fails. A disaster but if I restart the zigbe2mqtt service it works fine, works perfectly for for a time . please ,see all the log:

```
may 08 00:05:27 raspberrypi npm[9987]: zigbee2mqtt:info  2020-05-08 00:05:27: MQTT publish: topic 'zigbee2mqtt/0x00158d000418cd86', payload '{"battery":100,"voltage":3062,"linkquality":65,"click":"single"}'
may 08 00:05:30 raspberrypi npm[9987]: zigbee2mqtt:info  2020-05-08 00:05:30: MQTT publish: topic 'zigbee2mqtt/0x00158d000418cd86', payload '{"battery":100,"voltage":3062,"linkquality":63,"click":"single"}'
may 08 00:05:32 raspberrypi npm[9987]: zigbee2mqtt:info  2020-05-08 00:05:32: MQTT publish: topic 'zigbee2mqtt/0x00158d000418cd86', payload '{"battery":100,"voltage":3062,"linkquality":63,"click":"triple"}'
may 08 00:05:35 raspberrypi npm[9987]: zigbee2mqtt:info  2020-05-08 00:05:35: MQTT publish: topic 'zigbee2mqtt/0x00158d000418cd86', payload '{"battery":100,"voltage":3062,"linkquality":63,"click":"triple"}'
may 08 00:05:37 raspberrypi npm[9987]: zigbee2mqtt:info  2020-05-08 00:05:37: MQTT publish: topic 'zigbee2mqtt/0x00158d000418cd86', payload '{"battery":100,"voltage":3062,"linkquality":63,"click":"triple"}'
may 08 00:05:39 raspberrypi npm[9987]: zigbee2mqtt:info  2020-05-08 00:05:39: MQTT publish: topic 'zigbee2mqtt/0x00158d000418cd86', payload '{"battery":100,"voltage":3062,"linkquality":63,"click":"triple"}'
may 08 00:05:41 raspberrypi npm[9987]: zigbee2mqtt:info  2020-05-08 00:05:41: MQTT publish: topic 'zigbee2mqtt/0x00158d000418cd86', payload '{"battery":100,"voltage":3062,"linkquality":63,"click":"triple"}'
may 08 00:05:42 raspberrypi npm[9987]: zigbee2mqtt:info  2020-05-08 00:05:42: MQTT publish: topic 'zigbee2mqtt/Denon plug', payload '{"state":"ON","power":7.79,"voltage":null,"consumption":69.01,"temperature":35,"linkquality":107}'
may 08 00:05:43 raspberrypi npm[9987]: zigbee2mqtt:info  2020-05-08 00:05:43: MQTT publish: topic 'zigbee2mqtt/0x00158d000418cd86', payload '{"battery":100,"voltage":3062,"linkquality":63,"click":"double"}'
may 08 00:05:46 raspberrypi npm[9987]: zigbee2mqtt:info  2020-05-08 00:05:46: MQTT publish: topic 'zigbee2mqtt/0x00158d000418cd86', payload '{"battery":100,"voltage":3062,"linkquality":63,"click":"triple"}'
may 08 00:05:47 raspberrypi npm[9987]: zigbee2mqtt:info  2020-05-08 00:05:47: MQTT publish: topic 'zigbee2mqtt/Denon plug', payload '{"state":"ON","power":33.42,"voltage":null,"consumption":69.01,"temperature":35,"linkquality":105}'
may 08 00:05:47 raspberrypi npm[9987]: zigbee2mqtt:info  2020-05-08 00:05:47: MQTT publish: topic 'zigbee2mqtt/0x00158d000418cd86', payload '{"battery":100,"voltage":3062,"linkquality":63,"click":"single"}'
may 08 00:05:49 raspberrypi npm[9987]: zigbee2mqtt:info  2020-05-08 00:05:49: MQTT publish: topic 'zigbee2mqtt/Denon plug', payload '{"state":"ON","power":65.52,"voltage":null,"consumption":69.01,"temperature":35,"linkquality":107}'
may 08 00:05:49 raspberrypi npm[9987]: zigbee2mqtt:info  2020-05-08 00:05:49: MQTT publish: topic 'zigbee2mqtt/0x00158d000418cd86', payload '{"battery":100,"voltage":3062,"linkquality":65,"click":"single"}'
may 08 00:05:51 raspberrypi npm[9987]: zigbee2mqtt:info  2020-05-08 00:05:51: MQTT publish: topic 'zigbee2mqtt/0x00158d000418cd86', payload '{"battery":100,"voltage":3062,"linkquality":63,"click":"triple"}'
may 08 00:05:53 raspberrypi npm[9987]: zigbee2mqtt:info  2020-05-08 00:05:53: MQTT publish: topic 'zigbee2mqtt/0x00158d000418cd86', payload '{"battery":100,"voltage":3062,"linkquality":63,"click":"double"}'
may 08 00:05:55 raspberrypi npm[9987]: zigbee2mqtt:info  2020-05-08 00:05:55: MQTT publish: topic 'zigbee2mqtt/0x00158d000418cd86', payload '{"battery":100,"voltage":3062,"linkquality":63,"click":"triple"}'
may 08 00:05:57 raspberrypi npm[9987]: zigbee2mqtt:info  2020-05-08 00:05:57: MQTT publish: topic 'zigbee2mqtt/0x00158d000418cd86', payload '{"battery":100,"voltage":3062,"linkquality":63,"click":"double"}'
[... cut ...]
may 08 00:06:47 raspberrypi npm[9987]: zigbee2mqtt:info  2020-05-08 00:06:47: MQTT publish: topic 'zigbee2mqtt/0x00158d000418cd86', payload '{"battery":100,"voltage":3062,"linkquality":63,"click":"double"}'
may 08 00:06:48 raspberrypi npm[9987]: zigbee2mqtt:info  2020-05-08 00:06:48: MQTT publish: topic 'zigbee2mqtt/0x00158d000418cd86', payload '{"battery":100,"voltage":3062,"linkquality":63,"click":"single"}'

```
all detected clicks correspond to the number of pulses emitted, none has failed. and they are fast pulsations in less than a second

Can you investigate please? seems something no works correctly in zigbe2mqtt service. thanks in advance. Best regards

Koenkk, 2020-05-08:
I think this is a network issue.

- Make sure to check https://www.zigbee2mqtt.io/how_tos/how_to_improve_network_range.html
- How much devices do you have?

kitopopo, 2020-05-09:
Dear @koenkk,

I have an big antenna in the cordinator, anyway, I have several xiaomi samrt plug distributed around tmy home that are signal repeaters. In the zigbee2mqtt device map I have an excellent signal in my 4 xiaomi switches., In my network I have a whole of 28 zigbee devices. It is not a signal problem, as I have already told you if I restart the zigbe2mqtt service using "systemctl restart zigbee2mqtt" my devices work perfectly for a few hours. a cordial greeting and thanks again

Koenkk, 2020-05-09:
- In case single/double/tripple/many click is detected wrong -> Issue of device itself (as this passes the amount of clicks, can for sure not be fixed from zigbee2mqtt side.
- In case long/long_release is detected incorrectly -> it takes longer than the `long_timeout` before the messages arrive at zigbee2mqtt. Can be solved be either increasing the `long_timeout` or improving the network (so that messages arrive faster).

kitopopo, 2020-05-09:
But let's see, why do you say it is a problem with the device? I have 4 devices and all 4 fail. Did all 4 break down? And what do you tell me about restarting the zigbe2mqtt service to solve the problem 
temporarily? works well for 1-2 hours?

Koenkk, 2020-05-11:
That would indicate that the problem is in the adapter (restarting zigbee2mqtt restarts the adapter). If the adapter is not strong enough to keep up with all the messages problems like this can arise. Upgrading to a more powerful adapter would make sure this is not the problem (https://www.zigbee2mqtt.io/information/supported_adapters.html#texas-instruments-cc26x2r1)

kitopopo, 2020-05-13:
Hello dear @Koenkk , I will buy a CC26X2R1. Does it currently work as well as the cc2531 or is it having problems? the xiaomi smart plugs continue working as repeaters with the new coordinator?

Koenkk, 2020-05-13:
Yes it will work much better. Keep an eye on the stock of https://www.tindie.com/products/electrolama/zzh-cc2652r-multiprotocol-rf-stick/ it's the best option atm.

kitopopo, 2020-05-13:
But cc26x2r1 and cc2652r are differents or are the same? Is the cc2652r another model? What is the difference? cc2652r is like to cc2531

Koenkk, 2020-05-13:
Its the same

kitopopo, 2020-05-13:
Ok. I will buy you adapter. But now is sold out. Do i need a debugger for windows? The debugger is an purchase option but i don't found the link. Thanks.

Koenkk, 2020-05-14:
You don't need the debugger (can be flashed without)

kitopopo, 2020-05-14:
Nice device. I will test. Thanks koenkk. Best regards

kitopopo, 2020-05-14:
A last question , the zzh cc2652r works correctly with domoticz?

Koenkk, 2020-05-14:
If using it via zigbee2mqtt, there is no difference of using CC2531 or CC2652R

kitopopo, 2020-05-15:
Hello dear @koenkk, sorry for continuing to insist on the same problem, but I am quite angry, the neighbors have called my attention for trusting zigbe2mqtt. Yesterday at 4 am when trying to turn off (1 click) the light with a short press, a script was fired (quadruple click) that opened all the blinds at home and the news was set at a quite high volume in all speakers of my home.

Does everyone with these switches have this problem? or is it just me? can someone comment something else?

I understand that the easiest solution is to comment that it is necessary to expand my cc2531 to a zzh cc2652 .  I have always had 28 zigbee devices and they have all worked properly for a year, my four WXKG01LM has always worked well since zigbee's inception, something has happened, and it may be due to a dev version update (that was when I start having the problem).  
It is the only thing that I have changed. Update to the last zigbe2mqtt brand dev. I have changed all the batteries in my 4 sensors but the problem persists over and over again

As I have already mentioned in the lastt updates I only use a 4gb raspberry 4 with domoticz. My system is not saturated I only use it for domoticz, I think it has a lot of power to do this function. It has always been commented in zigbe2mqtt.io and the github forum  that the cc2531 supported many devices using repeaters, many more than 28. 

with all my respect i think you have not paid much attention to this problem and i do not know the reason why you are not investigating what the problem is. You have not requested to sniff. I do not understand why it is based on the devices problem or cc2531 problem .

The truth is that I notice that the cc2531 is not very powerful , but to use a xiaomi switch it should work without problems as it has worked the last year. 

I will not buy the new zzh 2652 because in the future the same will happen again if you do not give me support now for my problem (I do not understand why), If I buy the new zzh it will be because I want to improve my system but not because a switch fails, it is not logical, have always worked well

I will continue investigating what happens. burn the firmware again and I will start from 0 and I will return to old versions of dev brach, if I find out something I will comment it here.
If I can not solve my problem, I will return to my old xiaomi gateway that does not give any problem with the xiaomi switch , the truth that I am quite disappointed.

Could you check the latest dev branch to see if there have been any changes?

thanks for your help. Best regards

Koenkk, 2020-05-15:
We could indeed do a final check to make sure where the issue comes from.

If you could provide the debug logging at the moment the issue happens I can check this. Also please annotate the amount of clicks you would expect.

To enable debug logging set in `configuration.yaml`:
```yaml
advanced:
  log_level: debug
```

nug321, 2020-05-20:
Hi guys, I just took my button WXKG01LM from xiaomi gate and paired it to the zigbee2mqtt to do some test. Started with fresh battery in the switch and coming in the range of coordinator.

Yep, it acts crazy. All those are supposed to be 'click':

`May 20 09:41:55 rpi npm[4494]: zigbee2mqtt:debug 2020-05-20 09:41:55: Received Zigbee message from 'Button', type 'attributeReport', cluster 'genOnOff', data '{"onOff":0}' from endpoint 1 with groupID 0
May 20 09:41:55 rpi npm[4494]: zigbee2mqtt:debug 2020-05-20 09:41:55: Received Zigbee message from 'Button', type 'attributeReport', cluster 'genOnOff', data '{"onOff":1}' from endpoint 1 with groupID 0
May 20 09:41:55 rpi npm[4494]: zigbee2mqtt:info  2020-05-20 09:41:55: MQTT publish: topic 'zigbee2mqtt/Button', payload '{"battery":100,"voltage":3012,"linkquality":0,"click":"single"}'
May 20 09:41:56 rpi npm[4494]: zigbee2mqtt:debug 2020-05-20 09:41:56: Received Zigbee message from 'Button', type 'attributeReport', cluster 'genOnOff', data '{"onOff":0}' from endpoint 1 with groupID 0
May 20 09:41:56 rpi npm[4494]: zigbee2mqtt:debug 2020-05-20 09:41:56: Received Zigbee message from 'Button', type 'attributeReport', cluster 'genOnOff', data '{"onOff":1}' from endpoint 1 with groupID 0
May 20 09:41:56 rpi npm[4494]: zigbee2mqtt:info  2020-05-20 09:41:56: MQTT publish: topic 'zigbee2mqtt/Button', payload '{"battery":100,"voltage":3012,"linkquality":0,"click":"single"}'
May 20 09:41:58 rpi npm[4494]: zigbee2mqtt:debug 2020-05-20 09:41:58: Received Zigbee message from 'Button', type 'attributeReport', cluster 'genOnOff', data '{"onOff":1}' from endpoint 1 with groupID 0
May 20 09:41:58 rpi npm[4494]: zigbee2mqtt:debug 2020-05-20 09:41:58: Received Zigbee message from 'Button', type 'attributeReport', cluster 'genOnOff', data '{"onOff":0}' from endpoint 1 with groupID 0
May 20 09:41:59 rpi npm[4494]: zigbee2mqtt:debug 2020-05-20 09:41:59: Received Zigbee message from 'Button', type 'attributeReport', cluster 'genOnOff', data '{"onOff":0}' from endpoint 1 with groupID 0
May 20 09:42:00 rpi npm[4494]: zigbee2mqtt:debug 2020-05-20 09:42:00: Received Zigbee message from 'Button', type 'attributeReport', cluster 'genOnOff', data '{"onOff":1}' from endpoint 1 with groupID 0
May 20 09:42:00 rpi npm[4494]: zigbee2mqtt:info  2020-05-20 09:42:00: MQTT publish: topic 'zigbee2mqtt/Button', payload '{"battery":100,"voltage":3012,"linkquality":0,"click":"single"}'
May 20 09:42:01 rpi npm[4494]: zigbee2mqtt:debug 2020-05-20 09:42:01: Received Zigbee message from 'Button', type 'attributeReport', cluster 'genOnOff', data '{"onOff":1}' from endpoint 1 with groupID 0
May 20 09:42:01 rpi npm[4494]: zigbee2mqtt:debug 2020-05-20 09:42:01: Received Zigbee message from 'Button', type 'attributeReport', cluster 'genOnOff', data '{"onOff":0}' from endpoint 1 with groupID 0
May 20 09:42:03 rpi npm[4494]: zigbee2mqtt:debug 2020-05-20 09:42:03: Received Zigbee message from 'Button', type 'attributeReport', cluster 'genOnOff', data '{"onOff":1}' from endpoint 1 with groupID 0
May 20 09:42:03 rpi npm[4494]: zigbee2mqtt:info  2020-05-20 09:42:03: MQTT publish: topic 'zigbee2mqtt/Button', payload '{"battery":100,"voltage":3012,"linkquality":0,"click":"single"}'
May 20 09:42:03 rpi npm[4494]: zigbee2mqtt:debug 2020-05-20 09:42:03: Received Zigbee message from 'Button', type 'attributeReport', cluster 'genOnOff', data '{"onOff":0}' from endpoint 1 with groupID 0
May 20 09:42:03 rpi npm[4494]: zigbee2mqtt:info  2020-05-20 09:42:03: MQTT publish: topic 'zigbee2mqtt/Button', payload '{"battery":100,"voltage":3012,"linkquality":0,"click":"long"}'
May 20 09:42:06 rpi npm[4494]: zigbee2mqtt:debug 2020-05-20 09:42:06: Received Zigbee message from 'Button', type 'attributeReport', cluster 'genOnOff', data '{"onOff":1}' from endpoint 1 with groupID 0
May 20 09:42:06 rpi npm[4494]: zigbee2mqtt:info  2020-05-20 09:42:06: MQTT publish: topic 'zigbee2mqtt/Button', payload '{"battery":100,"voltage":3012,"linkquality":0,"click":"long_release","duration":3190}'
May 20 09:42:06 rpi npm[4494]: zigbee2mqtt:debug 2020-05-20 09:42:06: Received Zigbee message from 'Button', type 'attributeReport', cluster 'genOnOff', data '{"onOff":0}' from endpoint 1 with groupID 0
May 20 09:42:08 rpi npm[4494]: zigbee2mqtt:info  2020-05-20 09:42:08: MQTT publish: topic 'zigbee2mqtt/Button', payload '{"battery":100,"voltage":3012,"linkquality":0,"click":"long"}'
May 20 09:42:08 rpi npm[4494]: zigbee2mqtt:debug 2020-05-20 09:42:08: Received Zigbee message from 'Button', type 'attributeReport', cluster 'genOnOff', data '{"onOff":1}' from endpoint 1 with groupID 0
May 20 09:42:08 rpi npm[4494]: zigbee2mqtt:info  2020-05-20 09:42:08: MQTT publish: topic 'zigbee2mqtt/Button', payload '{"battery":100,"voltage":3012,"linkquality":0,"click":"long_release","duration":540}'
May 20 09:42:08 rpi npm[4494]: zigbee2mqtt:debug 2020-05-20 09:42:08: Received Zigbee message from 'Button', type 'attributeReport', cluster 'genOnOff', data '{"onOff":0}' from endpoint 1 with groupID 0
May 20 09:42:10 rpi npm[4494]: zigbee2mqtt:debug 2020-05-20 09:42:10: Received Zigbee message from 'Button', type 'attributeReport', cluster 'genOnOff', data '{"onOff":1}' from endpoint 1 with groupID 0
May 20 09:42:10 rpi npm[4494]: zigbee2mqtt:info  2020-05-20 09:42:10: MQTT publish: topic 'zigbee2mqtt/Button', payload '{"battery":100,"voltage":3012,"linkquality":0,"click":"single"}'
[... cut ...]
May 20 09:42:19 rpi npm[4494]: zigbee2mqtt:info  2020-05-20 09:42:19: MQTT publish: topic 'zigbee2mqtt/Button', payload '{"battery":100,"voltage":3012,"linkquality":0,"click":"long_release","duration":1682}'
May 20 09:42:19 rpi npm[4494]: zigbee2mqtt:info  2020-05-20 09:42:19: MQTT publish: topic 'zigbee2mqtt/Button', payload '{"battery":100,"voltage":3012,"linkquality":0,"click":"single"}'
May 20 09:42:22 rpi npm[4494]: zigbee2mqtt:info  2020-05-20 09:42:22: MQTT publish: topic 'zigbee2mqtt/Button', payload '{"battery":100,"voltage":3012,"linkquality":0,"click":"long"}'`

zigbee2mqtt version 1.13.0 (commit #55e6283)
All elements: 18 (including coordinator + 3 routers)

ArkadiuszNiemiec, 2020-05-21:
Some of my switches starting acting like @nug321 just today. I believe that this started to happen since I have switched the firmware from `default` to `source_routing` and added Xiaomi Wall Switch (working as a router).
I cannot verify the routing of switches due to lack of routes in the scan but the good one reports LQI of 31 and the bad one has lqi around 94.

kitopopo, 2020-05-22:
Hello dear @Koenkk,

Some days ago I updated to the latest version of raspbian buster, I saw that the latest version of npm was also updated, then I also updated to the latest development branch of zigbe2mqtt (I don't know if any new development branch has come out) days but I did update to anyway).
As I performed all of these steps, my xiaomi switches are now working properly and error-free. What a mistery
Anyway, when the new zzh stick is available, I will buy it as it is much more powerful and everything will work better. I hope everything works more very quickly. Thanks for your help. a cordial greeting

kitopopo, 2020-06-22:
> If using it via zigbee2mqtt, there is no difference of using CC2531 or CC2652R

dear @Koenkk ,

Some days ago there was stock again zzh cc2652  but now you are going to make the purchase in tindie shop and again it is out of stock. I have several questions, 

If i buy the cc2652 Do you support cc2652? 

Why don't you add information on how to configure the new cc2652 on the zigbe2mqtt.io website same as with cc2531?

Best regards, thanks in advanced

Koenkk, 2020-06-24:
All devices listed here are supported: https://www.zigbee2mqtt.io/information/supported_adapters.html . If the zzh is out-of-stock you can go for the: Texas Instruments LAUNCHXL-CC26X2R1

kitopopo, 2020-06-24:
Thanks @Koenkk ,

Which one is the best, the zzh cc2652r or conbe II?

Best regards

ArkadiuszNiemiec, 2020-06-24:
@kitopopo Conbee is not yet supported, you cannot run it in a stable setup.

kitopopo, 2020-06-24:
Dear friend, 

In my domoticz telegram group the people are using deconz (conbe ii) whith deconz plugin fir domoticz, i don't know the differences but they explain that have interface web for manage all option, update devices, zigbee2mtt map, all using a web app.
They explain good support por xiaomi switches, is possible use 7 o 8 differents clicks. I don' know, i am studying this option .

ArkadiuszNiemiec, 2020-06-27:
I have just received my zzh stick (bought it just to fix this issue but it did not help). Here are full logs of one of the clicks:
```
2020-06-27T10:49:37.832Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,27,68,129,0,0,6,0,253,54,1,1,0,27,0,74,38,41,0,0,7,24,32,10,0,0,16,0,97,171,28,190]
2020-06-27T10:49:37.834Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,27,68,129,0,0,6,0,253,54,1,1,0,27,0,74,38,41,0,0,7,24,32,10,0,0,16,0,97,171,28,190]
2020-06-27T10:49:37.834Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 27 - 2 - 4 - 129 - [0,0,6,0,253,54,1,1,0,27,0,74,38,41,0,0,7,24,32,10,0,0,16,0,97,171,28] - 190
2020-06-27T10:49:37.835Z zigbee-herdsman:adapter:zStack:znp:AREQ <-- AF - incomingMsg - {"groupid":0,"clusterid":6,"srcaddr":14077,"srcendpoint":1,"dstendpoint":1,"wasbroadcast":0,"linkquality":27,"securityuse":0,"timestamp":2696778,"transseqnumber":0,"len":7,"data":{"type":"Buffer","data":[24,32,10,0,0,16,0]}}
zigbee2mqtt:debug 2020-06-27 12:49:37: Received Zigbee message from '0x00158d0003e725c3', type 'attributeReport', cluster 'genOnOff', data '{"onOff":0}' from endpoint 1 with groupID 0
2020-06-27T10:49:37.836Z zigbee-herdsman:controller:log Received 'zcl' data '{"frame":{"Header":{"frameControl":{"frameType":0,"manufacturerSpecific":false,"direction":1,"disableDefaultResponse":true,"reservedBits":0},"transactionSequenceNumber":32,"manufacturerCode":null,"commandIdentifier":10},"Payload":[{"attrId":0,"dataType":16,"attrData":0}]},"address":14077,"endpoint":1,"linkquality":27,"groupID":0}'
2020-06-27T10:49:37.839Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
2020-06-27T10:49:37.908Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,27,68,129,0,0,6,0,253,54,1,1,0,27,0,232,56,41,0,0,7,24,32,10,0,0,16,0,97,171,28,2]
2020-06-27T10:49:37.910Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,27,68,129,0,0,6,0,253,54,1,1,0,27,0,232,56,41,0,0,7,24,32,10,0,0,16,0,97,171,28,2]
2020-06-27T10:49:37.911Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 27 - 2 - 4 - 129 - [0,0,6,0,253,54,1,1,0,27,0,232,56,41,0,0,7,24,32,10,0,0,16,0,97,171,28] - 2
2020-06-27T10:49:37.912Z zigbee-herdsman:adapter:zStack:znp:AREQ <-- AF - incomingMsg - {"groupid":0,"clusterid":6,"srcaddr":14077,"srcendpoint":1,"dstendpoint":1,"wasbroadcast":0,"linkquality":27,"securityuse":0,"timestamp":2701544,"transseqnumber":0,"len":7,"data":{"type":"Buffer","data":[24,32,10,0,0,16,0]}}
2020-06-27T10:49:37.915Z zigbee-herdsman:controller:log Received 'zcl' data '{"frame":{"Header":{"frameControl":{"frameType":0,"manufacturerSpecific":false,"direction":1,"disableDefaultResponse":true,"reservedBits":0},"transactionSequenceNumber":32,"manufacturerCode":null,"commandIdentifier":10},"Payload":[{"attrId":0,"dataType":16,"attrData":0}]},"address":14077,"endpoint":1,"linkquality":27,"groupID":0}'
zigbee2mqtt:debug 2020-06-27 12:49:37: Received Zigbee message from '0x00158d0003e725c3', type 'attributeReport', cluster 'genOnOff', data '{"onOff":0}' from endpoint 1 with groupID 0
2020-06-27T10:49:37.918Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
2020-06-27T10:49:37.969Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,27,68,129,0,0,6,0,253,54,1,1,0,24,0,40,72,41,0,0,7,24,32,10,0,0,16,0,97,171,28,177]
2020-06-27T10:49:37.970Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,27,68,129,0,0,6,0,253,54,1,1,0,24,0,40,72,41,0,0,7,24,32,10,0,0,16,0,97,171,28,177]
2020-06-27T10:49:37.970Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 27 - 2 - 4 - 129 - [0,0,6,0,253,54,1,1,0,24,0,40,72,41,0,0,7,24,32,10,0,0,16,0,97,171,28] - 177
2020-06-27T10:49:37.971Z zigbee-herdsman:adapter:zStack:znp:AREQ <-- AF - incomingMsg - {"groupid":0,"clusterid":6,"srcaddr":14077,"srcendpoint":1,"dstendpoint":1,"wasbroadcast":0,"linkquality":24,"securityuse":0,"timestamp":2705448,"transseqnumber":0,"len":7,"data":{"type":"Buffer","data":[24,32,10,0,0,16,0]}}
2020-06-27T10:49:37.973Z zigbee-herdsman:controller:log Received 'zcl' data '{"frame":{"Header":{"frameControl":{"frameType":0,"manufacturerSpecific":false,"direction":1,"disableDefaultResponse":true,"reservedBits":0},"transactionSequenceNumber":32,"manufacturerCode":null,"commandIdentifier":10},"Payload":[{"attrId":0,"dataType":16,"attrData":0}]},"address":14077,"endpoint":1,"linkquality":24,"groupID":0}'
zigbee2mqtt:debug 2020-06-27 12:49:37: Received Zigbee message from '0x00158d0003e725c3', type 'attributeReport', cluster 'genOnOff', data '{"onOff":0}' from endpoint 1 with groupID 0
2020-06-27T10:49:37.977Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
2020-06-27T10:49:38.098Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,27,68,129,0,0,6,0,253,54,1,1,0,24,0,137,103,41,0,0,7,24,32,10,0,0,16,0,97,171,28,63]
2020-06-27T10:49:38.098Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,27,68,129,0,0,6,0,253,54,1,1,0,24,0,137,103,41,0,0,7,24,32,10,0,0,16,0,97,171,28,63]
2020-06-27T10:49:38.099Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 27 - 2 - 4 - 129 - [0,0,6,0,253,54,1,1,0,24,0,137,103,41,0,0,7,24,32,10,0,0,16,0,97,171,28] - 63
2020-06-27T10:49:38.099Z zigbee-herdsman:adapter:zStack:znp:AREQ <-- AF - incomingMsg - {"groupid":0,"clusterid":6,"srcaddr":14077,"srcendpoint":1,"dstendpoint":1,"wasbroadcast":0,"linkquality":24,"securityuse":0,"timestamp":2713481,"transseqnumber":0,"len":7,"data":{"type":"Buffer","data":[24,32,10,0,0,16,0]}}
[... cut ...]
zigbee2mqtt:info  2020-06-27 12:49:39: MQTT publish: topic 'zigbee2mqtt/0x00158d0003e725c3', payload '{"battery":100,"voltage":3062,"linkquality":24,"click":"long_release","duration":274}'
2020-06-27T10:49:39.553Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
zigbee2mqtt:info  2020-06-27 12:49:39: MQTT publish: topic 'zigbee2mqtt/0x00158d0003e725c3', payload '{"battery":100,"voltage":3062,"linkquality":24,"click":""}'
zigbee2mqtt:info  2020-06-27 12:49:39: MQTT publish: topic 'zigbee2mqtt/0x00158d0003e725c3/click', payload 'long_release'
```

ArkadiuszNiemiec, 2020-06-27:
I have tried a quick fix and it looks like it's working: https://github.com/Koenkk/zigbee-herdsman-converters/pull/1356

I clear the timer when click down is received, so all but last one are ignored. This is what happened with the fix:
```
zigbee2mqtt:debug 2020-06-27 13:37:41: Received Zigbee message from '0x00158d0003e725c3', type 'attributeReport', cluster 'genOnOff', data '{"onOff":0}' from endpoint 1 with groupID 0
zigbee2mqtt:debug 2020-06-27 13:37:41: Received Zigbee message from '0x00158d0003e725c3', type 'attributeReport', cluster 'genOnOff', data '{"onOff":0}' from endpoint 1 with groupID 0
zigbee2mqtt:debug 2020-06-27 13:37:41: Received Zigbee message from '0x00158d0003e725c3', type 'attributeReport', cluster 'genOnOff', data '{"onOff":0}' from endpoint 1 with groupID 0
zigbee2mqtt:debug 2020-06-27 13:37:41: Received Zigbee message from '0x00158d0003e725c3', type 'attributeReport', cluster 'genOnOff', data '{"onOff":0}' from endpoint 1 with groupID 0
zigbee2mqtt:debug 2020-06-27 13:37:41: Received Zigbee message from '0x00158d0003e725c3', type 'attributeReport', cluster 'genOnOff', data '{"onOff":1}' from endpoint 1 with groupID 0
zigbee2mqtt:info  2020-06-27 13:37:41: MQTT publish: topic 'zigbee2mqtt/0x00158d0003e725c3', payload '{"battery":100,"voltage":3062,"linkquality":84,"click":"single"}'
zigbee2mqtt:info  2020-06-27 13:37:41: MQTT publish: topic 'zigbee2mqtt/0x00158d0003e725c3', payload '{"battery":100,"voltage":3062,"linkquality":84,"click":""}'
zigbee2mqtt:info  2020-06-27 13:37:41: MQTT publish: topic 'zigbee2mqtt/0x00158d0003e725c3/click', payload 'single'
```

I believe that there is still an issue when the `click down` comes before `click up` but I will investigate.

kitopopo, 2020-06-27:
I have problems every day with the xiaomi switch , this problem should be fixed. Some months ago the switches have worked correctly.

ArkadiuszNiemiec, 2020-06-27:
Now I have a problem with multiple double presses:
```
zigbee2mqtt:debug 2020-06-27 16:45:33: Received Zigbee message from 'office_desk_switch', type 'attributeReport', cluster 'genOnOff', data '{"32768":2}' from endpoint 1 with groupID 0
zigbee2mqtt:info  2020-06-27 16:45:33: MQTT publish: topic 'zigbee2mqtt/office_desk_switch', payload '{"battery":100,"voltage":3032,"linkquality":15,"click":"double"}'
zigbee2mqtt:info  2020-06-27 16:45:33: MQTT publish: topic 'zigbee2mqtt/office_desk_switch', payload '{"battery":100,"voltage":3032,"linkquality":15,"click":""}'
zigbee2mqtt:info  2020-06-27 16:45:33: MQTT publish: topic 'zigbee2mqtt/office_desk_switch/click', payload 'double'
zigbee2mqtt:debug 2020-06-27 16:45:33: Received Zigbee message from 'office_desk_switch', type 'attributeReport', cluster 'genOnOff', data '{"32768":2}' from endpoint 1 with groupID 0
zigbee2mqtt:info  2020-06-27 16:45:33: MQTT publish: topic 'zigbee2mqtt/office_desk_switch', payload '{"battery":100,"voltage":3032,"linkquality":15,"click":"double"}'
zigbee2mqtt:info  2020-06-27 16:45:33: MQTT publish: topic 'zigbee2mqtt/office_desk_switch', payload '{"battery":100,"voltage":3032,"linkquality":15,"click":""}'
zigbee2mqtt:info  2020-06-27 16:45:33: MQTT publish: topic 'zigbee2mqtt/office_desk_switch/click', payload 'double'
zigbee2mqtt:debug 2020-06-27 16:45:33: Received Zigbee message from 'office_desk_switch', type 'attributeReport', cluster 'genOnOff', data '{"32768":2}' from endpoint 1 with groupID 0
zigbee2mqtt:info  2020-06-27 16:45:33: MQTT publish: topic 'zigbee2mqtt/office_desk_switch', payload '{"battery":100,"voltage":3032,"linkquality":18,"click":"double"}'
zigbee2mqtt:info  2020-06-27 16:45:33: MQTT publish: topic 'zigbee2mqtt/office_desk_switch', payload '{"battery":100,"voltage":3032,"linkquality":18,"click":""}'
zigbee2mqtt:info  2020-06-27 16:45:33: MQTT publish: topic 'zigbee2mqtt/office_desk_switch/click', payload 'double'
zigbee2mqtt:debug 2020-06-27 16:45:33: Received Zigbee message from 'office_desk_switch', type 'attributeReport', cluster 'genOnOff', data '{"32768":2}' from endpoint 1 with groupID 0
zigbee2mqtt:info  2020-06-27 16:45:33: MQTT publish: topic 'zigbee2mqtt/office_desk_switch', payload '{"battery":100,"voltage":3032,"linkquality":15,"click":"double"}'
zigbee2mqtt:info  2020-06-27 16:45:33: MQTT publish: topic 'zigbee2mqtt/office_desk_switch', payload '{"battery":100,"voltage":3032,"linkquality":15,"click":""}'
zigbee2mqtt:info  2020-06-27 16:45:33: MQTT publish: topic 'zigbee2mqtt/office_desk_switch/click', payload 'double'
zigbee2mqtt:debug 2020-06-27 16:45:34: Received Zigbee message from 'office_desk_switch', type 'attributeReport', cluster 'genOnOff', data '{"32768":2}' from endpoint 1 with groupID 0
zigbee2mqtt:info  2020-06-27 16:45:34: MQTT publish: topic 'zigbee2mqtt/office_desk_switch', payload '{"battery":100,"voltage":3032,"linkquality":18,"click":"double"}'
zigbee2mqtt:info  2020-06-27 16:45:34: MQTT publish: topic 'zigbee2mqtt/office_desk_switch', payload '{"battery":100,"voltage":3032,"linkquality":18,"click":""}'
zigbee2mqtt:info  2020-06-27 16:45:34: MQTT publish: topic 'zigbee2mqtt/office_desk_switch/click', payload 'double'
zigbee2mqtt:debug 2020-06-27 16:45:34: Received Zigbee message from 'office_desk_switch', type 'attributeReport', cluster 'genOnOff', data '{"32768":2}' from endpoint 1 with groupID 0
zigbee2mqtt:info  2020-06-27 16:45:34: MQTT publish: topic 'zigbee2mqtt/office_desk_switch', payload '{"battery":100,"voltage":3032,"linkquality":15,"click":"double"}'
zigbee2mqtt:info  2020-06-27 16:45:34: MQTT publish: topic 'zigbee2mqtt/office_desk_switch', payload '{"battery":100,"voltage":3032,"linkquality":15,"click":""}'
zigbee2mqtt:info  2020-06-27 16:45:34: MQTT publish: topic 'zigbee2mqtt/office_desk_switch/click', payload 'double'
```
@Koenkk can I fix it in `zigbee-herdsman-converters` using timers and parameter or should it all be fixed on the coordinator side?

kitopopo, 2020-06-27:
Maybe @Koenkk can solve the problem. Thanks in advanced. Best regards

Koenkk, 2020-06-28:
@ArkadiuszNiemiec if I understand correctly, a single double click on the device results in multiple `Received Zigbee message from 'office_desk_switch'` messages? If so, could you sniff the traffic here? (maybe it reveals something about the root cause of this). https://www.zigbee2mqtt.io/how_tos/how_to_sniff_zigbee_traffic.html

ArkadiuszNiemiec, 2020-06-28:
@Koenkk yes, and the single click down results with multiple `{"onOff":0}` and one `{"onOff":1}`. I have only observed it when the switch is far from the coordinator. I will sniff it and update here.

ArkadiuszNiemiec, 2020-06-28:
@Koenkk I have registered a bad click with filter set to its address (`zbee_nwk.addr==0x36FD`):

[image]

Here is the file: https://send.firefox.com/download/1f69f9cd4a80ce47/#YX886jR3cZ1thSp2VZtsHQ
Here is the file of two signle clicks that worked correctly: https://send.firefox.com/download/a2525b89ba4f7f18/#lf-4_7iBNZCFGSUcHuOInA

In short, it shows same thing as the debug logs of Z2M. Can you see more in these packets?
By the way: I just cover the switch with my body to get these multiplied packets.

Koenkk, 2020-06-29:
Can you provide me your network key? (otherwise I cannot decrypt the sniff).

But first try if this fixes the issue. In `fromZigbee.js` change the `WXKG01LM_click` converter to:

```js
WXKG01LM_click: {
    cluster: 'genOnOff',
    type: ['attributeReport', 'readResponse'],
    convert: (model, msg, publish, options, meta) => {
        const deviceID = msg.device.ieeeAddr;
        const state = msg.data['onOff'];

        if (!store[deviceID]) {
            store[deviceID] = {};
        }

        const current = msg.meta.zclTransactionSequenceNumber;
        if (store[msg.device.ieeeAddr].transaction === current) return;
        store[msg.device.ieeeAddr].transaction = current;

        // 0 = click down, 1 = click up, else = multiple clicks
        if (state === 0) {
            store[deviceID].timer = setTimeout(() => {
                publish({click: 'long'});
                store[deviceID].timer = null;
                store[deviceID].long = Date.now();
            }, options.long_timeout || 1000); // After 1000 milliseconds of not releasing we assume long click.
        } else if (state === 1) {
            if (store[deviceID].long) {
                const duration = Date.now() - store[deviceID].long;
                publish({click: 'long_release', duration: duration});
                store[deviceID].long = false;
            }

            if (store[deviceID].timer) {
                clearTimeout(store[deviceID].timer);
                store[deviceID].timer = null;
                publish({click: 'single'});
            }
        } else {
            const clicks = msg.data['32768'];
            const payload = clickLookup[clicks] ? clickLookup[clicks] : 'many';
            publish({click: payload});
        }
    },
},
```

Does that fix the issue?

ArkadiuszNiemiec, 2020-06-29:
@Koenkk It's the smart way to fix it 😄 and it works:
```
Received Zigbee message from '0x00158d0003e725c3', type 'attributeReport', cluster 'genOnOff', data '{"onOff":0}' from endpoint 1 with groupID 0
Received Zigbee message from '0x00158d0003e725c3', type 'attributeReport', cluster 'genOnOff', data '{"onOff":0}' from endpoint 1 with groupID 0
message ignored!
Received Zigbee message from '0x00158d0003e725c3', type 'attributeReport', cluster 'genOnOff', data '{"onOff":0}' from endpoint 1 with groupID 0
message ignored!
Received Zigbee message from '0x00158d0003e725c3', type 'attributeReport', cluster 'genOnOff', data '{"onOff":0}' from endpoint 1 with groupID 0
message ignored!
Received Zigbee message from '0x00158d0003e725c3', type 'attributeReport', cluster 'genOnOff', data '{"onOff":0}' from endpoint 1 with groupID 0
message ignored!
Received Zigbee message from '0x00158d0003e725c3', type 'attributeReport', cluster 'genOnOff', data '{"onOff":0}' from endpoint 1 with groupID 0
message ignored!
Received Zigbee message from '0x00158d0003e725c3', type 'attributeReport', cluster 'genOnOff', data '{"onOff":1}' from endpoint 1 with groupID 0
zigbee2mqtt:info  2020-06-29 MQTT publish: topic 'zigbee2mqtt/0x00158d0003e725c3', payload '{"battery":100,"voltage":3042,"linkquality":15,"click":"single"}'
zigbee2mqtt:info  2020-06-29 MQTT publish: topic 'zigbee2mqtt/0x00158d0003e725c3', payload '{"battery":100,"voltage":3042,"linkquality":15,"click":""}'
zigbee2mqtt:info  2020-06-29 MQTT publish: topic 'zigbee2mqtt/0x00158d0003e725c3/click', payload 'single'
Received Zigbee message from '0x00158d0003e725c3', type 'attributeReport', cluster 'genOnOff', data '{"onOff":1}' from endpoint 1 with groupID 0
message ignored!
Received Zigbee message from '0x00158d0003e725c3', type 'attributeReport', cluster 'genOnOff', data '{"onOff":1}' from endpoint 1 with groupID 0
message ignored!
Received Zigbee message from '0x00158d0003e725c3', type 'attributeReport', cluster 'genOnOff', data '{"onOff":1}' from endpoint 1 with groupID 0
message ignored!
Received Zigbee message from '0x00158d0003e725c3', type 'attributeReport', cluster 'genOnOff', data '{"onOff":1}' from endpoint 1 with groupID 0
message ignored!
Received Zigbee message from '0x00158d0003e725c3', type 'attributeReport', cluster 'genOnOff', data '{"onOff":1}' from endpoint 1 with groupID 0
message ignored!
```

I will remove the log message from the code and you can merge it if you want. Thanks for the help.
https://github.com/Koenkk/zigbee-herdsman-converters/pull/1356

Koenkk, 2020-06-29:
@ArkadiuszNiemiec great! 

Changes will be available in the latest dev branch in a few hours (https://www.zigbee2mqtt.io/how_tos/how-to-switch-to-dev-branch.html)

closed 2020-06-29 by Koenkk

kitopopo, 2020-06-29:
> Can you provide me your network key? (otherwise I cannot decrypt the sniff).
>
> But first try if this fixes the issue. In `fromZigbee.js` change the `WXKG01LM_click` converter to:
>
> ```js [...]

Dear @Koenkk ,

This fix works also for the cc2531? I wait news from you. Thanks

Koenkk, 2020-06-29:
@kitopopo yes, fix is available in the dev branch now.

kitopopo, 2020-06-29:
Ok @Koenkk . I will update at the new dev branch this night. As I told you more than 1 month ago there was a problem between the buttons and zigbee2mqtt. thank you very much for solving it.

Best regards.

kitopopo, 2020-06-29:
Dear koenk, I am sorry to tell you this but the problem persists, zigbee2mqtt continues to detect double clicks if I make single clicks, I am already tired of the blinds being pulled up late at night, I think that if you do not solve the problem my next device will not be zzh cc2652r. I'm going to try ConBee II, I find it horrible that a button fails for so long with a raspberry pi 4 and an ssd. Thank you for your collaboration during all this time. You have been very kind

ArkadiuszNiemiec, 2020-06-29:
If you get double click instead of single click then it's a problem with button itself. Xiaomi switches reports presses down, up, double and triple click, if it reports a double click than its not a CC2531 or z2m problem. The problem that was solved here is that the button was reporting multiple double or single clicks instead of just one.

kitopopo, 2020-06-30:
Hello, thanks for your answer, I have 4 xiaomi switch buttons and all 4 fail with the same problem, it cannot be a button problem. There has come a time when not a single pulse worked. I have been more than 1 year with zigbe2mqtt and these buttons and they have never failed until I opened this issue approximately. I will not insist any more, thank you. Some time ago, restarting the zigbe2mqtt service solved the problem for a few hours, now it has gotten worse. Best regards.

ArkadiuszNiemiec, 2020-07-03:
@kitopopo these switches are terrible, I had a problem with one today, it was not responding for a minute and then it sent multiple long, triple and single messages (I was just single clicking it). I will never buy xiaomi switches again, next buy is Konke switch: https://www.zigbee2mqtt.io/devices/2AJZ4KPKEY.html
They are sold on Aliexpress.

kitopopo, 2020-07-03:
> @kitopopo these switches are terrible, I had a problem with one today, it was not responding for a minute and then it sent multiple long, triple and single messages (I was just single clicking it). [...]

Dear friend,

Thanks for the suggestion. But during 1 year aproximately my xiaomi switches have worked correctly. Maybe an wrong update of dev branch , i don't know the reasson exactly but they worked correctly during a year. If my switches don't have worked correctly during so long i would understand a hardware switch problem

Best Regards

ArkadiuszNiemiec, 2020-07-13:
These damned switches... I had the problem that switch reported the click up before click down. Again. I was able to log it:
```
zigbee2mqtt:debug 2020-07-14 01:14:34: Received Zigbee message from 'kitchen_switch', type 'attributeReport', cluster 'genOnOff', data '{"onOff":1}' from endpoint 1 with groupID 0
zigbee2mqtt:debug 2020-07-14 01:14:34: Received Zigbee message from 'kitchen_switch', type 'attributeReport', cluster 'genOnOff', data '{"onOff":0}' from endpoint 1 with groupID 0
zigbee2mqtt:info  2020-07-14 01:14:35: MQTT publish: topic 'zigbee2mqtt/kitchen_switch', payload '{"battery":100,"voltage":3042,"linkquality":111,"click":"long"}'
zigbee2mqtt:info  2020-07-14 01:14:35: MQTT publish: topic 'zigbee2mqtt/kitchen_switch', payload '{"battery":100,"voltage":3042,"linkquality":111,"click":""}'
zigbee2mqtt:info  2020-07-14 01:14:35: MQTT publish: topic 'zigbee2mqtt/kitchen_switch/click', payload 'long'
zigbee2mqtt:debug 2020-07-14 01:14:37: Received Zigbee message from 'kitchen_switch', type 'attributeReport', cluster 'genOnOff', data '{"onOff":0}' from endpoint 1 with groupID 0
zigbee2mqtt:debug 2020-07-14 01:14:37: Received Zigbee message from 'kitchen_switch', type 'attributeReport', cluster 'genOnOff', data '{"onOff":1}' from endpoint 1 with groupID 0
zigbee2mqtt:info  2020-07-14 01:14:37: MQTT publish: topic 'zigbee2mqtt/kitchen_switch', payload '{"battery":100,"voltage":3042,"linkquality":105,"click":"long_release","duration":2228}'
zigbee2mqtt:info  2020-07-14 01:14:37: MQTT publish: topic 'zigbee2mqtt/kitchen_switch', payload '{"battery":100,"voltage":3042,"linkquality":105,"click":"single"}'
zigbee2mqtt:info  2020-07-14 01:14:37: MQTT publish: topic 'zigbee2mqtt/kitchen_switch', payload '{"battery":100,"voltage":3042,"linkquality":105,"click":""}'
zigbee2mqtt:info  2020-07-14 01:14:37: MQTT publish: topic 'zigbee2mqtt/kitchen_switch', payload '{"battery":100,"voltage":3042,"linkquality":105,"click":""}'
zigbee2mqtt:info  2020-07-14 01:14:37: MQTT publish: topic 'zigbee2mqtt/kitchen_switch/click', payload 'single'
zigbee2mqtt:info  2020-07-14 01:14:37: MQTT publish: topic 'zigbee2mqtt/kitchen_switch/click', payload 'long_release'
```

It would only break one click but the logic is that `store[deviceID].long` is not set to `false` in this case.

Koenkk, 2020-07-14:
>  I believe that to solve this issue we have to ignore the click up event for long presses and relay only on timeouts or use another timeout if no click up event is received after long press detection.

A timeout should be fine I think, so if no click up is received (e.g.) 5 seconds after the click down is received publish it. Can you make a PR?

kitopopo, 2020-07-19:
Dear @Koenkk , 

I wanted to comment on my experience about pulsation failures on WXKG01LM devices, it was kind of funny and I don't quite understand it, I have organized and revised my wiring and I have exchanged ports usb 3.0 of my hub (keyboard and mouse and hdd) in my raspberry 4 and the problem has improved 99%. I would like to apologize because I thought that the problem was due to some update of the dev branch but I have observed it all weekend and now there are no errors. i don´t understand why it just failed this device.  is very strange, perhaps proximity or interference between a sonoff and the cc2531 due to channel 11, I don't know but now it doesn't fail after organizing my wiring. 
I take this opportunity to comment that my next device (when they have stock in tindie shop) will definitely be the zzh cc2652, especially for the good support that you give here. Thank you very much, I do not understand where the problem was but it seems that it was my problem. A cordial greeting.

ArkadiuszNiemiec, 2020-07-19:
@Koenkk This should be it: https://github.com/Koenkk/zigbee-herdsman-converters/pull/1418

kitopopo, 2020-07-21:
Good evening mr @Koenkk , @ArkadiuszNiemiec , 

First of all I would like to apologize for believing that the xiaomi switch WXKG01LM pulsation failure was due to a dev branch update. 
The problem was in a domoticz script that manages the xiaomi switch orders, after performing these tasks I would set the value of the switch to 'off' and this caused the switch to not work well in the following orders, resetting it back to status 'off' after each order. You have to let the switch stay on the last press received so that it works well on the next pulsation.  
In all the scripts of my 4 xiaomi switches I had since they returned to the initial value 'off', for this reason I thought that a physical problem of all my switches was impossible

I also take this opportunity to comment that my next device when the store has stock in stock will be the zzh 2652r and not the conve ii. I choose the zzh device mainly because of the great support it is giving and that Mr. @Koenkk  has given all this time. Surely the zzh will be a great device. Thank you very much for your support.  I am very grateful

Receive a warm greeting

kitopopo, 2020-08-17:
Dear @Koenkk ,

After two weeks working correctly today my switchs fails again, i can see in the zigbee2mqtt log that every pulsation is detected two times in the log, i attach the log file:

```
ago 17 15:33:32 raspberrypi npm[5913]: Zigbee2MQTT:info  2020-08-17 15:33:32: MQTT publish: topic 'zigbee2mqtt/Switch salon', payload '{"action":"single","battery":100,"linkquality":107,"voltage":3132}'
ago 17 15:33:32 raspberrypi npm[5913]: Zigbee2MQTT:info  2020-08-17 15:33:32: MQTT publish: topic 'zigbee2mqtt/Switch salon', payload '{"battery":100,"click":"single","linkquality":107,"voltage":3132}'

ago 17 15:36:56 raspberrypi npm[5913]: Zigbee2MQTT:info  2020-08-17 15:36:56: MQTT publish: topic 'zigbee2mqtt/Switch salon', payload '{"action":"single","battery":100,"linkquality":107,"voltage":3132}'
ago 17 15:36:56 raspberrypi npm[5913]: Zigbee2MQTT:info  2020-08-17 15:36:56: MQTT publish: topic 'zigbee2mqtt/Switch salon', payload '{"battery":100,"click":"single","linkquality":107,"voltage":3132}'
```

I test with the xiaomi door sensor but this sensor is detected only one time, the problem appear only in all xiaomi switchs WXKG01LM

how i can solve the problem?

ArkadiuszNiemiec, 2020-08-17:
Here are my logs on newest stable version:
```
Zigbee2MQTT:debug 2020-08-17 16:01:48: Received Zigbee message from 'office_desk_switch', type 'attributeReport', cluster 'genOnOff', data '{"onOff":0}' from endpoint 1 with groupID 0
Zigbee2MQTT:debug 2020-08-17 16:01:49: Received Zigbee message from 'office_desk_switch', type 'attributeReport', cluster 'genOnOff', data '{"onOff":1}' from endpoint 1 with groupID 0
Zigbee2MQTT:info  2020-08-17 16:01:49: MQTT publish: topic 'zigbee2mqtt/office_desk_switch', payload '{"battery":100,"linkquality":105,"voltage":3032,"action":"single"}'
Zigbee2MQTT:info  2020-08-17 16:01:49: MQTT publish: topic 'zigbee2mqtt/office_desk_switch', payload '{"battery":100,"linkquality":105,"voltage":3032,"click":"single"}'
Zigbee2MQTT:info  2020-08-17 16:01:49: MQTT publish: topic 'zigbee2mqtt/office_desk_switch', payload '{"battery":100,"linkquality":105,"voltage":3032,"action":""}'
Zigbee2MQTT:info  2020-08-17 16:01:49: MQTT publish: topic 'zigbee2mqtt/office_desk_switch/action', payload 'single'
Zigbee2MQTT:info  2020-08-17 16:01:49: MQTT publish: topic 'zigbee2mqtt/office_desk_switch', payload '{"battery":100,"linkquality":105,"voltage":3032,"click":""}'
Zigbee2MQTT:info  2020-08-17 16:01:49: MQTT publish: topic 'zigbee2mqtt/office_desk_switch/click', payload 'single'
```

The switch reports click once but it's published twice on MQTT: once as `action` and once as `click`. On my setup it does not break anything as I only parse `clicks`. It's also not a problem on Home Assistant. It's probably not a source of your switch failing.

[14 more replies]