Koenkk/zigbee2mqtt #3446: CC2531 adapter stopped working

Eddy555, 2020-04-29:
## What happened
CC2531 adapter stopped working after running succesfully for several weeks. My alternative adapter also does not work.
It's connected to 9 x Aqara temperature humidty sensors and has been working fine.

## What did you expect to happen
Adapter should work
It says the adapter cannot be found at the path, but I can find it using ls

## How to reproduce it (minimal and precise)
Restart Zigbee2mqtt and the errors appear

## Debug Info
Zigbee2mqtt version: 1.12.2
Adapter hardware: CC2531 (I have 2 and both do the same thing)
Adapter firmware version: Koenkk Z-stack Home 1.2 Coordinator 20190608 on one adapter
My alternative is running CC2531_DEFAULT_20190608

Log errors attached
[zigbee2mqtt errors.txt](https://github.com/Koenkk/zigbee2mqtt/files/4550174/zigbee2mqtt.errors.txt)

dreimer1986, 2020-04-29:
Some configuration information might be useful. Log says that your configured serial port is not right.

Eddy555, 2020-04-29:
Thanks, I've attached my config file.
As I say it was working fine then one day I saw it stopped receiving data part way through the day and hasn't worked since and I can't figure out why.
The config for adapter seems OK (unless I've overlooked something).
[zigbee2mqtt config.txt](https://github.com/Koenkk/zigbee2mqtt/files/4551614/zigbee2mqtt.config.txt)

Koenkk, 2020-04-29:
- What is the output of `ls -l /dev/serial/by-id`
- Can you change

```
  port: >-
    /dev/serial/by-id/usb-Texas_Instruments_TI_CC2531_USB_CDC___0X00124B000BE95916-if00
```

to 
```
  port: /dev/serial/by-id/usb-Texas_Instruments_TI_CC2531_USB_CDC___0X00124B000BE95916-if00
```

Eddy555, 2020-04-30:
total 0
crw-rw----    1 root     audio     166,   0 Apr 21 06:07 usb-Texas_Instruments_TI_CC2531_USB_CDC___0X00124B0018E21E95-if00

To be sure I've copied this back into the config and still get the same error.

I've changed the config to as you suggested but it appears to revert back and add >- after saving.
And the error still persists

BTW the addon is now updated to 1.13.0

dreimer1986, 2020-04-30:
Not clue if the CC2531 sets up several serial ports as my shiny new CC1352P_2 does, but did you try to change the port to:
/dev/serial/by-id/usb-Texas_Instruments_TI_CC2531_USB_CDC___0X00124B0018E21E95-if00

Eddy555, 2020-04-30:
Thanks, I've tried your suggestion and it's still not working. :-(

Koenkk, 2020-04-30:
Can you try reflashing the adapter?

Eddy555, 2020-04-30:
I'll give it a go and report back

spectreman51, 2020-04-30:
same problems with CC2531 and aqara devices. I tested with many raspberry (rpi2 & rpi zero), with many adaptaters. Always the same problems with aqara. They stopped working after one or two hours.

Eddy555, 2020-05-01:
It's not that the Aqara devices lose connection, it's that the CC2531 stops being found by the addon. It had worked solidly for serveral weeks and then the adapter can't be found (although it did briefly work yesterday again.)

z4rn0x, 2020-05-01:
Are you on a raspberry? If yes, have you tried another order supply?

Eddy555, 2020-05-01:
Yeah it's running on a Pi 3 B+. 
Do you mean use a different hardware to run Hassio?

z4rn0x, 2020-05-01:
Sorry, autocorrect, another power supply

Eddy555, 2020-05-01:
Ah right, that makes more sense (Damned autocorrect :-) )
I'll dig one out and try that.

spectreman51, 2020-05-02:
I changed the power supply for 3A charger --> same issue. 
If I watch MQTT explorer, the topic zigbee2mqtt/device disappears

Eddy555, 2020-05-11:
Is it worth trying a different hardware adapter? Is there another adpater people use?

closed 2020-07-18 by stale[bot]

Bobbik1, 2021-12-31:
I have similar issue, but it started after my friend ask me if I could try his adapter , because it stopped to work for him. It isn't work , but after i plug in my adapter and restart everything, it didn't work also. Zigbbe2mqtt doesn't contain any logs