Koenkk/zigbee2mqtt #3449: YPHIX dimmer

jeremyklamer, 2020-04-29:
I am running: Hassio on Raspberry pi 3 with the zigbee2mqtt 1.12.2 and Mosquitto broker 5.1 addons. Usb Stick: CC2531 

I recently bought the YPHIX dimmer https://www.ledlampendirect.nl/zigbee-230v-muurdimmer.html?sqr=zigbee%20dimmer&#box-additional 

The documentation for the dimmer can be found [here]((https://github.com/Koenkk/zigbee2mqtt/files/4552733/50208695_Manual_ENG.pdf))
I set the device to pairing mode and it is discovered but not as the YPHIX dimmer, it is discovered as the [ZigBee knob smart dimmer (ZG2835RAC) by Sunricher](https://www.zigbee2mqtt.io/devices/ZG2835RAC.html). This is all fine because it might be using the same components with a different brand. Problem however is that i can only control the on or off states. The other states are nog even logged in Zigbee2Mqtt.

Now my initial idea was that I was going to have to configure the whole device myself, but seeing as it is already configured I would like to troubleshoot and add the dimming option. Can someone set me off in the right direction to start working on this? 

Extra info: Sending on and off to the dimmer works fine. Changing the settings on the dimmer itself does not send any message

All ideas are welcome

Koenkk, 2020-04-29:
Please try changing the entry in `devices.js` to  (see https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html how to get to devices.js)

```js
{
    zigbeeModel: ['Micro Smart Dimmer'],
    model: 'ZG2835RAC',
    vendor: 'Sunricher',
    description: 'ZigBee knob smart dimmer',
    extend: generic.light_onoff_brightness,
    meta: {configureKey: 1},
    whiteLabel: [
        {vendor: 'YPHIX', model: '50208695},
    ],
    configure: async (device, coordinatorEndpoint) => {
        const endpoint = device.getEndpoint(1);
        await bind(endpoint, coordinatorEndpoint, ['genOnOff', 'genLevelCtrl']);
    },
},
```

thejman1989, 2020-04-30:
First of all, thanks for the reply and thanks for creating zigbee2mqtt! (Legend)

I have added the configuration as you said and the device was then sending back the brightness! Home assistant did not automatically pickup the changes so I thought ill just remove (using zigbee2mqtt/bridge/config/remove ) the device and pair it again, what can go wrong. For now it won't pair anymore. I will investigate more when I have time. 

When I get it working I'll create a PR.

closed 2020-07-02 by thejman1989