Koenkk/zigbee2mqtt #3451: MQTT Error at startup of zigbee2mqtt

SargonofAssyria, 2020-04-29:
startup info.
`info  2020-04-29 19:38:29: Starting zigbee2mqtt version 1.12.2-dev (commit #d8ec59b)
info  2020-04-29 19:38:29: Starting zigbee-herdsman...
debug 2020-04-29 19:38:29: Using zigbee-herdsman with settings: '{"network":{"panID":7778,"extendedPanID":[238,221,238,2
21,238,221,238,221],"channelList":[25],"networkKey":"HIDDEN"},"databasePath":"/home/pi/zigbee2mqtt/data/database.db","da
tabaseBackupPath":"/home/pi/zigbee2mqtt/data/database.db.backup","backupPath":"/home/pi/zigbee2mqtt/data/coordinator_bac
kup.json","serialPort":{"baudRate":115200,"rtscts":true,"path":"/dev/ttyACM1"},"adapter":{"concurrent":null}}'
info  2020-04-29 19:38:31: zigbee-herdsman started

At every startup i get this error.
`debug 2020-04-29 19:38:31: Setup reporting for '0x00158d0002b7d689' - 1 - genOnOff
info  2020-04-29 19:38:31: Successfully setup reporting for '0x00158d0002b7d689' - 1 - genOnOff
debug 2020-04-29 19:38:31: Setup reporting for '0x00158d0002b7d689' - 1 - genLevelCtrl
info  2020-04-29 19:38:31: Successfully setup reporting for '0x00158d0002b7d689' - 1 - genLevelCtrl
debug 2020-04-29 19:38:31: Setup reporting for '0x00158d0002b7d689' - 1 - lightingColorCtrl
debug 2020-04-29 19:38:31: Received Zigbee message from 'Overlooplamp', type 'readResponse', cluster 'lightingColorCtrl'
, data '{}' from endpoint 1 with groupID 0
error 2020-04-29 19:38:31: Not connected to MQTT server!
error 2020-04-29 19:38:31: Cannot send message: topic: 'homeassistant/light/0x00158d0002b7d689/light/config', payload: '
{"brightness":true,"color_temp":true,"schema":"json","command_topic":"zigbee2mqtt/Overlooplamp/set","state_topic":"zigbee2mqtt/Overlooplamp","json_attributes_topic":"zigbee2mqtt/Overlooplamp","name":"Overlooplamp_light","unique_id":"0x00158
d0002b7d689_light_zigbee2mqtt","device":{"identifiers":["zigbee2mqtt_0x00158d0002b7d689"],"name":"Overlooplamp","sw_vers
ion":"Zigbee2mqtt 1.12.2-dev","model":"Aqara smart LED bulb (ZNLDP12LM)","manufacturer":"Xiaomi"},"availability_topic":"
zigbee2mqtt/bridge/state"}
error 2020-04-29 19:38:31: Not connected to MQTT server!
error 2020-04-29 19:38:31: Cannot send message: topic: 'homeassistant/sensor/0x00158d0002b7d689/linkquality/config', pay
load: '{"icon":"mdi:signal","unit_of_measurement":"lqi","value_template":"{{ value_json.linkquality }}","state_topic":"z
igbee2mqtt/Overlooplamp","json_attributes_topic":"zigbee2mqtt/Overlooplamp","name":"Overlooplamp_linkquality","unique_id
":"0x00158d0002b7d689_linkquality_zigbee2mqtt","device":{"identifiers":["zigbee2mqtt_0x00158d0002b7d689"],"name":"Overlo
oplamp","sw_version":"Zigbee2mqtt 1.12.2-dev","model":"Aqara smart LED bulb (ZNLDP12LM)","manufacturer":"Xiaomi"},"avail
ability_topic":"zigbee2mqtt/bridge/state"}
error 2020-04-29 19:38:31: Failed to setup reporting for '0x00158d0002b7d689' - Error: Read 0x00158d0002b7d689/1 lightin
gColorCtrl(["colorCapabilities"], {"timeout":10000,"manufacturerCode":null,"disableDefaultResponse":true}) failed (Error
: Status 'UNSUPPORTED_ATTRIBUTE')
    at Endpoint.<anonymous> (/home/pi/zigbee2mqtt/node_modules/zigbee-herdsman/dist/controller/model/endpoint.js:182:23)
    at Generator.next (<anonymous>)
    at fulfilled (/home/pi/zigbee2mqtt/node_modules/zigbee-herdsman/dist/controller/model/endpoint.js:5:58)
`
It looks like the devices is working properly. Info is send to MQTT before the connection to the local MQTT server was established.
Other same Bulbs do not give this error.

Koenkk, 2020-04-30:
Should be fixed in latest dev branch (for hassio edge wait 4 hours before updating).

SargonofAssyria, 2020-05-01:
Tnx, that's solved.
There was an other error. Should I make a new issue?
`error 2020-05-01 09:51:53: Failed to setup reporting for '0x00158d0002b7d689' - Error: Read 0x00158d0002b7d689/1 lightingColorCtrl(["colorCapabilities"], {"timeout":10000,"manufacturerCode":null,"disableDefaultResponse":true}) failed (Error: Status 'UNSUPPORTED_ATTRIBUTE')
error 2020-05-01 09:52:34: Failed to execute LQI for 'Ganglamp'
`

closed 2020-05-01 by SargonofAssyria