Koenkk/zigbee2mqtt #3455: Osram Smart+ plug not supporting routing table?

Kosta-Github, 2020-04-30:
# Bug Report

It looks like that the [OSRAM Smart+ plug (AB3257001NJ)](https://www.zigbee2mqtt.io/devices/AB3257001NJ.html) does not support the `routing table` command? At least when trying to generate the network topology, the `routing table` was always failing on my 2 plugs with the error message `0x7cb03eaa0a09632f (55401)failed: routingTable, OSRAM Smart+ plug (AB3257001NJ)` within the graphviz diagram and on within the logs: `Failed to execute routing table for '0x7cb03eaa0a09632f'`.

## Debug Info

Zigbee2mqtt version: 1.13.0
Adapter hardware: CC2531
Adapter firmware version: zStack12, 20190608

Kosta-Github, 2020-04-30:
enabled `DEBUG` output for the driver as well, getting this for the `network` scan (note the `out of bounds` access error):
```
zigbee2mqtt    | zigbee2mqtt:info  2020-04-30 17:48:17: Starting network scan (includeRoutes 'true')
zigbee2mqtt    | 2020-04-30T15:48:17.246Z zigbee-herdsman:adapter:zStack:znp:SREQ --> ZDO - mgmtLqiReq - {"dstaddr":0,"startindex":0}
zigbee2mqtt    | 2020-04-30T15:48:17.247Z zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,3,37,49,0,0,0,23]
zigbee2mqtt    | 2020-04-30T15:48:17.256Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,1,101,49,0,85]
zigbee2mqtt    | 2020-04-30T15:48:17.257Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,1,101,49,0,85]
zigbee2mqtt    | 2020-04-30T15:48:17.257Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 1 - 3 - 5 - 49 - [0] - 85
zigbee2mqtt    | 2020-04-30T15:48:17.257Z zigbee-herdsman:adapter:zStack:znp:SRSP <-- ZDO - mgmtLqiReq - {"status":0}
zigbee2mqtt    | 2020-04-30T15:48:17.258Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
zigbee2mqtt    | 2020-04-30T15:48:17.267Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,72,69,177,0,0,0,3,0,3,221,221,221,221,221,221,221,221,237,229,119,15,0,111,13,0,162,66,18,2,1,49,221,221,221,221,221,221,221,221,117,103,9,10,170,62,176,124,141,181,21,2,1,1,221,221,221,221,221,221,221,221,47,99,9,10,170,62,176,124,105,216,21,2,1,1,185]
zigbee2mqtt    | 2020-04-30T15:48:17.268Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,72,69,177,0,0,0,3,0,3,221,221,221,221,221,221,221,221,237,229,119,15,0,111,13,0,162,66,18,2,1,49,221,221,221,221,221,221,221,221,117,103,9,10,170,62,176,124,141,181,21,2,1,1,221,221,221,221,221,221,221,221,47,99,9,10,170,62,176,124,105,216,21,2,1,1,185]
zigbee2mqtt    | 2020-04-30T15:48:17.269Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 72 - 2 - 5 - 177 - [0,0,0,3,0,3,221,221,221,221,221,221,221,221,237,229,119,15,0,111,13,0,162,66,18,2,1,49,221,221,221,221,221,221,221,221,117,103,9,10,170,62,176,124,141,181,21,2,1,1,221,221,221,221,221,221,221,221,47,99,9,10,170,62,176,124,105,216,21,2,1,1] - 185
zigbee2mqtt    | 2020-04-30T15:48:17.270Z zigbee-herdsman:adapter:zStack:znp:AREQ <-- ZDO - mgmtLqiRsp - {"srcaddr":0,"status":0,"neighbortableentries":3,"startindex":0,"neighborlqilistcount":3,"neighborlqilist":[{"extPandId":"0xdddddddddddddddd","extAddr":"0x000d6f000f77e5ed","nwkAddr":17058,"deviceType":2,"rxOnWhenIdle":0,"relationship":1,"permitJoin":2,"depth":1,"lqi":49},{"extPandId":"0xdddddddddddddddd","extAddr":"0x7cb03eaa0a096775","nwkAddr":46477,"deviceType":1,"rxOnWhenIdle":1,"relationship":1,"permitJoin":2,"depth":1,"lqi":1},{"extPandId":"0xdddddddddddddddd","extAddr":"0x7cb03eaa0a09632f","nwkAddr":55401,"deviceType":1,"rxOnWhenIdle":1,"relationship":1,"permitJoin":2,"depth":1,"lqi":1}]}
zigbee2mqtt    | 2020-04-30T15:48:17.271Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
zigbee2mqtt    | 2020-04-30T15:48:17.274Z zigbee-herdsman:adapter:zStack:znp:SREQ --> ZDO - mgmtRtgReq - {"dstaddr":0,"startindex":0}
zigbee2mqtt    | 2020-04-30T15:48:17.276Z zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,3,37,50,0,0,0,20]
zigbee2mqtt    | 2020-04-30T15:48:17.285Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,1,101,50,0,86]
zigbee2mqtt    | 2020-04-30T15:48:17.286Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,1,101,50,0,86]
zigbee2mqtt    | 2020-04-30T15:48:17.287Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 1 - 3 - 5 - 50 - [0] - 86
zigbee2mqtt    | 2020-04-30T15:48:17.288Z zigbee-herdsman:adapter:zStack:znp:SRSP <-- ZDO - mgmtRtgReq - {"status":0}
zigbee2mqtt    | 2020-04-30T15:48:17.288Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
zigbee2mqtt    | 2020-04-30T15:48:17.292Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,31,69,178,0,0,0,5,0,5,141,181,0,141,181,105,216,0,141,181,70,191,1,254,255,233,184,1,254,255,55,119,1,254,255,137]
zigbee2mqtt    | 2020-04-30T15:48:17.292Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,31,69,178,0,0,0,5,0,5,141,181,0,141,181,105,216,0,141,181,70,191,1,254,255,233,184,1,254,255,55,119,1,254,255,137]
zigbee2mqtt    | 2020-04-30T15:48:17.293Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 31 - 2 - 5 - 178 - [0,0,0,5,0,5,141,181,0,141,181,105,216,0,141,181,70,191,1,254,255,233,184,1,254,255,55,119,1,254,255] - 137
zigbee2mqtt    | 2020-04-30T15:48:17.294Z zigbee-herdsman:adapter:zStack:znp:AREQ <-- ZDO - mgmtRtgRsp - {"srcaddr":0,"status":0,"routingtableentries":5,"startindex":0,"routingtablelistcount":5,"routingtablelist":[{"destNwkAddr":46477,"routeStatus":"ACTIVE","nextHopNwkAddr":46477},{"destNwkAddr":55401,"routeStatus":"ACTIVE","nextHopNwkAddr":46477},{"destNwkAddr":48966,"routeStatus":"DISCOVERY_UNDERWAY","nextHopNwkAddr":65534},{"destNwkAddr":47337,"routeStatus":"DISCOVERY_UNDERWAY","nextHopNwkAddr":65534},{"destNwkAddr":30519,"routeStatus":"DISCOVERY_UNDERWAY","nextHopNwkAddr":65534}]}
zigbee2mqtt    | 2020-04-30T15:48:17.295Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
zigbee2mqtt    | 2020-04-30T15:48:17.299Z zigbee-herdsman:adapter:zStack:znp:SREQ --> ZDO - mgmtLqiReq - {"dstaddr":46477,"startindex":0}
zigbee2mqtt    | 2020-04-30T15:48:17.300Z zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,3,37,49,141,181,0,47]
zigbee2mqtt    | 2020-04-30T15:48:17.316Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,1,101,49,0,85]
[... cut ...]
zigbee2mqtt    |   "0x000d6f000fbf9c4e" [style="rounded, dashed, filled", fillcolor="#fff8ce", fontcolor="#000000", label="{motion_guest|0x000d6f000fbf9c4e (7910)|OSRAM SMART+ Motion Sensor (AC01353010G)|2020-04-30T15:25:59+02:00}"];
zigbee2mqtt    |   "0x000d6f000fbf71b2" [style="rounded, dashed, filled", fillcolor="#fff8ce", fontcolor="#000000", label="{motion_schlafzimmer|0x000d6f000fbf71b2 (47337)|OSRAM SMART+ Motion Sensor (AC01353010G)|2020-04-30T17:18:04+02:00}"];
zigbee2mqtt    | }'
```

Koenkk, 2020-05-02:
Closing this one in favour of https://github.com/Koenkk/zigbee-herdsman/issues/170 (as its a herdsman issue).

closed 2020-05-04 by Kosta-Github