Koenkk/zigbee2mqtt #3456: brightness_move: stop immediately turns bulb off

mdhom, 2020-04-30:
# Bug Report
## What happened
When sending `{"state":"ON","brightness_move":40}` my bulb starts dimming up as expected. As soon as i send `{"state":"ON","brightness_move":"stop"}` the bulb immediately goes completely dark, state On but brightness 1 (`{"state":"ON","brightness":1,"linkquality":52}`). Interestingly, when i move brightness in negative direction, the bulb takes the previous (254) brightness, so goes completely bright after stopping.

## What did you expect to happen
The bulb should keep the currently reached brightness and not go dark / bright again.

## How to reproduce it (minimal and precise)
Send `{"state":"ON","brightness_move":40}`, wait 1-2 seconds until lamp goes brighter, then send `{"state":"ON","brightness_move":"stop"}' and see that bulb jumps to dark  / bright state.

## Debug Info
Zigbee2mqtt version: 1.12.2, commit 7d27a54
Adapter hardware: CC2531
Adapter firmware version: zStack12 source routing rev 20190619

Koenkk, 2020-04-30:
- With what bulb is this?
- Can you provide the logging when running with herdsman debug logging?

To enable herdsman debug logging, see https://www.zigbee2mqtt.io/information/debug.html#zigbee-herdsman-debug-logging

mdhom, 2020-04-30:
- With all bulbs i have, tested with
  - Philips Hue Lightbar
  - Ikea Tradfri
- I have loggingg enabled via docker:
`docker run -d -it -v $(pwd)/zigbee2mqtt:/app/data --device=/dev/ttyACM0 -e TZ=Europe/Berlin -e DEBUG=* --restart always --name ZigbeeTwoMqtt koenkk/zigbee2mqtt`
You can find the log here: https://pastebin.com/NBbhv6rE (dimming started at line 236)

Koenkk, 2020-05-02:
I think the issue occurs because you also provide the `state`.  (`'{"state":"ON","brightness_move":-40}'`, can you try without: `'{"brightness_move":-40}'`

mdhom, 2020-05-02:
Actually i am setting the state because without, it seemed not to be possible to dimm up a switched off lamp, am i right?

Koenkk, 2020-05-02:
I expect the `brightness_move` alone also to turn on the bulb (and turn off once brightness 0 has been reached).

mdhom, 2020-05-02:
That seems not to be the case, i tested it right now and the bulb stays dark if currently switched off and dimming up..
State before test:
`{"state":"OFF","brightness":1,"linkquality":44}`
Command start move:
`{"brightness_move":40}`
Bulb stays dark...

mdhom, 2020-05-02:
Also, if dimming down, it ends at brightness 1 and state ON. Don't know if that is expected behaviour or not...

Koenkk, 2020-05-02:
That's indeed true. Should work after changing https://github.com/Koenkk/zigbee-herdsman-converters/blob/master/converters/toZigbee.js#L289 to ` await entity.command('genLevelCtrl', 'moveWithOnOff', payload, getOptions(meta.mapped));`

mdhom, 2020-05-02:
Thanks, tested it! Dimming up when lamp is initially off works now! When dimming completely down, the state is ON and brightness 1, is that how you would expect it to be? I think it should be off or at least brightness 0?

Koenkk, 2020-05-04:
Implemented this, also fixed the state reading. Note that the brightens always stays at 1 (the bulb returns this value), but this is overruled by the state "OFF". Available in latest dev branch, assuming this can be closed.

closed 2020-05-04 by Koenkk