Koenkk/zigbee2mqtt #3461: [featyre] debounce for groups

sjorge, 2020-04-30:
# Bug Report
## What happened

I noticed I get a attReport from 2 bulbs in a group usually within a second of eachother, the group update get published twice to mqtt with the sam payload.

## What did you expect to happen

I expected to be able to set 'debounce' for a group, but that had no effect.
Looking at the code, I don't think that is currently support. So I guess this is a feature request.

## How to reproduce it (minimal and precise)
Add 2 trådfri bulbs to a group, then change the state via the group.
- both bulbs change state
- both bulbs withing < 1 sec publish an attReport for genOnOff with the new state
- group gets published twice more (I think 3 times as once gets published form setting the state via the group too)

sjorge, 2020-07-02:
Still wanted

sjorge, 2020-08-01:
Unstale

sjorge, 2020-09-01:
Unstals

Koenkk, 2020-10-08:
I'm wondering that instead of debounce, we can also not publish a new group state when there is nothing different to be published. This has to advantage of not having to wait the debounce time before a new state gets published.

sjorge, 2020-10-08:
Should we not try to keep the behavior consistent between groups and devices?

Although both would work for my use case.

Koenkk, 2020-10-08:
Both yes and no, the no part is not to add features/settings for things that are not used. Besides that not republishing the "optimistic" state for groups when it didn't change makes a lot of sense, I can't imagine why you don't want this.

sjorge, 2020-10-08:
True but should this not be done for groups and devices then?

Koenkk, 2020-10-09:
I only meant the optimistic state for groups based on device changes. So a /set on a group will still publish a state update, even when identical. But when e.g. 2 bulbs in a group report they have been turned off, the group will only publish it's updated state once. 

Please try the `groups.js` changes from https://github.com/Koenkk/zigbee2mqtt/pull/4623/files

sjorge, 2020-10-09:
Seems to behave as expected.

Both triggered via remote or via mqtt will result in 1 publish

Koenkk, 2020-10-09:
Good, does it solve your OP issue? Because this is exactly what you described in "How to reproduce"

sjorge, 2020-10-09:
Yeah this is fine for my use case. my 6 bulb groups now no longer spam mqtt messages 1-3 sec after toggling them via de group.

Koenkk, 2020-10-09:
Great, merged I assume we can close this now :smile:

closed 2020-10-09 by Koenkk

sjorge, 2020-10-09:
Something is still broken after restarting z2m, I need to investigate more but it looks like it does not republish retained messages for groups now.

sjorge, 2020-10-10:
I've just verified this is indeed the case! :(

How I verified this:
- start the following mqtt subs
  - `mosquitto_sub -t 'zigbee2mqtt/office/floor_light' | jq .`
  - `mosquitto_sub -t 'zigbee2mqtt/office/floor_light/bulb' | jq .`
- restart zigbee2mqtt

Only the sub that was for a device got the payload published again as expected, the sub for the group did not.
This breaks most of my node-red integration, I assume it's a side effect of checking if the current payload changed vs the cached payload. Which after restart, would be the case and also be skipped.

I guess the one that publishes at startup should get called with a 'force' flag or something that will always (re)publish.

Koenkk, 2020-10-10:
@sjorge fixed

sjorge, 2020-10-10:
Looks good now after grabbing the latest dev.