Koenkk/zigbee2mqtt #3463: Error on 1 ZNLDP12LM - Xiaomi Aqara smart LED bulb (Router)

SargonofAssyria, 2020-05-01:
## Debug Info
Zigbee2mqtt version:  Starting zigbee2mqtt version 1.13.0-dev (commit #e792f8b)
Adapter hardware: CC1352P-2
Adapter firmware version: zStack3x0 20200417
I have 3 lightbulbs ZNLDP12LM. One of them gives this error.
`debug 2020-05-01 10:10:37: Setup reporting for '0x00158d0002b7d689' - 1 - genOnOff
info  2020-05-01 10:10:37: Successfully setup reporting for '0x00158d0002b7d689' - 1 - genOnOff
debug 2020-05-01 10:10:37: Setup reporting for '0x00158d0002b7d689' - 1 - genLevelCtrl
info  2020-05-01 10:10:38: Successfully setup reporting for '0x00158d0002b7d689' - 1 - genLevelCtrl
debug 2020-05-01 10:10:38: Setup reporting for '0x00158d0002b7d689' - 1 - lightingColorCtrl
debug 2020-05-01 10:10:38: Received Zigbee message from 'Overlooplamp', type 'readResponse', cluster 'lightingColorCtrl', data '{}' from endpoint 1 with groupID 0
error 2020-05-01 10:10:38: Failed to setup reporting for '0x00158d0002b7d689' - Error: Read 0x00158d0002b7d689/1 lightingColorCtrl(["colorCapabilities"], {"timeout":10000,"manufacturerCode":null,"disableDefaultResponse":true}) failed (Error: Status 'UNSUPPORTED_ATTRIBUTE')
`
It looks like the bulb is functioning normally (I am not sure).
Is this an error in the bulb or in the software?
How can I solve it?

Koenkk, 2020-05-02:
It's strange that only one of the bulbs gives `UNSUPPORTED_ATTRIBUTE` (that means the software between them is different). This results in reporting not being setup for this device, depending on your situation it might not be problematic (if you don't control it via a remote).

SargonofAssyria, 2020-05-02:
It seems that the bulb is working properly.
The error came when I changed the controller from CC2531 to CC1352P-2

closed 2020-05-02 by SargonofAssyria