Koenkk/zigbee2mqtt #3468: Not a bug - help needed - Ikea Tradfri motion sensor timout settings

sanchosk, 2020-05-01:
Hi. This is just a help request, not a support ticket.
I have a new version of the IKEA motion sensor from Tradfri line.
This version does not seem to have any setting for duration of the occupancy signal.
The pre-defined value seems to be 3 minutes, this is quite a lot for the usage I need the device for.

Does anyone have any experience with setting some lower values for the duration, please?

nhausman1, 2020-05-14:
I don't have a Tradfri motion sensor myself but I do have Aqara's version. By most accounts I've seen online, [github ticket](https://github.com/Koenkk/zigbee2mqtt/issues/2324#issuecomment-554667879) and [reddit post](https://www.reddit.com/r/tradfri/comments/cogibl/motion_detector_time/ewi1hse/), the new Tradfri motion sensor only reports motion once every 3 minutes. It will ignore all motion events in the meantime. Aqara's "sleep" time is slightly less at 1 minute.

HereIsFrank, 2020-06-28:
No, shorter durations are not possible. The new version of Tradfri motion sensor sends an "occupancy":false message after exactly 3 minutes.  But the sensor is **not** ignoring a motion event in the meantime. 

I have been studying the sensor's behaviour. After a true signal, no new signal is sent for the first 100 seconds (this time always varies by a few seconds for me). If a new movement is detected in the time window between 100 seconds and 3 minutes, the sensor immediately sends a new true signal. If no movement is detected in this time window, the sensor sends a false signal exactly 3 minutes after the true signal.

MattWestb, 2020-07-29:
@HereIsFrank The Ikeas motion sensors and other light steering devices are not sending "motion".
The ikea its sending the command "Light on with off time X" and no "off " or "no motion". If motion its retrigged its sending one new "light on with of time X". The off command or no motion its  produced in the Z2M and not by the device or the zigbee mech.

closed 2020-09-05 by github-actions[bot]