Koenkk/zigbee2mqtt #3469: Device ID naming [feature request]

to4ko, 2020-05-01:
Hi! Will it be possible to change devices id naming convention when homeassistant: true found in configuration?

Reason - it's not possible to get entity last_updated\last_triggered for entities with id like 0x00158...

I know that it's possible to rename entities in HA. But what if there are 2-3 HA instances and qround 130+ devices?

 
# Bug Report
## What happened

## What did you expect to happen

## How to reproduce it (minimal and precise)

## Debug Info
Zigbee2mqtt version:
Adapter hardware: CC2531, CC2530, CC26X2R1 or CC1352P-2
Adapter firmware version:

alting, 2020-05-01:
Confirm!

sjorge, 2020-05-01:
Doesn't setting the friendly_name propagate to HA? I'm pretty sure it did last I looked at it.

to4ko, 2020-05-01:
> Doesn't setting the friendly_name propagate to HA? I'm pretty sure it did last I looked at it.

nope ( 

friendly_name set in z2m devices.yaml to Test_Relay

[image]

to4ko, 2020-05-11:
Looks like removing 0x00 nameing will not solve the case. It's HA that is not able to operate with entities name stating with number...

So only option - change naming to HA style sensor.temperature_0x00158 instead of sensor.0x00158...

closed 2020-05-12 by to4ko