Koenkk/zigbee2mqtt #3470: All sensors get "off" state at HA restart

tuga49, 2020-05-01:
Hi,

I'm facing an issue but don't know how to solve. I got some xiaomi window contact sensor (binary_sensor). When I restart Home Assistant, all my sensors go to "off" state and after some seconds, they get the real state. For example, sensor A is "on" and sensor B is "off". If I restart HA, when it is available again, both sensors are "off" and then after +- 20 seconds, sensor B keeps "off" because is it previous state, and sensor A go to "on". That make fake triggers happens. I have automations from "off" to "on" and it get triggered by the sensor "A" in this case, and I don't wan't that because his real state is currently "on" and I don't wan't to trigger the automation again. This is normal to happen? is there any way to keep the last state of the sensor imediatly after HA restart, so it don't make fake triggers? Thank's for any help ;)

Kosta-Github, 2020-05-01:
you might want to use the `retain` flag/functionality for those sensors: https://www.zigbee2mqtt.io/information/configuration.html#example
or globally for all:
https://www.zigbee2mqtt.io/information/configuration.html#all-devices

see also: http://www.steves-internet-guide.com/mqtt-retained-messages-example/

tuga49, 2020-05-01:
> you might want to use the `retain` flag/functionality for those sensors: https://www.zigbee2mqtt.io/information/configuration.html#example
> or globally for all: [...]

Hi. Thank's for your reply. How do I add it to all devices and not one by one? this globally link you send, is to use inside a device I think. Can I use retain by defualt for all sensors? Where do I add the retain?

tuga49, 2020-05-01:
> you might want to use the `retain` flag/functionality for those sensors: https://www.zigbee2mqtt.io/information/configuration.html#example
> or globally for all: [...]

I added retain: true for each device and it works now. Really thank's mate. I tried retain but on mqtt config in HA and not on zigbee2mqtt devices. But anyway, is there any way to set retain only one time and avoid to do that in every device? if yes, where do I add it? thanks

Kosta-Github, 2020-05-01:
looks like there is no global setting/option for this... maybe can file a new feature request?

tuga49, 2020-05-01:
> looks like there is no global setting/option for this... maybe can file a new feature request?

Sure. Anyway, thank's for your help ;) my problem is solved. Thank's bro

closed 2020-05-01 by tuga49