Koenkk/zigbee2mqtt #3471: Zigbee2mqtt TLS connection to Mosquitto fails

jonaspaulo, 2020-05-01:
Hi,

Starting a fresh setup with the following versions:
CC2531 running latest Zigbee 1.2 HA FW: CC2531_DEFAULT_20190608
Zigbee2mqtt 1.13.0 (commit #55e6283) running on docker
Mosquitto (Eclipse Oficial Version 1.69) running on docker

I am trying to connect zigbee2mqtt to Mosquitto via a TLS connection (mqtts) but not being able to. I have configured everything as per below:

**Cert generation:**
```
sudo openssl genrsa -des3 -out ca.key 2048
sudo openssl req -new -x509 -days 1826 -key ca.key -out ca.crt 
sudo openssl genrsa -out server.key 2048
sudo openssl req -new -out server.csr -key server.key
sudo openssl x509 -req -in server.csr -CA ca.crt -CAkey ca.key -CAcreateserial -out server.crt -days 360

```

**mosquitto.conf:**
```
connection_messages true
log_timestamp true
log_dest file /mosquitto/log/mosquitto.log
log_dest stdout
log_timestamp_format %Y-%m-%d %H:%M:%S
persistence true
persistence_file mosquitto.db
persistence_location /mosquitto/data/
port 8883
allow_anonymous false
cafile /mosquitto/certs/ca.crt
keyfile /mosquitto/certs/server.key
certfile /mosquitto/certs/server.crt
tls_version tlsv1.2
socket_domain ipv4
```

**Zigbee2mqtt configuration.yaml:**
```
homeassistant: true
permit_join: false
mqtt:
  base_topic: zigbee2mqtt
  server: 'mqtt://127.0.0.1:8883'
  ca: '/ca.crt'
  client_id: CC2531NAS
serial:
  port: /dev/ttyACM0
advanced:
  network_key: GENERATE
  channel: 11
devices: devices.yaml
groups: groups.yaml
```

**On Zigbee2mqtt container logs:**
`zigbee2mqtt:info  2020-05-01 19:28:33: Connecting to MQTT server at mqtt://127.0.0.1:8883 zigbee2mqtt:error 2020-05-01 19:28:43: Not connected to MQTT server!`

**On Mosquitto container logs:**
```
2020-05-01 17:31:35: New connection from 127.0.0.1 on port 8883
2020-05-01 17:31:35: OpenSSL Error[0]: error:140260FC:SSL routines:ACCEPT_SR_CLNT_HELLO:unknown protocol
2020-05-01 17:31:35: Socket error on client <unknown>, disconnecting.
```

I have tested the the ca.crt in MqttFX MacOS app and it works with the same Mosquitto server.
I also found this issue which can explain this behaviour?  https://zigbee2mqtt.discourse.group/t/zigbee2mqtt-mosquitto/1068/2

Any tips?
Thanks a lot

jonaspaulo, 2020-05-02:
Hi,

Some news on this. First I tried to create a new server cert and ca with a Hostname instead of the IP I was using previously (127.0.0.1). But this didn't work. Nevertheless I stuck with this new certificate.

I noticed two things I had to change to make it work:

1- We indeed need to change the protocol on zigbee2mqtt configuration.yaml to mqtts: instead of mqtt: despite the fact the one has defined the certificate. This removes the error above on the mosquitto side of "unknown protocol".

2- Then I tried to use client side authentication instead of server side as above (only with the ca.crt). For this I created a new client cert and key from the ca.crt and ca.key of the server and defined them on the zigbee2mqtt configuration.yaml file which indeed worked:

```
mqtt:
  base_topic: zigbee2mqtt
  server: 'mqtts://HOSTNAME:8883'
  ca: '/certs/ca.crt'
  key: '/certs/client.key'
  cert: '/certs/client.crt'
  client_id: CC2531NAS
```

So in short it seems that server side authentication only is not working (at least on zigbee2mqtt side) for some reason since it works with another client and client side authentication with client certificates is working just fine.

Thanks!

noohi, 2020-05-07:
I'm having the same issue "Not connected to MQTT server" as of recently.
Node-red and Home Assistant connect just fine, haven't touched MQTT server (Mosquitto) settings.
All in docker containers.

Tried rebooting one or the other, nothing helps and it seemingly randomly starts working again on multiple reboots of Z2M.

closed 2020-07-26 by stale[bot]

h3po, 2020-12-01:
i've got exactly the same issue, but without tls involved. mosquitto acks the incoming connection and then closes it with "socket error". zigbee2mqtt doesn't log anything except "Not connected to MQTT server!" even on debug level. the problem comes and goes between restarts, without changing the config. The only explanation I can come up with is a timing problem or memory corruption.

h3po, 2020-12-01:
wireshark capture of the tcp session: [zigbee2mqtt_mosquitto_socketerror.zip](https://github.com/Koenkk/zigbee2mqtt/files/5622480/zigbee2mqtt_mosquitto_socketerror.zip)
wireshark screenshot:
[image]
relevant mosquitto log output:
```
Dec 01 13:11:10 containers mosquitto[14441]: 1606824670: New connection from 172.18.0.5 on port 1883.    
Dec 01 13:11:10 containers mosquitto[14441]: 1606824670: Sending CONNACK to 172.18.0.5 (0, 5)        
Dec 01 13:11:10 containers mosquitto[14441]: 1606824670: Socket error on client <unknown>, disconnecting.
```
relevant zigbee2mqtt log output:
```
Dec 01 13:06:29 containers zigbee2mqtt[14738]: Zigbee2MQTT:info  2020-12-01 13:06:29: Connecting to MQTT server at mqtt://mosquitto.homeassistant                                                                                            
Dec 01 13:06:29 containers zigbee2mqtt[14738]: Zigbee2MQTT:debug 2020-12-01 13:06:29: Using MQTT client ID: 'zigbee2mqtt'                                                                                                                    
Dec 01 13:06:39 containers zigbee2mqtt[14738]: Zigbee2MQTT:error 2020-12-01 13:06:39: Not connected to MQTT server!
```
mosquitto configuration:
```
use_username_as_clientid true
persistence true
persistence_location /mosquitto/data
log_type error
log_type warning
log_type notice
log_type information
log_type debug
allow_anonymous false
password_file /mosquitto/config/passwords.txt
```
zigbee2mqtt configuration:
```
homeassistant: true
permit_join: true
mqtt:
  base_topic: zigbee2mqtt
  server: 'mqtt://mosquitto.homeassistant'
  client_id: zigbee2mqtt
  user: zigbee2mqtt
  password: redacted
serial:
  port: /dev/ttyACM0
advanced:
  log_level: debug
  homeassistant_legacy_triggers: false
  network_key: [redacted]
  pan_id: [redacted]
  ext_pan_id: [redacted]
  channel: 20
devices: [...]
```

h3po, 2020-12-01:
this might be related, but supposedly doesn't affect mqtt.js v4.2.1 which you are using https://github.com/mqttjs/MQTT.js/issues/1189

Koenkk, 2020-12-01:
@h3po since this is an issue in the used library and not Zigbee2MQTT itself, it would be good if you can make a small example and submit an issue here: https://github.com/mqttjs/MQTT.js/issues

h3po, 2020-12-03:
sorry i have no clue about node. can you maybe fix zigbee2mqtt so the debug flag gets passed to the library logger?

gabric098, 2020-12-15:
Hi @h3po I'm experiencing the same issue ad you described. Did you make any progress in figuring out how to make it behave?

gabric098, 2020-12-15:
Just a note here. If you're using self signed SSL certificate you must remember to set the `reject_unauthorized` flag to `false`

That worked for me. 
I feel like the flag's name is not super clear but it's clearly documented in the docs:

```
# Optional: disable self-signed SSL certificates (default: true)
  reject_unauthorized: true
```

h3po, 2020-12-16:
> Hi @h3po I'm experiencing the same issue you described. Did you make any progress in figuring out how to make it behave?
No, it comes and goes as I restart the machine. Restarting the service itself does not help.

gabric098, 2020-12-16:
> > Hi @h3po I'm experiencing the same issue you described. Did you make any progress in figuring out how to make it behave?
> 
> No, it comes and goes as I restart the machine. Restarting the service itself does not help.

I guess this https://github.com/Koenkk/zigbee2mqtt/issues/3471#issuecomment-745419570 doesn't apply to your case right?

sheran29, 2023-09-11:
hi
  i am also facing same issue by certificate. it  is working fine if i am using only mqtt. but if i use with zigbee it is not connecting to server.
  getting a error message ie. mqtt fails to connect