Koenkk/zigbee2mqtt #3474: BUG containers fails start docker

nexro567, 2020-05-01:
Describe the bug
Containers fails with starting after adding new devices. I can not fix it anymore.

To Reproduce
Steps to reproduce the behavior:

Go to '...'
Click on '....'
Scroll down to '....'
See error
Expected behavior

Steps I took to try resolve the problem
Open a new container in docker. Still same problem.
I added in config
advanced: pan_id: 0x1a63

No fix, after that I reset local state, still no change :(

Pertinent logs
sing '/app/data' as data directory
Creating configuration file...

zigbee2mqtt@1.13.0 start /app
node index.js

�[32mzigbee2mqtt:info �[39m 2020-05-01 22:33:26: Logging to console and directory: '/app/data/log/2020-05-01.22-33-26' filename: log.txt
�[32mzigbee2mqtt:info �[39m 2020-05-01 22:33:26: Starting zigbee2mqtt version 1.13.0 (commit #55e6283)
�[32mzigbee2mqtt:info �[39m 2020-05-01 22:33:26: Starting zigbee-herdsman...
�[32mzigbee2mqtt:info �[39m 2020-05-01 22:33:28: zigbee-herdsman started
�[32mzigbee2mqtt:info �[39m 2020-05-01 22:33:28: Coordinator firmware version: '{"type":"zStack12","meta":{"transportrev":2,"product":0,"majorrel":2,"minorrel":6,"maintrel":3,"revision":20190608}}'
�[32mzigbee2mqtt:info �[39m 2020-05-01 22:33:28: Currently 0 devices are joined:
�[33mzigbee2mqtt:warn �[39m 2020-05-01 22:33:28: permit_join set to true in configuration.yaml.
�[33mzigbee2mqtt:warn �[39m 2020-05-01 22:33:28: Allowing new devices to join.
�[33mzigbee2mqtt:warn �[39m 2020-05-01 22:33:28: Set permit_join to false once you joined all devices.
�[32mzigbee2mqtt:info �[39m 2020-05-01 22:33:28: Zigbee: allowing new devices to join.
�[32mzigbee2mqtt:info �[39m 2020-05-01 22:33:28: Connecting to MQTT server at mqtt://localhost
�[32mzigbee2mqtt:info �[39m 2020-05-01 22:33:28: Connected to MQTT server
�[32mzigbee2mqtt:info �[39m 2020-05-01 22:33:28: MQTT publish: topic 'zigbee2mqtt/bridge/state', payload 'online'
�[32mzigbee2mqtt:info �[39m 2020-05-01 22:33:28: MQTT publish: topic 'zigbee2mqtt/bridge/config', payload '{"version":"1.13.0","commit":"55e6283","coordinator":{"type":"zStack12","meta":{"transportrev":2,"product":0,"majorrel":2,"minorrel":6,"maintrel":3,"revision":20190608}},"log_level":"info","permit_join":true}'
�[32mzigbee2mqtt:info �[39m 2020-05-01 22:33:33: MQTT publish: topic 'zigbee2mqtt/bridge/config/devices', payload '[{"ieeeAddr":"0x00124b00193e51cc","type":"Coordinator","networkAddress":0,"friendly_name":"Coordinator","softwareBuildID":"zStack12","dateCode":"20190608","lastSeen":1588372413635}]'
�[32mzigbee2mqtt:info �[39m 2020-05-01 22:33:33: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"devices","message":[{"ieeeAddr":"0x00124b00193e51cc","type":"Coordinator","networkAddress":0,"friendly_name":"Coordinator","softwareBuildID":"zStack12","dateCode":"20190608","lastSeen":1588372413637}]}'
�[32mzigbee2mqtt:info �[39m 2020-05-01 22:33:50: MQTT publish: topic 'zigbee2mqtt/bridge/state', payload 'offline'
�[32mzigbee2mqtt:info �[39m 2020-05-01 22:33:50: Disconnecting from MQTT server
�[32mzigbee2mqtt:info �[39m 2020-05-01 22:33:50: zigbee-herdsman stopped
Using '/app/data' as data directory

zigbee2mqtt@1.13.0 start /app
node index.js

�[32mzigbee2mqtt:info �[39m 2020-05-01 22:34:19: Logging to console and directory: '/app/data/log/2020-05-01.22-34-19' filename: log.txt
�[32mzigbee2mqtt:info �[39m 2020-05-01 22:34:19: Starting zigbee2mqtt version 1.13.0 (commit #55e6283)
�[32mzigbee2mqtt:info �[39m 2020-05-01 22:34:19: Starting zigbee-herdsman...
�[32mzigbee2mqtt:info �[39m 2020-05-01 22:34:20: zigbee-herdsman started
�[32mzigbee2mqtt:info �[39m 2020-05-01 22:34:20: Coordinator firmware version: '{"type":"zStack12","meta":{"transportrev":2,"product":0,"majorrel":2,"minorrel":6,"maintrel":3,"revision":20190608}}'
�[32mzigbee2mqtt:info �[39m 2020-05-01 22:34:20: Currently 0 devices are joined:
�[33mzigbee2mqtt:warn �[39m 2020-05-01 22:34:20: permit_join set to true in configuration.yaml.
�[33mzigbee2mqtt:warn �[39m 2020-05-01 22:34:20: Allowing new devices to join.
�[33mzigbee2mqtt:warn �[39m 2020-05-01 22:34:20: Set permit_join to false once you joined all devices.
�[32mzigbee2mqtt:info �[39m 2020-05-01 22:34:20: Zigbee: allowing new devices to join.
�[32mzigbee2mqtt:info �[39m 2020-05-01 22:34:21: Connecting to MQTT server at mqtt://localhost
�[32mzigbee2mqtt:info �[39m 2020-05-01 22:34:21: Connected to MQTT server
�[32mzigbee2mqtt:info �[39m 2020-05-01 22:34:21: MQTT publish: topic 'zigbee2mqtt/bridge/state', payload 'online'
�[32mzigbee2mqtt:info �[39m 2020-05-01 22:34:21: MQTT publish: topic 'zigbee2mqtt/bridge/config', payload '{"version":"1.13.0","commit":"55e6283","coordinator":{"type":"zStack12","meta":{"transportrev":2,"product":0,"majorrel":2,"minorrel":6,"maintrel":3,"revision":20190608}},"log_level":"info","permit_join":true}'
�[32mzigbee2mqtt:info �[39m 2020-05-01 22:34:21: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"entity_not_found","message":{"friendly_name":"0xccccccfffeb3c6a7"}}'
�[31mzigbee2mqtt:error�[39m 2020-05-01 22:34:21: Entity '0xccccccfffeb3c6a7' is unknown
�[32mzigbee2mqtt:info �[39m 2020-05-01 22:34:26: MQTT publish: topic 'zigbee2mqtt/bridge/config/devices', payload '[{"ieeeAddr":"0x00124b00193e51cc","type":"Coordinator","networkAddress":0,"friendly_name":"Coordinator","softwareBuildID":"zStack12","dateCode":"20190608","lastSeen":1588372466148}]'
�[32mzigbee2mqtt:info �[39m 2020-05-01 22:34:26: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"devices","message":[{"ieeeAddr":"0x00124b00193e51cc","type":"Coordinator","networkAddress":0,"friendly_name":"Coordinator","softwareBuildID":"zStack12","dateCode":"20190608","lastSeen":1588372466159}]}'
�[32mzigbee2mqtt:info �[39m 2020-05-01 22:34:51: Starting network scan (includeRoutes 'false')
�[31mzigbee2mqtt:error�[39m 2020-05-01 22:34:51: Adapter disconnected, stopping
�[32mzigbee2mqtt:info �[39m 2020-05-01 22:34:51: MQTT publish: topic 'zigbee2mqtt/bridge/state', payload 'offline'
�[32mzigbee2mqtt:info �[39m 2020-05-01 22:34:51: Disconnecting from MQTT server
�[31mzigbee2mqtt:error�[39m 2020-05-01 22:34:51: Failed to stop zigbee
npm ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! zigbee2mqtt@1.13.0 start: node index.js
npm ERR! Exit status 1
npm ERR!
npm ERR! Failed at the zigbee2mqtt@1.13.0 start script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.

npm ERR! A complete log of this run can be found in:
npm ERR! /root/.npm/_logs/2020-05-01T22_34_51_639Z-debug.log
Using '/app/data' as data directory

mgrzelczyk, 2020-05-03:
I have the same error.

Using '/app/data' as data directory                                                                                                                                                                                                                                                                                        
Creating configuration file...                                                                                                                                                                                                                                                                                             
                                                                                                                                                                                                                                                                                                                           
> zigbee2mqtt@1.13.0 start [...]
                                                                                                                                                                                                                                                                                                                           
zigbee2mqtt:info  2020-05-03 15:15:12: Logging to console and directory: '/app/data/log/2020-05-03.15-15-11' filename: log.txt                                                                                                                                                                                             
zigbee2mqtt:info  2020-05-03 15:15:15: Starting zigbee2mqtt version 1.13.0 (commit #55e6283)                                                                                                                                                                                                                               
zigbee2mqtt:info  2020-05-03 15:15:15: Starting zigbee-herdsman...                                                                                                                                                                                                                                                         
zigbee2mqtt:error 2020-05-03 15:15:23: Error while starting zigbee-herdsman                                                                                                                                                                                                                                                
zigbee2mqtt:error 2020-05-03 15:15:23: Failed to start zigbee                                                                                                                                                                                                                                                              
zigbee2mqtt:error 2020-05-03 15:15:23: Exiting...                                                                                                                                                                                                                                                                          
zigbee2mqtt:error 2020-05-03 15:15:23: Error: Error while opening serialport 'Error: Error: No such file or directory, cannot open /dev/ttyACM0'                                                                                                                                                                           
    at Znp.<anonymous> (/app/node_modules/zigbee-herdsman/dist/adapter/z-stack/znp/znp.js:113:32)                                                                                                                                                                                                                          
    at Generator.next (<anonymous>)                                                                                                                                                                                                                                                                                        
    at /app/node_modules/zigbee-herdsman/dist/adapter/z-stack/znp/znp.js:8:71                                                                                                                                                                                                                                              
    at new Promise (<anonymous>)                                                                                                                                                                                                                                                                                           
    at __awaiter (/app/node_modules/zigbee-herdsman/dist/adapter/z-stack/znp/znp.js:4:12)                                                                                                                                                                                                                                  
    at SerialPort.<anonymous> (/app/node_modules/zigbee-herdsman/dist/adapter/z-stack/znp/znp.js:111:49)                                                                                                                                                                                                                   
[... cut ...]
                                                                                                                                                                                                                                                                                                                           
npm ERR! A complete log of this run can be found in:                                                                                                                                                                                                                                                                       
npm ERR!     /root/.npm/_logs/2020-05-03T15_15_23_523Z-debug.log

aureliosaraiva, 2020-05-03:
I too

```
zigbee2mqtt:info  2020-05-03 18:09:06: Starting zigbee2mqtt version 1.13.0 (commit #unknown)
zigbee2mqtt:info  2020-05-03 18:09:06: Starting zigbee-herdsman...
zigbee2mqtt:error 2020-05-03 18:09:13: Error while starting zigbee-herdsman
zigbee2mqtt:error 2020-05-03 18:09:13: Failed to start zigbee
zigbee2mqtt:error 2020-05-03 18:09:13: Exiting...
zigbee2mqtt:error 2020-05-03 18:09:13: Error: Failed to connect to the adapter (Error: SRSP - SYS - ping after 6000ms)
    at ZStackAdapter.<anonymous> (/zigbee2mqtt-1.13.0/node_modules/zigbee-herdsman/dist/adapter/z-stack/adapter/zStackAdapter.js:71:23)
npm ERR! code ELIFECYCLE
    at Generator.throw (<anonymous>)
    at rejected (/zigbee2mqtt-1.13.0/node_modules/zigbee-herdsman/dist/adapter/z-stack/adapter/zStackAdapter.js:6:65)
npm ERR! errno 1
npm
 ERR! zigbee2mqtt@1.13.0 start: node index.js
npm ERR! Exit status 1
npm ERR! 
npm ERR! Failed at the zigbee2mqtt@1.13.0 start script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.
```

nexro567, 2020-05-03:
@aureliosaraiva 

I solved the problem for 50%. 

1. I made a backup of my files in the config map. 
2. Did a reset in Zigbee2MQTTAssistant -> status -> Reset Local State. 
3. After that I reset the CC2531 sniffer. You find two buttons. The button nearby the USB port is the reset button.
4. After that I removed the docker image and container. Restart the server and made a new docker image and container. Put the old config lines without devices in the new configuration.yaml file.

The problem, after the steps above I experiencing now is when I restart the server that the zigbee2mqtt container fails after a minute. After the container has failed for a minute I can restart it. After that I have no more problems anymore till I restart the server.. 

I hope it will work again with the steps above.

dartheide, 2020-05-07:
Same problem, switch back to 1.12.2 problem solved.

docker pull koenkk/zigbee2mqtt:1.12.2

nexro567, 2020-05-10:
Thanks @dartheide that works!

nexro567, 2020-05-24:
I found also a fault in my configuration of docker. Check your docker container HomeAssistant or another container where you use your serial device of zigbee, because your serial device can be used by only one container.

https://github.com/Koenkk/zigbee2mqtt/issues/3487#issuecomment-632402095

dartheide, 2020-05-24:
With the latest version (1.13.1) my problem is solved. so 
docker pull koenkk/zigbee2mqtt:latest is working.

closed 2020-07-26 by stale[bot]