Koenkk/zigbee2mqtt #3475: No mqtt-messages from devices with 1.13.0

chrjangit, 2020-05-02:
## What happened
Upgraded to 1.13.0, and after that there are no messages from my devices (Ikea and Aqara). No errors in logs. When pressing buttons, no activity in command window. The whole start sequence looks good. Only new might be "genOnOff from '901' failed (undefined)".

## What did you expect to happen
Nothing new.

## How to reproduce it (minimal and precise)
npm start

## Debug Info
Zigbee2mqtt version: 1.13.0
Adapter hardware: Conbee2
Adapter firmware version: 0x264a0700

chrjangit, 2020-05-02:
With:
advanced:
    availability_timeout: 10
...there is a message zigbee2mqtt:debug 2020-05-02 16:37:37: Failed to ping 'room/outlet1'

chrjangit, 2020-05-02:
I suspect it is a problem with the Conbee II.

Koenkk, 2020-05-02:
Probably a conbee issue indeed, please post the details here: https://github.com/Koenkk/zigbee-herdsman/issues/72

closed 2020-07-10 by stale[bot]