Koenkk/zigbee2mqtt #3477: Cannot pair Philips LTG002

vdkeybus, 2020-05-02:
I have a set of 2 (as in 'a pair', but avoiding that term in this context) Philips GU10 with Bluetooth.

Initially one of them paired about a month ago, but the signal quality was not good. At some point the link must have been lost, and I used the Philips dimmer switch to reset it, in an effort to re-pair it.

Since then it won't pair to my Zigbee network at all. Also the other bulb won't pair. I also tried removing the ID from the system, but that failed because the device could not be found !? (This really should be possible, and should require you to reset the device, rather than not removing it because the device should announce its departure. But this is maybe not relevant now.)

Manually removed it from database.db and data/database.db. Still shows up in HomeAssistant, but at least now Z2M says that the device is unknown. Still no pairing.

Any idea on how to debug this. I would like not to reinstall or reset Z2M, if possible.

Thanks.

Koenkk, 2020-05-04:
Make sure to check https://www.zigbee2mqtt.io/information/FAQ.html#why-does-my-device-not-or-fail-to-pair. For Hue also try touchlink reset: https://www.zigbee2mqtt.io/information/touchlink.html

lb20, 2020-05-05:
I have a very similar issue to this, but with some Xiaomi motion sensors. Interested to hear what solution works for you; I've tried many, but haven't tried a couple of things listed on that support page yet. I Will report back here also.

closed 2020-07-12 by stale[bot]