Koenkk/zigbee2mqtt #3479: MAC no ack (233) on all Tradfri (LED1732G11) at the same time

mdhom, 2020-05-03:
# Bug Report
## What happened
My mesh was running stable for about two days, everythin working fine. Then i noticed a Tradfri bulb not working anymore (getting MAC no ack 233 errors when trying to set anything). I tested alle the other bulbs in my home, no Tradfri working anymore but all Non-Tradfri bulbs (and sensors) working perfectly.

## How to reproduce it (minimal and precise)
I noticed two events timely relatively close to when i noticed the problem:
- One Hue bulb lost its power due to a cable slipping out of the power outlet. I couldn't replug it immediately (only two hours later) because my kids were slipping close to it
- I ran another CC2531 USB stick attached to my PC for debugging purposes (see #3456 and #3322) for about an hour. That stick was used previously for coordination, but after i switched to the currently used stick, i repaired all devices, so no bulb is paired to the one mentioned here. With the test setup, the only things i did was pairing another IKEA Bulb (LED1842G3) and paring a Scenic friends of Hue switch (see #3322).

## Debug Info
Zigbee2mqtt version: 1.12.2 (7d27a54)
Adapter hardware: CC2531
Adapter firmware version: zStack12 20190619 (Source Routing)

mdhom, 2020-05-03:
I sniffed the traffic when trying to send an on/off command to one of the Tradfri bulbs:
[TradfriMacNoAck.zip](https://github.com/Koenkk/zigbee2mqtt/files/4569735/TradfriMacNoAck.zip)

mdhom, 2020-05-03:
I also sniffed the traffic for a bulb showing No network route (205) error:
- Bulb address: 0x680ae2fffea13fe9

See attached wireshark capture: [TradfriNoRoute.zip](https://github.com/Koenkk/zigbee2mqtt/files/4569757/TradfriNoRoute.zip)

Koenkk, 2020-05-04:
The errors are valid, zigbee2mqtt sends the commands but the bulbs doesn't respond.

Assuming the bulb has power, it probably has a bad signal, make sure to check: https://www.zigbee2mqtt.io/how_tos/how_to_improve_network_range.html

mdhom, 2020-05-05:
Hm i can't believe it is bad signal as it worked for few days without any problem. I have already setup all the measures for improving network range, but i can't talk to an IKEA bulb which is only 4-5m away from the coordinator.
Also i found that not only my ikea bulbs but also an ikea symfonisk input device does not talk to zigbee2mqtt anymore.

Koenkk, 2020-05-06:
When reconnecting the bulbs to mains, do you see a device announce? (please sniff the traffic of this).

closed 2020-07-12 by stale[bot]