Koenkk/zigbee2mqtt #3480: Can't pair Gledopto GL-FL-004TZ

itavero, 2020-05-03:
I'm trying to bind two [Gledopto GL-FL004TZ](https://www.zigbee2mqtt.io/devices/GL-FL-004TZ.html) floodlights that I have mounted on the wall outside.
In the past I used them with a Philips Hue bridge without any issues (paired straight away), but I'm migrating everything to zigbee2mqtt, so I also wanted to migrate these lights.

Today I removed the floodlights from the Philips Hue bridge via their app.
After that I made it possible to switch the lights on/off quickly (normally there is no light switch for them, other than the circuit breaker).

After enabling the pairing mode (via Zigbee2MqttAssistant), I tried to reset a single light about 20 times, but I did not succeed in binding it. I put the log level of zigbee2mqtt to debug and kept an eye on the log while giving it some more attempts, but there were no messages related to pairing at all.

When resetting the floodlight, it blinks green 4 times, after that the cold white LEDs go on and then the cold and warm white LEDs are on.
What I get from the instructions on the website and in the manual that came with the lights, this means that the light is reset.

By the way, there are some other Philips Hue lights outside (near these floodlights, with clear line of sight) that I have already paired with zigbee2mqtt. So the floodlights should be within reach of the network.

What else can I try?
I'm at the point of giving up and just using the Philips Hue bridge instead. So frustrating... (especially since binding 29 other devices went pretty smooth, some of them even after attempting to pair these floodlights).

## Debug Info
Zigbee2mqtt version: 1.13.0
Adapter hardware: CC26X2R1
Adapter firmware version: 20191106

itavero, 2020-05-03:
Decided to move my Raspberry Pi and CC26X2R1 outside, to debug this issue further and behold:
```
info  2020-05-03 16:28:05: Device '0x00124b001eae05e7' joined
info  2020-05-03 16:28:05: Starting interview of '0x00124b001eae05e7'
info  2020-05-03 16:28:05: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"device_connected","message":{"friendly_name":"0x00124b001eae05e7"}}'
info  2020-05-03 16:28:05: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"pairing","message":"interview_started","meta":{"friendly_name":"0x00124b001eae05e7"}}'
info  2020-05-03 16:28:05: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"device_announced","message":"announce","meta":{"friendly_name":"0x00124b001eae05e7"}}'
warn  2020-05-03 16:28:06: Received message from unsupported device with Zigbee model 'GL-FL-004TZS'
warn  2020-05-03 16:28:06: Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html.
```
Apparently it's a slightly different model. I'll try to follow the instructions to add support for it and open up a PR once it works (or seems to work). 😉

itavero, 2020-05-03:
By duplicating the config of the GL-FL-004TZ in Koenkk/zigbee-herdsman-converters, the device can be paired and I can control it via MQTT directly. I'll create a PR for this as soon as I have time.

Home Assistant discovery is not yet working though.

itavero, 2020-05-03:
In a similar fashion, I tried to copy the line for the `GL-FL-004TZ` in `lib/extension/homeassistant.js` and renamed it to `GL-FL-004TZS`.
Unfortunately, after restarting zigbee2mqtt, it still does not show up in HA.

Koenkk, 2020-05-04:
Good to you managed to pair the device. Assuming this can be closed now.

closed 2020-05-04 by Koenkk