Koenkk/zigbee2mqtt #3481: Tuya Zigbee TYZB02 / TS0042 - NO COMPONENTS

Ciubas, 2020-05-03:
This is the: TS0042 but recognize as: _TYZB02_keyjhapk | NO COMPONENTS Maybe this device is not (yet) supported by Zigbee2Mqtt?

Board: CC2531
Version of Zigbee2MqttAssistant (this software): 0.3.147 
Version of Zigbee2Mqtt: 1.8.0
Coordinator version: 20190608

[image]

robertcsakany, 2020-05-16:
I'm using it. It supperted from: https://github.com/Koenkk/zigbee-herdsman-converters/commit/cbbbbe653d525bdb51cf9ddf0c3e1575bcdf567d

Ciubas, 2020-05-16:
> I'm using it. It supperted from: [Koenkk/zigbee-herdsman-converters@cbbbbe6](https://github.com/Koenkk/zigbee-herdsman-converters/commit/cbbbbe653d525bdb51cf9ddf0c3e1575bcdf567d)

How you update this file on your HA? :)

Koenkk, 2020-05-17:
You are using a old version of zigbee2mqtt, first update to the latest one (1.13.0).

Ciubas, 2020-05-17:
[image]

This is my problem, how? Where? I read this 4 times and don't have clue where.

Koenkk, 2020-05-17:
I expect to get some update notification, please ask this question here: https://github.com/danielwelch/hassio-zigbee2mqtt (I don't use hassio so don't know).

Ciubas, 2020-05-17:
Thanks. I use supervisor in docker.

Ciubas, 2020-05-18:
I find the problem - I dont know how, but the repo link in Addons deleted, so I dinn't have aupdates! Now its ok :D 

And its working! :D

closed 2020-05-18 by Koenkk