Koenkk/zigbee2mqtt #3482: INNR SP220 pairing impossible

Markus1970, 2020-05-03:
Hi All,
I use at the moment Version 1.13.0 on my Bridge. Tried today to pair a Innr Sp220 plug which seems to be the new type of the Innr Sp120 plug. But pairing fails. Also nothing visible in the Logfile. Is this one not supported at the moment? Is there something i can do to get it working?

Thanks

Markus

Alexis6464, 2020-05-03:
Hi,
I have the same problem with my GLEDOPTO GL-S-007Z.
They work before, but i reinstalled my HASSIO in a new machine with the lastest version.

But all my xiaomi device is working fine.

I don't know why...

Thanks,
Alexis

sjorge, 2020-05-03:
@markus1540 have you tried stopping z2m, removing and inserting the usb, starting z2m before you try to pair.

markus1540, 2020-05-03:
> @markus1540 have you tried stopping z2m, removing and inserting the usb, starting z2m before you try to pair.

No but thanks for asking. 😬

Markus1970, 2020-05-04:
I guess,I've found the issue.. I've swapped few weeks ago to another cc2530 Module. Connected now GND to P2_0 and pairing was working. Did thid always in the past on the CC2530 but forgotten to connect this on the last one.

Thanks 
Markus

Alexis6464, 2020-05-06:
> Hi,
> I have the same problem with my GLEDOPTO GL-S-007Z.
> They work before, but i reinstalled my HASSIO in a new machine with the lastest version.
> 
> But all my xiaomi device is working fine.
> 
> I don't know why...
> 
> Thanks,
> Alexis

With HASSIO 0.109.3, my problem it's solved.

closed 2020-05-06 by Markus1970