Koenkk/zigbee2mqtt #3483: Wall socket ZNCZ04LM restore state before power fail

divanikus, 2020-05-03:
Hello,

I have a couple of ZNCZ04LM wall sockets which work just fine, but if I experience a power fail, they do not come back on power restore, until I press their ON button. I've tried to pair it Aquara Hub, and there they have an option to restore state after powerfail in their config. It works, wall socket restores it's state when I plug it back to power. I've tried to switch off Aquara Hub too, and seems like it works even without it. 
But when I pair it back to zigbee2mqtt, nothing. Unit doesn't turn ON on power restore. 

Is there a way to configure it?

divanikus, 2020-05-03:
So ok, the proposed fix for ZNCZ02LM doesn't work, because ZNCZ04LM doesn't have the corresponding converter in zigbee-herdsman-converters. Simply usnig ZNCZ02LM doesn't work, because seems like it has different domain/attribute codes. But I can't figure out how to edit ZNCZ02LM's converter to match ZNCZ04LM data. 

Some raw data about domains and attributes of 04 can be found here: https://github.com/dresden-elektronik/deconz-rest-plugin/issues/2583

divanikus, 2020-05-20:
I made a PR with basic implementation of this feature. Thanks to community for help
https://github.com/Koenkk/zigbee-herdsman-converters/pull/1250

Koenkk, 2020-05-21:
Merged your pr, big thanks for adding this! Assuming this can be closed now.

closed 2020-05-21 by Koenkk