Koenkk/zigbee2mqtt #3486: Feature request: send cluster commands

jd1900, 2020-05-03:
It would be great to send cluster messages to groups for debugging. I haven't found a way to do that.

For instance, if I want to send a message to change brightness using clusters to see how a device behaves, how can I do that?

For instance, a message like this (which is created by another device):
`zigbee2mqtt:debug 2020-05-03 20:03:52: Received Zigbee message from 'xxxx', type 'commandMove', cluster 'genLevelCtrl', data '{"movemode":1,"rate":255}' from endpoint 1 with groupID 3`

Koenkk, 2020-05-04:
I'm not sure what you mean, but the example you gave (`commandMove`) can be send to a group with `brighntess_move` (documentation: https://www.zigbee2mqtt.io/information/mqtt_topics_and_message_structure.html#zigbee2mqttfriendly_nameset).

jd1900, 2020-05-04:
There are several clusters like "genLevelCtrl", "genOnOff", "genBasic", "genScenes", "genTime", "genOta".

Can we directly deal with the clusters commands? Instead of sending device specific commands?

AFAIK (I may be wrong) if we send cluster commands we don't have to worry about specific device implementations. Or we can "try" some functionalities not displayed in the device data.

Koenkk, 2020-05-04:
Sending these commands can be done by adding a toZigbee to https://github.com/Koenkk/zigbee-herdsman-converters/blob/master/converters/toZigbee.js and adding it to the `toZigbee` array in `devices.js`.

jd1900, 2020-05-04:
That's what I'm asking for as a feature request.

Adding that could facilitate debugging devices and adding new devices and new functionalities to already added ones.

closed 2020-07-11 by stale[bot]