Koenkk/zigbee2mqtt #3487: Failed to start zigbee

aureliosaraiva, 2020-05-03:
# Bug Report
## What happened
Some errors started to occur in the log and I can't upload the zigbee service. This occurred during the pairing of some devices.

## What did you expect to happen
Let the zigbee network work

## How to reproduce it (minimal and precise)
I dont know

## Debug Info
Zigbee2mqtt version: 1.13.0
Adapter hardware: CC2531
Adapter firmware version:

```
zigbee2mqtt:info  2020-05-03 18:09:06: Starting zigbee2mqtt version 1.13.0 (commit #unknown)
zigbee2mqtt:info  2020-05-03 18:09:06: Starting zigbee-herdsman...
zigbee2mqtt:error 2020-05-03 18:09:13: Error while starting zigbee-herdsman
zigbee2mqtt:error 2020-05-03 18:09:13: Failed to start zigbee
zigbee2mqtt:error 2020-05-03 18:09:13: Exiting...
zigbee2mqtt:error 2020-05-03 18:09:13: Error: Failed to connect to the adapter (Error: SRSP - SYS - ping after 6000ms)
    at ZStackAdapter.<anonymous> (/zigbee2mqtt-1.13.0/node_modules/zigbee-herdsman/dist/adapter/z-stack/adapter/zStackAdapter.js:71:23)
npm ERR! code ELIFECYCLE
    at Generator.throw (<anonymous>)
    at rejected (/zigbee2mqtt-1.13.0/node_modules/zigbee-herdsman/dist/adapter/z-stack/adapter/zStackAdapter.js:6:65)
npm ERR! errno 1
npm
 ERR! zigbee2mqtt@1.13.0 start: `node index.js`
npm ERR! Exit status 1
npm ERR! 
npm ERR! Failed at the zigbee2mqtt@1.13.0 start script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.
``` 

```
zigbee2mqtt:info  2020-05-03 18:09:06: Starting zigbee2mqtt version 1.13.0 (commit #unknown)
zigbee2mqtt:info  2020-05-03 18:09:06: Starting zigbee-herdsman...
zigbee2mqtt:error 2020-05-03 18:09:13: Error while starting zigbee-herdsman
zigbee2mqtt:error 2020-05-03 18:09:13: Failed to start zigbee
zigbee2mqtt:error 2020-05-03 18:09:13: Exiting...
zigbee2mqtt:error 2020-05-03 18:09:13: Error: Failed to connect to the adapter (Error: SRSP - SYS - ping after 6000ms)
    at ZStackAdapter.<anonymous> (/zigbee2mqtt-1.13.0/node_modules/zigbee-herdsman/dist/adapter/z-stack/adapter/zStackAdapter.js:71:23)
npm ERR! code ELIFECYCLE
    at Generator.throw (<anonymous>)
    at rejected (/zigbee2mqtt-1.13.0/node_modules/zigbee-herdsman/dist/adapter/z-stack/adapter/zStackAdapter.js:6:65)
npm ERR! errno 1
npm
 ERR! zigbee2mqtt@1.13.0 start: node index.js
npm ERR! Exit status 1
npm ERR! 
npm ERR! Failed at the zigbee2mqtt@1.13.0 start script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.
```

nexro567, 2020-05-03:
I solved the problem for 50%. 

1. I made a backup of my files in the config map. 
2. Did a reset in Zigbee2MQTTAssistant -> status -> Reset Local State. 
3. After that I reset the CC2531 sniffer. You find two buttons. The button nearby the USB port is the reset button.
4. After that I removed the docker image and container. Restart the server and made a new docker image and container. Put the old config lines without devices in the new configuration.yaml file.

The problem, after the steps above I experiencing now is when I restart the server that the zigbee2mqtt container fails after a minute. After the container has failed for a minute I can restart it. After that I have no more problems anymore till I restart the server.. 

The fault after a reset is the same error. An npm error..

I hope it will work again with the steps above. 
**After the steps above I need to pair all devices again. Home Assistant knows the addresses but are not available till the sniffer has paired all devices.**

nexro567, 2020-05-10:
Switch back to 1.12.2, your problem will be solved.

Terminal:
docker pull koenkk/zigbee2mqtt:1.12.2

Repair all devices.

trashkpi, 2020-05-21:
Got the same issue on Hassio

nexro567, 2020-05-21:
I found also a fault in my configuration of docker. Check your docker container HomeAssistant or another container where you use your serial device of zigbee, because your serial device can be used by only one container.

closed 2020-07-20 by aureliosaraiva