Koenkk/zigbee2mqtt #3488: Innr SP120 configuring error message on boot

vdbrink, 2020-05-03:
# Bug Report
## What happened
When I start zigbee2mqtt I get the next error messages about my Innr SP120 socket while booting. This is in the 1.13.0 version but also 1.12 but I didn't saw it in a much older version.
The socket itself just works normally as expected.

```
MQTT publish: topic 'zigbee2mqtt/innrsmartplug1', payload '{"state":"OFF","linkquality":57,"power":0,"voltage":231,"current":0}'
zigbee npm[3915]: zigbee2mqtt:info   Configuring 'innrsmartplug1'
zigbee npm[3915]: zigbee2mqtt:error  Failed to configure 'innrsmartplug1', attempt 2 (Error: Read 0x00158d000342b933/1 haElectricalMeasurement(["acVoltageMultiplier","acVoltageDivisor","acCurrentMultiplier"], {"timeout":10000,"manufacturerCode":null,"disableDefaultResponse":true}) failed (Error: Status 'UNSUPPORTED_ATTRIBUTE')
at Endpoint.<anonymous> (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/controller/model/endpoint.js:182:23)
at Generator.next (<anonymous>)
at fulfilled (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/controller/model/endpoint.js:5:58))
```

## What did you expect to happen
No error at boot time for the configuration of the Innr sp120.

## How to reproduce it (minimal and precise)
Just start zigbee2mqtt with a innr SP120 in the list of devices. 

## Debug Info
Zigbee2mqtt version: 1.13.0 (also 1.12.x)
Adapter hardware: CC2531
Adapter firmware version: 20190608

## Impact
Very low

Koenkk, 2020-05-04:
Fix will be available in the latest dev branch in a few hours. Thanks for reporting.

closed 2020-05-04 by Koenkk

macmeck, 2020-05-20:
@Koenkk , can you tell into which release this fix will find its way?

Koenkk, 2020-05-21:
@macmeck 1.13.1 (released yesterday)

macmeck, 2020-05-23:
@Koenkk , great work, thanks! seems to be working for me after the upgrade to 1.13.1