Koenkk/zigbee2mqtt #3490: Tradfri devices cannot pair with z2m using the deconz stack "unsupported_attribute"

hutchinsane, 2020-05-04:
When trying to pair any Ikea/Tradfri Zigbee devices (Tried: Symfonisk sound controller, Tradfri On/Off switch, Tradfri light remote) the pairing process errors out during configuration with "unsupported attribute" and the devices cannot be used. 

Additionally, the z2m logs show a bunch of "no converters available" messages. The logs attached are specifically for the Symfonisk controller.

The corresponding logs can be found at:
https://pastebin.com/vGw1d41K

Koenkk, 2020-05-04:
Besides the error, does it report battery?

hutchinsane, 2020-05-05:
It does not finish the pairing process and never appears in my device list so I would assume not. Or do you mean in general? Because pairing with the CC2531 works and there it does report battery level.

Koenkk, 2020-05-06:
On line 63 it does `zigbee2mqtt:info  2020-05-04 06:06:56: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"pairing","message":"interview_successful","meta":{"friendly_name":"0xec1bbdfffe398315","model":"E1744","vendor":"IKEA","description":"SYMFONISK sound controller","supported":true}}'`. Only the configure fails therefore I ask if it reports battery.

hutchinsane, 2020-05-07:
My apologies, I assumed that configure unsuccessful was equivalent to pairing unsuccessful. I've just repaired it and waited for a while, and it doesn't seem to reporting any battery status values, neither to the logs nor to the MQTT topic.

Koenkk, 2020-05-07:
@hutchinsane the error is conbee specific, please post the log here so the team can take a look at it: https://github.com/Koenkk/zigbee-herdsman/issues/72

ChrisHae, 2020-05-08:
Please try the new version of the deconz adapter. I uploaded some fixes. Please tell if it worked.

hutchinsane, 2020-05-08:
@Koenkk Alright, thank you very much, I'll post the logs in the correct thread!
@ChrisHae Hi, I just reinstalled z2m and the issue still seems to persist, unfortunately!

closed 2020-05-08 by hutchinsane

palitu, 2020-05-24:
is there a reason this was closed?  I am having the same issue.

palitu, 2020-05-25:
@hutchinsane ?
is there a reason this was closed? I am having the same issue.

ChrisHae, 2020-05-26:
Do you have retested it with the latest version of the dev-branch? With the fix i uploaded 14 days ago the unsupported attribute error should be gone.