Koenkk/zigbee2mqtt #3498: vibration sensor DJT11LM refresh time

kitopopo, 2020-05-05:
Dear friend,

I have done the installation to close my awnings based on the value of strenght, everything works correctly. I only see that the refresh time of the strenght value is 300 seconds and in 300 seconds many gusts of wind can occur and can damage it.

Could the update time of the strenght value be cut in half?
I've been looking at the configuration files fromZigbee.js and
devices.js. I have seen constants of 300 seconds, 3600 .. But I prefer to ask first.

Is possible change the refresh time in the vibration sensor DJT11LM? This time depends on zigbee2mqtt or xiaomi sensor?

## What did you expect to happen

It would be interesting to be able to vary the refresh time of each device depending on the needs of each sensor

## How to reproduce it (minimal and precise)

making a vibration every 5 minutes only change the strenght value

## Debug Info
Zigbee2mqtt version:1.12.2-dev
Adapter hardware: CC2531
Adapter firmware version: DEFAULT_20190608

SargonofAssyria, 2020-05-06:
Are you sure that that can be set? Is that not a fixed value in the DJT11LM?

Koenkk, 2020-05-06:
AFAIK it's not possible to change the interval of strength reporting (it's something in the firmware of the DJT11LM)

kitopopo, 2020-05-06:
Tnank dear friend.

kitopopo, 2020-05-06:
> ¿Estás seguro de que eso se puede configurar? ¿No es ese un valor fijo en el DJT11LM?
> Entonces se deben usar los diferentes ángulos. Se envían casi de inmediato.
> O solo el ángulo único solo. Eso podría ser suficiente.
> Según el manual, este dispositivo tiene 3 modos.
> `1．SecurityMode. Monitor the door and window switch status. 2．KnockMode. Control other smart devices via wireless connection. 3．BedMode. Monitor the person's bed activity, to help determine user’s quality of sleep`

Interesting, Sorry, I read the answer too late before closing the Issue.  So should we get in touch again with @ stas-demydiuk or could @mr koenkk modify it?

kitopopo, 2020-05-06:
> > ¿Estás seguro de que eso se puede configurar? ¿No es ese un valor fijo en el DJT11LM? 
> > Entonces se deben usar los diferentes ángulos. Se envían casi de inmediato. 
> > O solo el ángulo único solo. Eso podría ser suficiente. 
> > Según el manual, este dispositivo tiene 3 modos.
> > `1．SecurityMode. Monitor the door and window switch status. 2．KnockMode. Control other smart devices via wireless connection. 3．BedMode. Monitor the person's bed activity, to help determine user’s quality of sleep`
> 
> Interesante, lo siento, leí la respuesta demasiado tarde antes de cerrar el número.
Every minute or so I have verified that a new detected vibration can occur and note that the value of strenght is always the same for 5 minutes, but I did not notice that the value of the angles is changing, I will check it.
Entonces, ¿deberíamos ponernos en contacto nuevamente con @stas-demydiuk  o podría @Koenkk modificarlo?

kitopopo, 2020-05-06:
Hi, i see always the same values during the first 5 minutes, Do you obatin differents values?

```
may 06 12:07:42 raspberrypi npm[10993]: zigbee2mqtt:info  2020-05-06 12:07:42: MQTT publish: topic 'zigbee2mqtt/0x00158d0002a51c0a', payload '{"battery":100,"voltage":3025,"linkquality":115,"angle":5,"angle_x":27,"angle_y":-43,"angle_z":35,"angle_x_absolute":63,"angle_y_absolute":133,"strength":14,"action":"vibration"}'
may 06 12:09:19 raspberrypi npm[10993]: zigbee2mqtt:info  2020-05-06 12:09:19: MQTT publish: topic 'zigbee2mqtt/0x00158d0002a51c0a', payload '{"battery":100,"voltage":3025,"linkquality":115,"angle":5,"angle_x":27,"angle_y":-43,"angle_z":35,"angle_x_absolute":63,"angle_y_absolute":133,"strength":14,"action":"vibration"}'
may 06 12:10:27 raspberrypi npm[10993]: zigbee2mqtt:info  2020-05-06 12:10:27: MQTT publish: topic 'zigbee2mqtt/0x00158d0002a51c0a', payload '{"battery":100,"voltage":3025,"linkquality":115,"angle":5,"angle_x":27,"angle_y":-43,"angle_z":35,"angle_x_absolute":63,"angle_y_absolute":133,"strength":28}'
```

> ¿Estás seguro de que eso se puede configurar? ¿No es ese un valor fijo en el DJT11LM?
> Entonces se deben usar los diferentes ángulos. Se envían casi de inmediato.
> O solo el ángulo único solo. Eso [...]

Hi, i see always the same values during the first 5 minutes, Do you obatin differents values?

```
may 06 12:07:42 raspberrypi npm[10993]: zigbee2mqtt:info  2020-05-06 12:07:42: MQTT publish: topic 'zigbee2mqtt/0x00158d0002a51c0a', payload '{"battery":100,"voltage":3025,"linkquality":115,"angle":5,"angle_x":27,"angle_y":-43,"angle_z":35,"angle_x_absolute":63,"angle_y_absolute":133,"strength":14,"action":"vibration"}'
may 06 12:09:19 raspberrypi npm[10993]: zigbee2mqtt:info  2020-05-06 12:09:19: MQTT publish: topic 'zigbee2mqtt/0x00158d0002a51c0a', payload '{"battery":100,"voltage":3025,"linkquality":115,"angle":5,"angle_x":27,"angle_y":-43,"angle_z":35,"angle_x_absolute":63,"angle_y_absolute":133,"strength":14,"action":"vibration"}'
may 06 12:10:27 raspberrypi npm[10993]: zigbee2mqtt:info  2020-05-06 12:10:27: MQTT publish: topic 'zigbee2mqtt/0x00158d0002a51c0a', payload '{"battery":100,"voltage":3025,"linkquality":115,"angle":5,"angle_x":27,"angle_y":-43,"angle_z":35,"angle_x_absolute":63,"angle_y_absolute":133,"strength":28}'
```

kitopopo, 2020-05-06:
I have read that by making a hardware modification, a solder on the motion sensor modifies this refresh value to obtain almost every minute refresh of information, perhaps the same could be done in this sensor but for this sensor I did not find information

SargonofAssyria, 2020-05-06:
If you want the angle, which is send almost immediately, in Domotizc then you should contact  @stas-demydiuk 
Or you could use MQTT to get the value in Domoticz.
For a good wind detection the sensor should be placed in such a way that the angle changes by movement.

kitopopo, 2020-05-06:
You are right, I have found that by fully opening the awning and closing it afterwards the angle values ​​change in less than 1 minute

`may 06 13:08:21 raspberrypi npm[10993]: zigbee2mqtt:info  2020-05-06 13:08:21: MQTT publish: topic 'zigbee2mqtt/0x00158d0002a51c0a', payload '{"battery":100,"voltage":3025,"linkquality":115,"angle":5,"angle_x":6,"angle_y":-78,"angle_z":11,"angle_x_absolute":84,"angle_y_absolute":168,"strength":83,"action":"vibration"}'`

`may 06 13:08:55 raspberrypi npm[10993]: zigbee2mqtt:info  2020-05-06 13:08:55: MQTT publish: topic 'zigbee2mqtt/0x00158d0002a51c0a', payload '{"battery":100,"voltage":3025,"linkquality":115,"angle":5,"angle_x":6,"angle_y":-78,"angle_z":11,"angle_x_absolute":84,"angle_y_absolute":168,"strength":83,"action":"tilt"}'`

`may 06 13:09:05 raspberrypi npm[10993]: zigbee2mqtt:info  2020-05-06 13:09:05: MQTT publish: topic 'zigbee2mqtt/0x00158d0002a51c0a', payload '{"battery":100,"voltage":3025,"linkquality":115,"angle":5,"angle_x":11,"angle_y":-62,"angle_z":25,"angle_x_absolute":79,"angle_y_absolute":152,"strength":83}'`

`may 06 13:09:12 raspberrypi npm[10993]: zigbee2mqtt:info  2020-05-06 13:09:12: MQTT publish: topic 'zigbee2mqtt/0x00158d0002a51c0a', payload '{"battery":100,"voltage":3025,"linkquality":115,"angle":5,"angle_x":11,"angle_y":-62,"angle_z":25,"angle_x_absolute":79,"angle_y_absolute":152,"strength":83,"action":"tilt"}'`

`may 06 13:09:12 raspberrypi npm[10993]: zigbee2mqtt:info  2020-05-06 13:09:12: MQTT publish: topic 'zigbee2mqtt/0x00158d0002a51c0a', payload '{"battery":100,"voltage":3025,"linkquality":115,"angle":5,"angle_x":14,"angle_y":-57,"angle_z":29,"angle_x_absolute":76,"angle_y_absolute":147,"strength":83}'`

`may 06 13:09:38 raspberrypi npm[10993]: zigbee2mqtt:info  2020-05-06 13:09:38: MQTT publish: topic 'zigbee2mqtt/0x00158d0002a51c0a', payload '{"battery":100,"voltage":3025,"linkquality":115,"angle":5,"angle_x":14,"angle_y":-57,"angle_z":29,"angle_x_absolute":76,"angle_y_absolute":147,"strength":83,"action":"tilt"}'
`
`may 06 13:09:39 raspberrypi npm[10993]: zigbee2mqtt:info  2020-05-06 13:09:39: MQTT publish: topic 'zigbee2mqtt/0x00158d0002a51c0a', payload '{"battery":100,"voltage":3025,"linkquality":115,"angle":5,"angle_x":27,"angle_y":-43,"angle_z":35,"angle_x_absolute":63,"angle_y_absolute":133,"strength":83}'
`

I think that the angle value is not enough to detect a lot or little wind,i am not sure but I think that for this the strenght value is better. My question is that if the angle values ​​are updated for less than 1 minute why the strenght value doesn't? who can solve this?

kitopopo, 2020-05-06:
Dear friends,

Mr @koenkk , now te strenghts values are refreshing every minute. I don,t understand anything.
have you made any modifications? thanks in advanced

`may 06 13:33:25 raspberrypi npm[10993]: zigbee2mqtt:info  2020-05-06 13:33:25: MQTT publish: topic 'zigbee2mqtt/0x00158d0002a51c0a', payload '{"battery":100,"voltage":3035,"linkquality":115,"angle":5,"angle_x":6,"angle_y":-78,"angle_z":11,"angle_x_absolute":84,"angle_y_absolute":168,"strength":103}'`

`may 06 13:33:41 raspberrypi npm[10993]: zigbee2mqtt:info  2020-05-06 13:33:41: MQTT publish: topic 'zigbee2mqtt/0x00158d0002a51c0a', payload '{"battery":100,"voltage":3035,"linkquality":115,"angle":5,"angle_x":6,"angle_y":-78,"angle_z":11,"angle_x_absolute":84,"angle_y_absolute":168,"strength":103,"action":"vibration"}'
`

`may 06 13:34:25 raspberrypi npm[10993]: zigbee2mqtt:info  2020-05-06 13:34:25: MQTT publish: topic 'zigbee2mqtt/0x00158d0002a51c0a', payload '{"battery":100,"voltage":3035,"linkquality":115,"angle":5,"angle_x":6,"angle_y":-78,"angle_z":11,"angle_x_absolute":84,"angle_y_absolute":168,"strength":68}'
`

`may 06 13:34:45 raspberrypi npm[10993]: zigbee2mqtt:info  2020-05-06 13:34:45: MQTT publish: topic 'zigbee2mqtt/0x00158d0002a51c0a', payload '{"battery":100,"voltage":3035,"linkquality":115,"angle":5,"angle_x":6,"angle_y":-78,"angle_z":11,"angle_x_absolute":84,"angle_y_absolute":168,"strength":68,"action":"vibration"}'
`

`may 06 13:35:25 raspberrypi npm[10993]: zigbee2mqtt:info  2020-05-06 13:35:25: MQTT publish: topic 'zigbee2mqtt/0x00158d0002a51c0a', payload '{"battery":100,"voltage":3035,"linkquality":115,"angle":5,"angle_x":6,"angle_y":-78,"angle_z":11,"angle_x_absolute":84,"angle_y_absolute":168,"strength":70}'
`

`may 06 13:35:45 raspberrypi npm[10993]: zigbee2mqtt:info  2020-05-06 13:35:45: MQTT publish: topic 'zigbee2mqtt/0x00158d0002a51c0a', payload '{"battery":100,"voltage":3035,"linkquality":115,"angle":5,"angle_x":6,"angle_y":-78,"angle_z":11,"angle_x_absolute":84,"angle_y_absolute":168,"strength":70,"action":"vibration"}'
`

`may 06 13:36:24 raspberrypi npm[10993]: zigbee2mqtt:info  2020-05-06 13:36:24: MQTT publish: topic 'zigbee2mqtt/0x00158d0002a51c0a', payload '{"battery":100,"voltage":3035,"linkquality":115,"angle":5,"angle_x":6,"angle_y":-78,"angle_z":11,"angle_x_absolute":84,"angle_y_absolute":168,"strength":69}'
`

`may 06 13:36:53 raspberrypi npm[10993]: zigbee2mqtt:info  2020-05-06 13:36:53: MQTT publish: topic 'zigbee2mqtt/0x00158d0002a51c0a', payload '{"battery":100,"voltage":3035,"linkquality":115,"angle":5,"angle_x":6,"angle_y":-78,"angle_z":11,"angle_x_absolute":84,"angle_y_absolute":168,"strength":69,"action":"vibration"}'
`

`may 06 13:37:24 raspberrypi npm[10993]: zigbee2mqtt:info  2020-05-06 13:37:24: MQTT publish: topic 'zigbee2mqtt/0x00158d0002a51c0a', payload '{"battery":100,"voltage":3035,"linkquality":115,"angle":5,"angle_x":6,"angle_y":-78,"angle_z":11,"angle_x_absolute":84,"angle_y_absolute":168,"strength":72}'
`

Koenkk, 2020-05-06:
I didn't change anything, so probably the sensor has some different behaviour then we initially thought.

kitopopo, 2020-05-16:
Hi dear @koenkk,

Yesterday I updated to the latest version of raspbian buster, I saw that the latest version of npm was also updated, then I also updated to the latest dev branch of zigbe2mqtt (I don't know if any new dev branch will have come out these days but i did the update anyway). 
Since I performed all these steps my xiaomi switches now work correctly and without errors. What a mistery. 
Anyway when the new zzh stick is available I will buy it since it is much more powerful and everything will work better. Thanks for your help. a cordial greeting

kitopopo, 2020-05-22:
Dear @Koenkk ,

If the refresh of the strenght value information depends on the xiaomi firmware, I can add 2 vibration sensors on the awning. Do you know if both of them would refresh the strenght value at the same time? Or does this vary depending on when i pair the sensor with zigbe2mqtt? that is, I would like to obtain the strenght every 2.5 minutes, would it be possible if I add a sensor to zigbee2mqtt first and after 2.5 minutes I add the other sensor?

Koenkk, 2020-05-22:
@kitopopo I don't know what the timing behaviour will be. Only way to find this out is to try.

kitopopo, 2020-05-22:
Ok, thanks again. Best regards

closed 2020-05-22 by Koenkk