Koenkk/zigbee2mqtt #3499: OOM Killed on QNAP332X ContainerStation since zigbee2mqtt 1.9

kabakakao, 2020-05-05:
# Bug Report
## What happened
I am running zigbee2mqtt 1.8 in my QNAP ContainerStation. This works great with the official docker Container.
But when I use any newer Version of zigbee2mqtt container it directly crashes with following output:
`Using '/app/data' as data directory                                                                                                                                                                                            
                                                                                                                                                                                                                               
<--- Last few GCs --->                                                                                                                                                                                                         
                                                                                                                                                                                                                               
                                                                                                                                                                                                                               
<--- JS stacktrace --->                                                                                                                                                                                                        
                                                                                                                                                                                                                               
                                                                                                                                                                                                                               
#                                                                                                                                                                                                                              
# Fatal process OOM in insufficient memory to create an Isolate                                                                                                                                                                
#  `

## How to reproduce it (minimal and precise)

I tried different configuration of RAM and CPU. But nothing is working for me.
I have 16GB of RAM installed in my QNAP

## Debug Info
Zigbee2mqtt version: >1.8
Hardware: QNAP 332X (ARM) (16GB RAM)

kabakakao, 2020-06-03:
No one with an answer or a tip :(
Are there chances to check where the error comes from? When  googling around, it seems to come from nodejs. But I'm not sure what else to check

Koenkk, 2020-06-03:
- Make sure to check with 1.13.1 first
- If that doesn't help provide debug logging/herdsman debug logging. https://www.zigbee2mqtt.io/information/debug.html

kabakakao, 2020-06-03:
I tried 1.13.1 already
Now, I tried to activate debug-logging. But no log-file was created :(

closed 2020-07-26 by stale[bot]