Koenkk/zigbee2mqtt #3501: orvibo thermostat

ARoozitalab, 2020-05-05:
hi,I want to use orvibo thermostat FCU_ACCP with hassio and zigbee2mqtt.
 in debug state and changing thermostat parameter i have these commands:

```
zigbee2mqtt:warn  2020-04-30 12:57:10: Received message from unsupported device with Zigbee model '9eca5b60791f40549e7059ae6771c2f4'
zigbee2mqtt:warn  2020-04-30 12:57:10: Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html.
zigbee2mqtt:debug 2020-04-30 12:58:08: Device '0x00124b000adca903' announced itself
zigbee2mqtt:info  2020-04-30 12:58:08: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"device_announced","message":"announce","meta":{"friendly_name":"0x00124b000adca903"}}'
zigbee2mqtt:debug 2020-04-30 12:59:01: Received Zigbee message from '0x00124b000aeaef8a', type 'attributeReport', cluster 'hvacThermostat', data '{"localTemp":2730}' from endpoint 1 with groupID 0

zigbee2mqtt:debug 2020-04-30 13:01:30: Received Zigbee message from '0x00124b000aeaef8a', type 'attributeReport', cluster 'hvacFanCtrl', data '{"fanMode":1}' from endpoint 1 with groupID 0
zigbee2mqtt:debug 2020-04-30 13:01:41: Received Zigbee message from '0x00124b000aeaef8a', type 'attributeReport', cluster 'hvacFanCtrl', data '{"fanMode":2}' from endpoint 1 with groupID 0

zigbee2mqtt:debug 2020-04-30 13:02:09: Received Zigbee message from '0x00124b000aeaef8a', type 'attributeReport', cluster 'hvacThermostat', data '{"runningMode":4}' from endpoint 1 with groupID 0
zigbee2mqtt:debug 2020-04-30 13:02:25: Received Zigbee message from '0x00124b000aeaef8a', type 'attributeReport', cluster 'hvacThermostat', data '{"runningMode":3}' from endpoint 1 with groupID 0
zigbee2mqtt:debug 2020-04-30 13:02:37: Received Zigbee message from '0x00124b000aeaef8a', type 'attributeReport', cluster 'hvacThermostat', data '{"runningMode":6}' from endpoint 1 with groupID 0

zigbee2mqtt:debug 2020-04-30 13:03:10: Received Zigbee message from '0x00124b000aeaef8a', type 'attributeReport', cluster 'genOnOff', data '{"onOff":0}' from endpoint 1 with groupID 0
zigbee2mqtt:debug 2020-04-30 13:03:33: Received Zigbee message from '0x00124b000aeaef8a', type 'attributeReport', cluster 'genOnOff', data '{"onOff":1}' from endpoint 1 with groupID 0

zigbee2mqtt:debug 2020-04-30 13:04:16: Received Zigbee message from '0x00124b000aeaef8a', type 'attributeReport', cluster 'hvacThermostat', data '{"occupiedCoolingSetpoint":1200}' from endpoint 1 with groupID 0
zigbee2mqtt:debug 2020-04-30 13:04:16: Received Zigbee message from '0x00124b000aeaef8a', type 'attributeReport', cluster 'hvacThermostat', data '{"occupiedCoolingSetpoint":1300}' from endpoint 1 with groupID 0
zigbee2mqtt:debug 2020-04-30 13:04:11: Received Zigbee message from '0x00124b000aeaef8a', type 'attributeReport', cluster 'hvacThermostat', data '{"occupiedCoolingSetpoint":1100}' from endpoint 1 with groupID 0

zigbee2mqtt:info  2020-04-30 13:21:03: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"device_announced","message":"announce","meta":{"friendly_name":"0x00124b000adca903"}}'
zigbee2mqtt:debug 2020-04-30 13:21:59: Received Zigbee message from '0x00124b000aeaef8a', type 'attributeReport', cluster 'hvacThermostat', data '{"localTemp":2630}' from endpoint 1 with groupID 0
```
so i add these line to device.js 👍 

```
{
    zigbeeModel: ['9eca5b60791f40549e7059ae6771c2f4'],
    model: 'FCU_ACCP',
    vendor: 'Orvibo',
    description: 'orvibo thermostat', 
    supports: 'temperature , heating/cooling system control,fan', 
    fromZigbee: [fz.on_off,fz.thermostat_att_report, fz.generic_fan_mode], 
    toZigbee: [tz.on_off,
	tz.factory_reset, tz.thermostat_local_temperature_calibration,
	tz.thermostat_occupancy, tz.thermostat_occupied_heating_setpoint, tz.thermostat_unoccupied_heating_setpoint,
	tz.thermostat_occupied_cooling_setpoint, tz.thermostat_unoccupied_cooling_Setpoint,
	tz.thermostat_setpoint_raise_lower, tz.thermostat_remote_sensing,
	tz.thermostat_control_sequence_of_operation, tz.thermosat_sysem_mode, tz.thermostat_weekly_schedule,
	tz.thermostat_clear_weekly_schedule, tz.thermostat_relay_status_log,
	tz.thermostat_temperature_setpoint_hold, tz.hermostat_temperature_Setpoint_hold_duration, tz.fan_mode,
	],
    meta:{configurekey: 1},
    configure: async (device, coordinatorEndpoint) => {
	const endpoint = device.getEndpoint(1);
	await bind(Endpoint, coordinatorEndpoint, ['hvacFanCtrl', 'hvacThermostat', 'genOnOff', 'hvacThermostat']);
	await configureReporting.thermostatTemprature(endpoint);
	await configureReporting.thermostatSystemMode(endpoint);
	await configureReporting.thermostatOccupiedHeatingSetpoint(endpoint);
	await configureReporting.thermostatUnoccupiedHeatingSetpoint(endpoint);
	await configureReporting.thermostatOccupiedCoolingSetpoint(endpoint);
	await configureReporting.thermostatUnoccupiedCoolingSetpoint(endpoint);
	await configureReporting.fanMode(endpoint);
       },
},
```

after adding my device indevice.js  i have all parameter  and very thing is ok

```
#----------------------------------------------------thermostat
  - platform: "mqtt"
    name : "TERMO TEMP"
    state_topic: "zigbee2mqtt/0x00124b000aeaef8a"
    availability_topic: "zigbee2mqtt/bridge/state"
    icon: "mdi:signal"
    unit_of_measurement: "C"
    value_template: "{{ value_json.local_temperature }}"
    
  - platform: "mqtt"
    name : "TERMO cooling TEMP"
    state_topic: "zigbee2mqtt/0x00124b000aeaef8a"
    availability_topic: "zigbee2mqtt/bridge/state"
    icon: "mdi:signal"
    unit_of_measurement: "C"
    value_template: "{{ value_json.occupied_cooling_setpoint }}"
    
  - platform: "mqtt"
    name : "fan mode"
    state_topic: "zigbee2mqtt/0x00124b000aeaef8a"
    availability_topic: "zigbee2mqtt/bridge/state"
    icon: "mdi:signal"
    unit_of_measurement: ""
    value_template: "{{ value_json.fan_mode }}"
    
  - platform: "mqtt"
    name : "fan state"
    state_topic: "zigbee2mqtt/0x00124b000aeaef8a"
    availability_topic: "zigbee2mqtt/bridge/state"
    icon: "mdi:signal"
    unit_of_measurement: ""
    value_template: "{{ value_json.fan_state }}"  
    
  - platform: "mqtt"
    name : "running mode"
    state_topic: "zigbee2mqtt/0x00124b000aeaef8a"
    availability_topic: "zigbee2mqtt/bridge/state"
    icon: "mdi:signal"
    unit_of_measurement: ""
    value_template: "{{ value_json.running_mode }}"    
```

but I can't set  AC parameter by Zigbee . for example I cant set fan mode or running mode or ...
I used this code but it doesn't worked
```
climate:
  - platform: "mqtt"
    availability_topic: "zigbee2mqtt/bridge/state"
    min_temp: "7"
    max_temp: "30"
    modes: 
      - "heat"
      - "cool"
      - "precoling"
    mode_state_topic: true
    mode_state_template: "{{ value_json.running_mode }}"
    mode_command_topic: true
    current_temperature_topic: true
    current_temperature_template: "{{ value_json.local_temperature }}"
    temperature_state_topic: true
    temperature_state_template: "{{ value_json.occupied_cooling_setpoint }}"
    temperature_command_topic: "occupied_heating_setpoint"
    temp_step: 1
    action_topic: true
    action_template: "{{ value_json.running_mode }}"
    fan_modes: 
      - "auto"
      - "low"
      - "medium"
      - "high"
    fan_mode_command_topic: true
    fan_mode_state_topic: true
    fan_mode_state_template: "{{ value_json.fan_mode }}"
```

Koenkk, 2020-05-06:
- For setting  fan mode use `zigbee2mqtt/FRIENDLY_NAME/set` with payload `{"fan_mode": "medium"}`.
- Running mode cannot be set currently
- Cooling temp use the same as point 1 with payload `{"occupied_cooling_setpoint": 10}`

ARoozitalab, 2020-05-07:
thanks
I changed climate src:

```
climate:
  - platform: "mqtt"
    availability_topic: "zigbee2mqtt/bridge/state"
    min_temp: "7"
    max_temp: "30"
    modes: 
      - "heat"
      - "cool"
      - "precoling"
    mode_state_topic: true
    mode_state_template: "{{ value_json.running_mode }}"
    mode_command_topic: true
    current_temperature_topic: true
    current_temperature_template: "{{ value_json.local_temperature }}"
    temperature_state_topic: true
    temperature_state_template: "{{ value_json.occupied_cooling_setpoint }}"
    temperature_command_topic: "occupied_heating_setpoint"
    temp_step: 1
    action_topic: true
    action_template: "{{ value_json.running_mode }}"
    fan_modes: 
      - "{\"fan_mode\": \"auto\"}"
      - "{\"fan_mode\": \"low\"}"
      - "{\"fan_mode\": \"medium\"}"
      - "{\"fan_mode\": \"high\"}"
    fan_mode_command_topic: "zigbee2mqtt/0x00124b000aeaef8a/set"
    fan_mode_state_topic: "zigbee2mqtt/0x00124b000aeaef8a"
    fan_mode_state_template: "{{ value_json.fan_mode }}" 
```

but it doesn't work and i have this logs  when i change fan mode:

```
zigbee2mqtt:error 2020-05-07 15:17:22: Failed to call 'EntityPublish' 'onMQTTMessage' (TypeError: Cannot read property 'key' of undefined
    at /app/lib/extension/publish.js:163:56
    at Array.find (<anonymous>)
    at EntityPublish.onMQTTMessage (/app/lib/extension/publish.js:163:42)
    at Controller.callExtensionMethod (/app/lib/controller.js:348:44)
    at Controller.onMQTTMessage (/app/lib/controller.js:242:14)
    at MQTT.emit (events.js:223:5)
    at MQTT.onMessage (/app/lib/mqtt.js:95:14)
    at MqttClient.emit (events.js:223:5)
    at MqttClient._handlePublish (/app/node_modules/mqtt/lib/client.js:1257:12)
    at MqttClient._handlePacket (/app/node_modules/mqtt/lib/client.js:401:12))
```

Koenkk, 2020-05-07:
`zigbee2mqtt:error 2020-05-07 15:17:22: Failed to call 'EntityPublish' 'onMQTTMessage' (TypeError: Cannot read property 'key' of undefined`

please provide the debug logging of what occured before this.

To enable debug logging set in `configuration.yaml`:
```yaml
advanced:
  log_level: debug
```

ARoozitalab, 2020-05-08:
i set log_level to debug
for fan mode :
```
zigbee2mqtt:debug 2020-05-07 17:05:07: Received MQTT message on 'zigbee2mqtt/0x00124b000aeaef8a/set' with data '{"fan_mode": "medium"}'
zigbee2mqtt:error 2020-05-07 17:05:07: Failed to call 'EntityPublish' 'onMQTTMessage' (TypeError: Cannot read property 'key' of undefined
    at /app/lib/extension/publish.js:163:56
    at Array.find (<anonymous>)
    at EntityPublish.onMQTTMessage (/app/lib/extension/publish.js:163:42)
    at Controller.callExtensionMethod (/app/lib/controller.js:348:44)
    at Controller.onMQTTMessage (/app/lib/controller.js:242:14)
    at MQTT.emit (events.js:223:5)
    at MQTT.onMessage (/app/lib/mqtt.js:95:14)
    at MqttClient.emit (events.js:223:5)
    at MqttClient._handlePublish (/app/node_modules/mqtt/lib/client.js:1257:12)
    at MqttClient._handlePacket (/app/node_modules/mqtt/lib/client.js:401:12))
```
for cooling set point

```
zigbee2mqtt:debug 2020-05-07 17:08:17: Received MQTT message on 'zigbee2mqtt/0x00124b000aeaef8a/set' with data '{"occupied_cooling_setpoint": 18}'
zigbee2mqtt:debug 2020-05-07 17:08:17: Publishing 'set' 'occupied_cooling_setpoint' to '0x00124b000aeaef8a'
zigbee2mqtt:error 2020-05-07 17:08:17: Publish 'set' 'occupied_cooling_setpoint' to '0x00124b000aeaef8a' failed: 'Error: Write 0x00124b000aeaef8a/1 hvacThermostat({"occupiedCoolingSetpoint":1800}, {"timeout":10000,"manufacturerCode":null,"disableDefaultResponse":true}) failed (Error: Status 'INVALID_DATA_TYPE')'
zigbee2mqtt:debug 2020-05-07 17:08:17: Error: Write 0x00124b000aeaef8a/1 hvacThermostat({"occupiedCoolingSetpoint":1800}, {"timeout":10000,"manufacturerCode":null,"disableDefaultResponse":true}) failed (Error: Status 'INVALID_DATA_TYPE')
    at Endpoint.<anonymous> (/app/node_modules/zigbee-herdsman/dist/controller/model/endpoint.js:158:23)
    at Generator.next (<anonymous>)
    at fulfilled (/app/node_modules/zigbee-herdsman/dist/controller/model/endpoint.js:5:58)
zigbee2mqtt:info  2020-05-07 17:08:17: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"zigbee_publish_error","message":"Publish 'set' 'occupied_cooling_setpoint' to '0x00124b000aeaef8a' failed: 'Error: Write 0x00124b000aeaef8a/1 hvacThermostat({\"occupiedCoolingSetpoint\":1800}, {\"timeout\":10000,\"manufacturerCode\":null,\"disableDefaultResponse\":true}) failed (Error: Status 'INVALID_DATA_TYPE')'","meta":{"friendly_name":"0x00124b000aeaef8a"}}'

```

Koenkk, 2020-05-08:
This occurs because one of your `toZigbee` converters does not exists, so specific in `toZigbee: []` but it's not in `toZigbee.js`.

closed 2020-07-18 by stale[bot]