Koenkk/zigbee2mqtt #3502: Docker and USR TCP232 - CC2530 dont connect

dossidr, 2020-05-05:
Hello Experts,

My Environment:
Synology DS918+ --> Docker --> latest --> USR TCP232 --> configuration.yaml:
`permit_join: true
mqtt:
  base_topic: zigbee2mqtt
  server: 'mqtt://192.168.178.44'
serial:
  port: 'tcp://192.168.178.4:20108'
`
USR Settings:
Seriel Port:

`Baude Rate = 115200
Data Size = 8
Parity = none
Stop Bits = 1
Local Port Number = 20108
Remote Port Number = 1775
Work Mode = TCP Client
Remote Server Address = 192.168.178.44
RESET = 
LINK = 1
INDEX =
Similar RFC2217 = 
`
Has anybody any idea how can connect the device behind?
The Device is an CC2530 

Best Regards.............

kabakakao, 2020-05-06:
Can you try to connect the Port out of zigbee2mqtt?

dossidr, 2020-05-06:
Yes connection from WIN10 client using USR VCOM Toolchain is possible. ASCII Data Ouput is possible. I want send picture later. Currently im away frome home. 
Best regards...

closed 2020-07-12 by stale[bot]