Koenkk/zigbee2mqtt #3508: QBCZ11LM Red Blinking

to4ko, 2020-05-07:
# Bug Report
## What happened
QBCZ11LM started slow red blinking once paired with Z2M (2-3 times every minute). While they were on MI gateway 2 - solid blue LED only.

## What did you expect to happen
Have no RED blinking

## How to reproduce it (minimal and precise)

## Debug Info
Zigbee2mqtt version: 1.13
Adapter hardware: CC2538
Adapter firmware version: {'type': 'zStack30x', 'meta': {'transportrev': 2, 'product': 2, 'majorrel': 2, 'minorrel': 7, 'maintrel': 2, 'revision': 20200211}}

to4ko, 2020-05-10:
looks like this is the case 

zigbee2mqtt:error 2020-05-10 23:36:12: Failed to execute LQI for 'Socket_PC_MB'

any ideas what can be done?

Koenkk, 2020-05-11:
Looks like connection issue, make sure to check https://www.zigbee2mqtt.io/how_tos/how_to_improve_network_range.html

to4ko, 2020-05-11:
Its really strange as this specofic socket located in 3m from cc2538 stick...

to4ko, 2020-05-12:
> Looks like connection issue, make sure to check https://www.zigbee2mqtt.io/how_tos/how_to_improve_network_range.html

once re-paired - issue is gone. closing.

to4ko, 2020-08-17:
issue is still there (((( pairing same socket\plug to test zigbee network (only couple of devices) resolves the issue... 

@Koenkk  chief is it possiblt that z2m simply flooded with messages (145 devices on single usb stick) and missing some communication with devices?

Koenkk, 2020-08-17:
Could be the case, I'm not sure what the reason it has to start flashing red. Are you able to sniff the traffic at the moment it starts flashing red? (from the solid color)

to4ko, 2020-08-28:
@Koenkk so the sniff results for xiaomi zigbee plug (short network address 0x2897). It's working fine. No issues. Red LED blinks not all the time (like when there is no connection) - it blinks in series (8 red blinks then pause for approx 15-20 sec and repeat). This is matching this 
[image]

and red blinks at this command which is suppressed all the time

[image]

please advise why this commands can be supressed?

full sniff attached
[xiaomi_red_led blinks.pcapng.gz](https://github.com/Koenkk/zigbee2mqtt/files/5144271/xiaomi_red_led.blinks.pcapng.gz)

Koenkk, 2020-08-29:
What do you mean with "supressed" ?

to4ko, 2020-08-29:
> What do you mean with "supressed" ?

yeah there are some issues with my English. sorry.

i'm just reading from sniff. 

[image]

Koenkk, 2020-08-29:
Could you make a similar sniff with the Xiaomi gateway so we can see the differences?

closed 2020-10-01 by to4ko