Koenkk/zigbee2mqtt #3510: RTCGQ11LM reports "occupancy=false" instead of "true"

mr2c12, 2020-05-07:
- Z2m 1.13.0
- Coordinator CC1352P2
- 33 devices (both routers and end devices)

I currently have 3 RTCGQ11LM in three very different positions (one is 3 mt from the coordinator, 2 upstair and connect throught a CC2530+cc2591 router.

I suffer from this problem: like once a week, the motion fails to trigger the correct automation in Home Assistant. This problem occurs in any of the RTCGQ11LM installed.
It's really difficult to reproduce and quite frustrating actually.

Today I was able to get it on the z2m log at least!

### 20:50 triggered (correct) - 20:52 end of motion (correct)

> info  2020-05-07 20:50:38: MQTT publish: topic 'zigbee2mqtt/0x00158d00045161f4', payload '{"battery":100,"voltage":3005,"illuminance":0,"linkquality":96,"last_seen":"2020-05-07T19:50:38.315Z","occupancy":true,"illuminance_lux":0}'
> info  2020-05-07 20:52:08: MQTT publish: topic 'zigbee2mqtt/0x00158d00045161f4', payload '{"battery":100,"voltage":3005,"illuminance":0,"linkquality":96,"last_seen":"2020-05-07T19:50:38.315Z","occupancy":false,"illuminance_lux":0}'

### 20:53 end of motion (WRONG!!) - 20:55 end of motion (correct)

> info  2020-05-07 20:53:41: MQTT publish: topic 'zigbee2mqtt/0x00158d00045161f4', payload '{"battery":100,"voltage":3005,"illuminance":0,"linkquality":90,"last_seen":"2020-05-07T19:53:41.721Z","occupancy":false,"illuminance_lux":0}'
> info  2020-05-07 20:55:27: MQTT publish: topic 'zigbee2mqtt/0x00158d00045161f4', payload '{"battery":100,"voltage":3005,"illuminance":0,"linkquality":96,"last_seen":"2020-05-07T19:55:27.356Z","occupancy":false,"illuminance_lux":0}'

### 20:55 triggered (correct) - 20:56 end of motion (correct)

> info  2020-05-07 20:55:27: MQTT publish: topic 'zigbee2mqtt/0x00158d00045161f4', payload '{"battery":100,"voltage":3005,"illuminance":0,"linkquality":93,"last_seen":"2020-05-07T19:55:27.394Z","occupancy":true,"illuminance_lux":0}'
> info  2020-05-07 20:56:57: MQTT publish: topic 'zigbee2mqtt/0x00158d00045161f4', payload '{"battery":100,"voltage":3005,"illuminance":0,"linkquality":93,"last_seen":"2020-05-07T19:55:27.394Z","occupancy":false,"illuminance_lux":0}'

(I can provide the entire log, but it does not seem really helpful. Just a bunch of messages from other devices and not a the same time of the "problematic" message occurred at 2020-05-07 20:53:41.

I have really spent a lot of time in the recorded issue for this sensor.
I doubt this is related to #2274, #1498 since this happens a few times a week and, as you can see, it magically turn back to normal at the very next motion triggered. It never required me to re-pair the device. Waiting for motion to retrigger is enough.
Also #2830 seems unrelated, because there is not a illuminance change in those messages.

I understand that I should provide a zigbee sniffing log but it's VERY HARD to get it.
A z2m log set to debug level could be of any more help? (since it is so much easier)

Useful info:
- this problem occurs for any of my motion sensors (not a faulty unit)
- it used to happen even with older versions of z2m 
- coordinator fw has been updated

Can you please help me to understand what is going on?

Ton1965, 2020-05-08:
Still it looks quite close to #2274 to me.

Just look at the log you sent more attentively. Two correct message pair have something in common: both occupancy start and occupancy clear have same last_seen attribute although the messages themselves occurred at different time. This is explainable since occupancy clear is not really obtained from the device, this message is sent by z2m after the specified timeout so last_seen is not updated.

The incorrect pair is different, the log entries have both different message time and different last_seen values. This means that both messages were generated in response to zigbee commands sent from the device. 

Also in both correct message pairs all other attributes are the same and in the wrong pair link_quality field is different. 

This resembles very much what I used to have. My explanation is that the device sends each attribute change as a separate zigbee message. z2m converts each such message into an mqtt message with the full set of attributes according to HA requirements. Sometimes zigbee messages go in quick succession and either z2m or coordinator as a physical device loose some of them. This results in an a wrong sequence of mqtt messages being sent to HA.

Looks like the real culprit is not z2m but rather the coordinator firmware. Since the time I updated my C1352P2 firmware to the latest version available on the dev branch I have not observed a single failure.

Try to reflash your coordinator, it is simple and does not require re-pairing of the devices.

mr2c12, 2020-05-08:
Thanks @Ton1965 , will do for sure (I am already on dev firmware, version 12/03/2020).
Will update again a few days after the new firmware has been flashed.

Ton1965, 2020-05-08:
Actually I am on the same firmware, so now I am confused. I attributed that behavior change to the newer firmware version and now I do not know what to think about.

I have several devices (contact sensors and occupancy sensors). Most of them connect through a router, some are attached directly to the coordinator. I had problems with both types and initially thought the problem was related to some radio interference. Firmware change was like a last resort but it did really help. I remember though that I erased the previous data at least twice when flashing but I do not think it is important.

mr2c12, 2020-05-08:
Pretty much the same setup.
My 4 RTCGQ11LM suffer from this problem even if they are placed in very different situation (as I stated above, one is very close to the coordinator, one really far).

About radio interferences: I had huge problem during the past week due to the new DSL router and my WiFi on channel 11 (while my zigbee network is on channel 25 or 26). I fixed switchin wifi to channel 6. But I got the error with RTCGQ11LM while the interference has been solved. And I didn't think about it because the message arrived somehow to the coordinator.

Anyway, since it is really an easy procedure, I will update my coordinator firmware and try to "erase previous data".

Koenkk, 2020-05-08:
It's likely the same as #2274 (to confirm I would need to have debug logging)

To enable debug logging set in `configuration.yaml`:
```yaml
advanced:
  log_level: debug
```

mr2c12, 2020-05-08:
@Koenkk loglevel set to debug via mqtt message

If it is related to #2274, I should get two messages ("occupancy=false"-"occupancy=true") in short sequence, is it correct?
Because, as reported above, I see just one message in my log.

Will try to capture the problem with log set to debug and report here.

mr2c12, 2020-05-11:
Closing this issue since it's strictly related to #2274. I will continue there with some log and debug stuff.

closed 2020-05-11 by mr2c12