Koenkk/zigbee2mqtt #3511: Xiaomi curtain position

trdb33, 2020-05-08:
Hello I'm using zigbee2mqtt with a lot with Xiaomi products, and noticed that the Xiaomi Curtain is reporting his position only when it's triggered, the state of the curtain is unknown after each Hass reboot and curtain automation can't be triggered since I'm using curtain position in automations conditions
I don't think the Zigbee2mqtt and hass versions are important because I have the probleme for more than four months with a lot of updates
To sum : Hass don't know curtain position after each reboot

I hope the issue is clear, thank you ;)

Koenkk, 2020-05-08:
Make sure to correctly setup according to https://www.zigbee2mqtt.io/integration/home_assistant.html#mqtt-discovery

trdb33, 2020-05-09:
Hey I added 
```
  birth_message:
    topic: 'hass/status'
    payload: 'online'
  will_message:
    topic: 'hass/status'
    payload: 'offline'
```
to mqtt section and now the issue is resolved, thank you ;)

closed 2020-05-09 by Koenkk