Koenkk/zigbee2mqtt #3513: support for new device plug

zhapwey, 2020-05-08:
Hi, I'm trying to get the settings out of a new device.
If someone could help me, because I am trying to test with other configurations of other smart plug devices but it does not work.

It seems to have been synchronized correctly, but the ping does not work correctly.

I copy the database.db of the device.
{"id":3,"type":"Router","ieeeAddr":"0x0015bc002f003826","nwkAddr":34644,"manufId":6548,"manufName":"Gewiss","powerSource":"Mains (single phase)","modelId":"GWA1526_SmartPlug","epList":[1,2],"endpoints":{"1":{"profId":49353,"epId":1,"devId":1,"inClusterList":[5,6],"outClusterList":[],"clusters":{},"binds":[{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0x00124b0008bdfc77","endpointID":1}]},"2":{"profId":260,"epId":2,"devId":81,"inClusterList":[0,1794,3,9,2820,6,4,5,2],"outClusterList":[0,25,10,3,1030],"clusters":{"genBasic":{"attributes":{"modelId":"GWA1526_SmartPlug","manufacturerName":"Gewiss","powerSource":1,"zclVersion":1,"dateCode":"2017-06-30 04:50"}}},"binds":[]}},"dateCode":"2017-06-30 04:50","zclVersion":1,"interviewCompleted":true,"meta":{},"lastSeen":1588939554333}

thanks,

sjorge, 2020-05-08:
Try adding this to devices.js

```
    {
        zigbeeModel: ['GWA1526_SmartPlug'],
        model: 'GWA1526',
        description: 'SmartPlug',
        supports: 'on/off',
        vendor: 'Gewiss',
        fromZigbee: [fz.on_off],
        toZigbee: [tz.on_off],
    },
```

That should at least get on/off working, hopefully.

zhapwey, 2020-05-11:
zigbee2mqtt:error 2020-05-11 09:44:10: Publish 'set' 'state' to '0x0015bc002f003826' failed: 'Error: Command 0x0015bc002f003826/1 genOnOff.off({}, {"timeout":6000,"manufacturerCode":null,"disableDefaultResponse":false}) failed (Error: Timeout - 34644 - 1 - 2 - 6 - 11 after 6000ms)'
zigbee2mqtt:debug 2020-05-11 09:44:10: Error: Command 0x0015bc002f003826/1 genOnOff.off({}, {"timeout":6000,"manufacturerCode":null,"disableDefaultResponse":false}) failed (Error: Timeout - 34644 - 1 - 2 - 6 - 11 after 6000ms)
    at Endpoint.<anonymous> (/opt/zigbee/node_modules/zigbee-herdsman/dist/controller/model/endpoint.js:344:23)
    at Generator.throw (<anonymous>)
    at rejected (/opt/zigbee/node_modules/zigbee-herdsman/dist/controller/model/endpoint.js:6:65)
zigbee2mqtt:info  2020-05-11 09:44:10: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"zigbee_publish_error"","message":"Publish 'set' 'state' to '0x0015bc002f003826' failed: 'Error: Command 0x0015bc002f003826/1 genOnOff.off({}, {\"timeout\":6000,\"manufacturerCode\":null,\"disableDefaultResponse\":false}) failed (Error: Timeout - 34644 - 1 - 2 - 6 - 11 after 6000ms)'","meta":{"friendly_name":"0x0015bc002f003826"}}'
zigbee2mqtt:info  2020-05-11 09:44:29: MQTT publish: topic 'zigbee2mqtt/0x0015bc002f003826/availability', payload 'offline'
zigbee2mqtt:error 2020-05-11 09:44:29: Failed to ping '0x0015bc002f003826'

I already tried that and also:

    {
        zigbeeModel: ['GWA1526_SmartPlug'],
        model: 'GWA1526_SmartPlug',
        description: 'Smart+ plug',
        supports: 'on/off',
        vendor: 'GEWIS',
        fromZigbee: [fz.on_off, fz.generic_power],
        toZigbee: [tz.on_off],
        meta: {configureKey: 3},
        configure: async (device, coordinatorEndpoint) => {
            const endpoint = device.getEndpoint(1);
            await bind(endpoint, coordinatorEndpoint, ['genOnOff', 'seMetering']);
            await configureReporting.onOff(endpoint);
            await configureReporting.instantaneousDemand(endpoint);
            await endpoint.read('seMetering', ['multiplier', 'divisor']);
        },
    },

But nothing doesn't work,

Thanks,

closed 2020-07-18 by stale[bot]