Koenkk/zigbee2mqtt #3514: Xiaomi Aqara double wall switch (WXKG02LM) sends false (double) events

guardiande, 2020-05-08:
# Bug Report
## What happened

Sometimes clicking on the buttons of the Xiaomi Aqara double wall switch (WXKG02LM) cause multiple events to arrive in z2m.

Here's a debug log (the left button was bouncing, the right click is known good example):

```
debug 2020-05-08 20:12:19: Received Zigbee message from 'double-switch-1', type 'attributeReport', cluster 'genMultistateInput', data '{"presentValue":1}' from endpoint 1 with groupID 0
info  2020-05-08 20:12:19: MQTT publish: topic 'zigbee2mqtt/double-switch-1', payload '{"battery":100,"voltage":3195,"linkquality":34,"click":"left"}'
debug 2020-05-08 20:12:19: Received Zigbee message from 'double-switch-1', type 'attributeReport', cluster 'genMultistateInput', data '{"presentValue":1}' from endpoint 1 with groupID 0
info  2020-05-08 20:12:19: MQTT publish: topic 'zigbee2mqtt/double-switch-1', payload '{"battery":100,"voltage":3195,"linkquality":34,"click":"left"}'
debug 2020-05-08 20:12:20: Received Zigbee message from 'double-switch-1', type 'attributeReport', cluster 'genMultistateInput', data '{"presentValue":1}' from endpoint 1 with groupID 0
info  2020-05-08 20:12:20: MQTT publish: topic 'zigbee2mqtt/double-switch-1', payload '{"battery":100,"voltage":3195,"linkquality":34,"click":"left"}'
debug 2020-05-08 20:12:20: Received Zigbee message from 'double-switch-1', type 'attributeReport', cluster 'genMultistateInput', data '{"presentValue":1}' from endpoint 1 with groupID 0
info  2020-05-08 20:12:20: MQTT publish: topic 'zigbee2mqtt/double-switch-1', payload '{"battery":100,"voltage":3195,"linkquality":31,"click":"left"}'
debug 2020-05-08 20:12:28: Received Zigbee message from 'double-switch-1', type 'attributeReport', cluster 'genMultistateInput', data '{"presentValue":1}' from endpoint 2 with groupID 0
info  2020-05-08 20:12:28: MQTT publish: topic 'zigbee2mqtt/double-switch-1', payload '{"battery":100,"voltage":3195,"linkquality":34,"click":"right"}'
```

## What did you expect to happen

One event per button press in z2m.

## How to reproduce it (minimal and precise)

Repeatedly hit the buttons and watch the log until you notice multiple events appearing in z2m instead of just one.

## Debug Info
Zigbee2mqtt version: 1.12.2 (commit #911cd58)
Adapter hardware: CC2531
Adapter firmware version: {"type":"zStack12","meta":{"transportrev":2,"product":0,"majorrel":2,"minorrel":6,"maintrel":3,"revision":20190619}}

guardiande, 2020-05-08:
Just double checked against 1.13.0 - same result.

Koenkk, 2020-05-09:
Looking at your log, it seems that this is what the switch actually sends this (multiple `Received Zigbee message from`). 

I'm not sure if we can do much about it from the z2m side, sniffing the traffic could confirm where the issue comes from. https://www.zigbee2mqtt.io/how_tos/how_to_sniff_zigbee_traffic.html

guardiande, 2020-05-09:
Here's a false event that I recorded, alternating between left and right button presses one at a time:

```
info  2020-05-09 17:19:34: MQTT publish: topic 'zigbee2mqtt/double-switch-1', payload '{"battery":100,"voltage":3175,"linkquality":34,"click":"left"}'
info  2020-05-09 17:19:39: MQTT publish: topic 'zigbee2mqtt/double-switch-1', payload '{"battery":100,"voltage":3175,"linkquality":34,"click":"right"}'
info  2020-05-09 17:19:42: MQTT publish: topic 'zigbee2mqtt/double-switch-1', payload '{"battery":100,"voltage":3175,"linkquality":34,"click":"left"}'
info  2020-05-09 17:19:42: MQTT publish: topic 'zigbee2mqtt/double-switch-1', payload '{"battery":100,"voltage":3175,"linkquality":31,"click":"left"}'
```

As you can see there are two click at 17:19:42 and the recording looks like the device is really sending multiple clicks:

[image]

And here's the sequence of packets that relate to the click in question:

**The first four look completely identical to me going from `0x02f8` to `0xcdaa`**

[image]

[image]

[image]

[image]

**This packet seems identical with the exception of the source and destination now going from `0xcdaa` to `0x0000`**

[image]

**Now the `Counter` value changes from `76` to `77`. I guess this is the button release event.**

[image]

[image]

guardiande, 2020-05-09:
Also wondering why all three presses result in different packet counts:

First press -> 3
Second press -> 2
Third press -> 7

Koenkk, 2020-05-09:
Could you share the pcapng file?

guardiande, 2020-05-09:
Sure :)

[WXKG02LM.zip](https://github.com/Koenkk/zigbee2mqtt/files/4604294/WXKG02LM.zip)

Koenkk, 2020-05-11:
It looks that indeed the button transmits multiple messages (maybe an issue with the firmware of the button?). However I can think we can work around it easily.

In `fromZigbee.js` change the corresponding converter (https://github.com/Koenkk/zigbee-herdsman-converters/blob/367567b875654dd9c79303d46e619f69a7527ba8/converters/fromZigbee.js#L1464) to

```js
WXKG02LM_click_multistate: {
    cluster: 'genMultistateInput',
    type: ['attributeReport', 'readResponse'],
    convert: (model, msg, publish, options, meta) => {
        const last = store[msg.device.ieeeAddr];
        const current = msg.meta.zclTransactionSequenceNumber;
        if (last !== current) return;
        store[msg.device.ieeeAddr] = current;

        const button = getKey(model.endpoint(msg.device), msg.endpoint.ID);
        const value = msg.data['presentValue'];

        const actionLookup = {
            0: 'long',
            1: null,
            2: 'double',
        };

        const action = actionLookup[value];

        if (button) {
            return {click: button + (action ? `_${action}` : '')};
        }
    },
},
```

guardiande, 2020-05-11:
Looking good! Thank you!

```
if (last !== current) return;
```

has to be

```
if (last == current) return;
```

though :)

Koenkk, 2020-05-12:
Good catch, added the fix. Will be available in the dev branch later today. Assuming this can be closed.

closed 2020-05-12 by Koenkk

Anthael, 2021-08-13:
Hi
I have the same issue from button WXKG11LM with 2 events per click.
May be possible to handle a new state ie 'none' ?
And after the click is released, revert the state to none. 
After the first click, the state always stayed to Single.