Koenkk/zigbee2mqtt #3516: Xiaomi\Aqara plugs\sockets Blue LED Control

to4ko, 2020-05-08:
# Bug Report
## What happened
There is a Blue LED on Aqara\Xiomi plugs\sockets. This led can be switched off via MI Home application. Would be great to have this option via Zigbee2MQTT

## What did you expect to happen
Have availability to turn ON\OFF Blue LED on Aqara\Xiaomi plugs\sockets

## How to reproduce it (minimal and precise)

## Debug Info
Zigbee2mqtt version: 1.13
Adapter hardware: CC2538
Adapter firmware version:

Koenkk, 2020-05-09:
Could you sniff the traffic of the Xiaomi bridge when setting this option? If you post the log here together with the network key of your Xiaomi bridge I can add it.

https://www.zigbee2mqtt.io/how_tos/how_to_sniff_zigbee_traffic.html

closed 2020-07-18 by stale[bot]

ZZJHONS, 2022-12-08:
Does anybody know if this was implemented? Else I would need to buy a sniffer and an aqara hub to log this