Koenkk/zigbee2mqtt #3518: cache_state as device specific option

mr2c12, 2020-05-09:
This is more a "feature request" than an "issue".

At the moment, "cache_state" is a global settings, i.e. every device is affected.
Would it be possible to make it per-device?
Given that this breaks the integration with home assistant, I would like to change it only for one sensor and leave the others as they are.

Koenkk, 2020-05-11:
What is the use case for this?

mr2c12, 2020-05-11:
I'm suffering from the problem with RTCGQ11LM, where two consecutives messages (illuminance, occupancy) cause false negatives (i.e. no motion is triggered).
I read in some related issues that you could disable caching, but partial json state messages will broke Home Assistant integration.

**If I could** enable cache only for RTCGQ11LM, I could then use some nodered automation to populate appropriates mqtt topics.
The idea is that, from {"illuminance": "10", "occupancy": true} I could create and populate

- mymotionsensor/illuminance
- mymotionsensor/occupancy

I understand this scenario is just a "very bad hack" to get the sensor working reliably (and I know the problem is with the sensor-z2m interaction or sort of).
But at the moment, if I disable caching, I have make some nodered automation for every sensor I own (>30).

Maybe an overkill at this point. But, if it is not difficult to get, it could be useful.
I am sorry but I can't think of another reasonable use case.
I'll let you decide whether to consider it useful or not.

Koenkk, 2020-05-11:
Let's first see what comes out of the original issue as this option does seem to be very specific.

closed 2020-07-18 by stale[bot]