Koenkk/zigbee2mqtt #3520: Problem booting zigbee2mqtt after upgrade

estebanz01, 2020-05-10:
# Bug Report
## What happened
After upgrading to zigbee2mqtt to latest version, I get the following error:
```
May 10 04:29:07 raspberrypi npm[18546]: zigbee2mqtt:info  2020-05-10 04:29:07: Logging to console and directory: '/opt/zigbee2mqtt/data/log/2020-05-10.04-29-07' filename: log.txt
May 10 04:29:10 raspberrypi npm[18546]: zigbee2mqtt:info  2020-05-10 04:29:10: Starting zigbee2mqtt version 1.13.0 (commit #55e6283)
May 10 04:29:10 raspberrypi npm[18546]: zigbee2mqtt:info  2020-05-10 04:29:10: Starting zigbee-herdsman...
May 10 04:29:14 raspberrypi npm[18546]: zigbee2mqtt:info  2020-05-10 04:29:14: zigbee-herdsman started
May 10 04:29:14 raspberrypi npm[18546]: zigbee2mqtt:info  2020-05-10 04:29:14: Coordinator firmware version: '{"type":"zStack12","meta":{"transportrev":2,"product":0,"majorrel":2,"minorrel":6,"maintrel":3,"revision":20190608}}'
May 10 04:29:14 raspberrypi npm[18546]: zigbee2mqtt:info  2020-05-10 04:29:14: Currently 2 devices are joined:
May 10 04:29:14 raspberrypi npm[18546]: zigbee2mqtt:error 2020-05-10 04:29:14: Failed to call 'Configure' 'onZigbeeStarted' (TypeError: zigbeeHerdsmanConverters.findByDevice is not a function
May 10 04:29:14 raspberrypi npm[18546]:     at Zigbee.resolveEntity (/opt/zigbee2mqtt/lib/zigbee.js:226:54)
May 10 04:29:14 raspberrypi npm[18546]:     at Configure.onZigbeeStarted (/opt/zigbee2mqtt/lib/extension/configure.js:70:48)
May 10 04:29:14 raspberrypi npm[18546]:     at Controller.callExtensionMethod (/opt/zigbee2mqtt/lib/controller.js:348:44))
May 10 04:29:14 raspberrypi npm[18546]: zigbee2mqtt:error 2020-05-10 04:29:14: Failed to call 'OnEvent' 'onZigbeeStarted' (TypeError: zigbeeHerdsmanConverters.findByDevice is not a function
May 10 04:29:14 raspberrypi npm[18546]:     at Zigbee.resolveEntity (/opt/zigbee2mqtt/lib/zigbee.js:226:54)
May 10 04:29:14 raspberrypi npm[18546]:     at OnEvent.onZigbeeStarted (/opt/zigbee2mqtt/lib/extension/onEvent.js:10:48)
May 10 04:29:14 raspberrypi npm[18546]:     at Controller.callExtensionMethod (/opt/zigbee2mqtt/lib/controller.js:348:44))
May 10 04:29:14 raspberrypi npm[18546]: (node:18598) UnhandledPromiseRejectionWarning: TypeError: zigbeeHerdsmanConverters.findByDevice is not a function
May 10 04:29:14 raspberrypi npm[18546]:     at Zigbee.resolveEntity (/opt/zigbee2mqtt/lib/zigbee.js:226:54)
May 10 04:29:14 raspberrypi npm[18546]:     at Controller.start (/opt/zigbee2mqtt/lib/controller.js:120:40)
May 10 04:29:14 raspberrypi npm[18546]: (Use `node --trace-warnings ...` to show where the warning was created)
May 10 04:29:14 raspberrypi npm[18546]: (node:18598) UnhandledPromiseRejectionWarning: Unhandled promise rejection. This error originated either by throwing inside of an async function without a catch block, or by rejecting a promise which was not handled with .catch(). To terminate the node process on unhandled promise rejection, use the CLI flag `--unhandled-rejections=strict` (see https://nodejs.org/api/cli.html#cli_unhandled_rejections_mode). (rejection id: 1)
May 10 04:29:14 raspberrypi npm[18546]: (node:18598) [DEP0018] DeprecationWarning: Unhandled promise rejections are deprecated. In the future, promise rejections that are not handled will terminate the Node.js process with a non-zero exit code.
```

And zigbee2mqtt never boots up. I changed zigbee2mqtt to 1.12.2 and it works as expected.

I'm not sure if I'm on the latest firmware version as the firmware table does not match information as how it is reported on MQTT, so apologies on advance for that.

## What did you expect to happen

Zigbee2mqtt to work as expected.

## How to reproduce it (minimal and precise)
* Run zigbee2mqtt on 1.12.2 with a CC2531 dongle.
* Pair a device.
* Upgrade to 1.13.1.
**Expected behaviour:** Zigbee2mqtt runs and acknowledge the paired device and reports data.
**Current behaviour:** Zigbee2mqtt fails to boot.

## Debug Info
Zigbee2mqtt version: 1.13.1
Adapter hardware: CC2531
Adapter firmware version:
* type: zStack12
* meta:
  * transportrev: 2
  * product: 0
  * majorrel: 2
  * minorrel: 6
  * maintrel: 3
  * revision: 20190608

Koenkk, 2020-05-11:
This is probably related to an incompatible version of zigbee-herdsman-converters (could be modified when you e.g. tried to add support for a new device). If that's the case make sure the correct version is installed. Can be done by executing `npm ci` in the Zigbee2mqtt directory (if running bare metal) or by removing the zigbee-herdsman-converters volume when running through docker.

estebanz01, 2020-05-11:
Yep, that fixed it. My bad! thanks for the help.

closed 2020-05-11 by estebanz01