Koenkk/zigbee2mqtt #3521: Xiaomi A6121 Smart Lock Support

olili, 2020-05-10:
I plan to buy such a device ( https://www.zigbee2mqtt.io/devices/A6121.html )

Question: is it fully supported by Z2M? 

I spent a look into the converters and only found some code for receiving data and not for sending data to the device.  So I assume that sending the command to lock/unlock is not yet implemented. Is that correct?
What is the problem? Will there be the chance to overcome these limitations in the upcoming future?

Thx in advance,

angeck, 2020-05-10:
Attention le A6121 ne permet pas de vérouiller/dévérouiller.

On peut uniquement avoir le statut de clé insérée, clé non reconnue et clé perdue.

Ce n'est pas une limitation de Z2M mais le produit qui est fait ainsi, c'est pareil sur l'application XIAOMI HOME.

Koenkk, 2020-05-11:
Unlocking/locking from zigbee2mqtt indeed does not seem to be supported for now.

If I remember correctly @dzungpv initially added this device. Do you know the reason why this was not added?

dzungpv, 2020-05-11:
> Unlocking/locking from zigbee2mqtt indeed does not seem to be supported for now.
> 
> If I remember correctly @dzungpv initially added this device. Do you know the reason why this was not added?

Those lock are only sensor, so just report  lock/unlock state. So it not posible to lock/unlock remotely.

closed 2020-07-18 by stale[bot]

667bdrm, 2020-09-12:
Today disassembled that lock, looks like there are no servo motor for lock/unlock. There are only solenoid for jam lock to forbid open by mechanical key. Here is a pinout of my electrical cylinder variant.

1 - black   - key cpu data/vcc (DC +3.27v)
2 - green  - cylinder movement contact switch 1
3 - white   - cylinder movement contact switch 2
4 - red      - cylinder key opening jam (DC +3.27v)
5 - yellow - ground

2-3 shortened when cylinder is moved out of neutral position.

Here is a third party disassembled photos https://fccid.io/2AP8E-A6121/Internal-Photos/Internal-Photos-4033605

Hope that helps someone.

osousa, 2020-10-20:
> 1 - black - key cpu data/vcc (DC +3.27v)

Hey,

do you think its possible to lock/unlock by changing the solenoid position and ignoring the codes from keys?

667bdrm, 2020-10-23:
> do you think its possible to lock/unlock by changing the solenoid position and ignoring the codes from keys?

Currently I seen only the hardware modification way. You could disconnect key data contacts, Don't remember if this works with original firmware. Currently I see only the way to connect the cylinder to the own controller (esp8266 or similar), so you could set +3.27 voltage on solenoid pin and it will jam the cylinder. I didn't analyzed controller protocol that connected to the TYZS5 and used for communication between keys and solenoid, so can't tell if you could just replace firmware for TYZS5 (EFR32,  https://www.silabs.com/wireless/zigbee).

osousa, 2020-10-23:
> > do you think its possible to lock/unlock by changing the solenoid position and ignoring the codes from keys?
> 
> Currently I see only the hardware modification way. You could disconnect key data contacts, Don't remember if this works with original firmware. Currently I see only the way to connect the cylinder to the own controller (esp8266 or similar), so you could set +3.27 voltage on solenoid pin and it will jam the cylinder. I didn't analyzed controller protocol that connected to the TYZS5 and used for communication between keys and solenoid, so can't tell if you could just replace firmware for TYZS5 (EFR32, https://www.silabs.com/wireless/zigbee).

Thank you for the info. It works indeed with a transistor controlling the solenoid. It closes and opens the lock.

If anyone is interested in exchanging the TYZS5 for an ESP8285, it will work.