Koenkk/zigbee2mqtt #3523: Zemismart downlight

krasevych, 2020-05-10:
# Bug Report
## new light - Zemismart Model: LXT56-LS27LX1.6 dont support

## after success join it should work correctly

## get it:  https://www.aliexpress.com/item/4000012905509.html?spm=a2g0s.9042311.0.0.72dd4c4d2RUthw, and then try to join.

## Debug Info
log: `zigbee2mqtt:warn  2020-05-09 13:14:05: Received message from unsupported device with Zigbee model 'LXT56-LS27LX1.6'`
device info:
```
Zigbee Id is 0x00158d000448cbcb.
Address on Zigbee Network is 0x2F7A.
Link quality is 61/255.
Model is LXT56-LS27LX1.6 by 3A Smart Home DE.
Hardware version is 1.
Firmware version is 1000-0001.
This device type is Router
```

Koenkk, 2020-05-11:
Added, will be supported in the dev branch in a few hours.

closed 2020-05-11 by Koenkk

krasevych, 2020-05-11:
but I thought that it is `Zemismart` not Pry, is it?

Koenkk, 2020-05-11:
I already reverted that change in favor of your pr