Koenkk/zigbee2mqtt #3524: Aqara opple switch no release payload after long press > ~4 sec.

frostnatt, 2020-05-10:
# Bug Report
## What happened
Aqara Opple Switch (WXCJKG13LM) does not send a "release" payload after > ~4sec. long press.
Same issue was already reported here:
https://github.com/Koenkk/zigbee2mqtt/issues/2938

## What did you expect to happen

## How to reproduce it (minimal and precise)

## Debug Info
Zigbee2mqtt version: 1.13.0
Adapter hardware: CC1352P-2
Adapter firmware version: 20191106

IVI053, 2020-05-11:
Can confirm this issue.
But it looks like it can not be fixed in z2m as the switch does not send a message in that case. I verified this by enabling herdsman debug, but maybe somebody else should double check this.

< 4 seconds:
```
  zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,5,69,196,104,229,1,214] +6s
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,5,69,196,104,229,1,214] +1ms
  zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,32] +1ms
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,5,69,196,104,229,1,214,254,32] +0ms
  zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 5 - 2 - 5 - 196 - [104,229,1,214,254] - 32 +0ms
  zigbee-herdsman:adapter:zStack:znp:AREQ <-- ZDO - srcRtgInd - {"dstaddr":58728,"relaycount":1,"relaylist":[65238]} +6s
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [] +1ms
  zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,28,68,129,0,0,18,0] +45ms
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,28,68,129,0,0,18,0] +1ms
  zigbee-herdsman:adapter:zStack:unpi:parser <-- [104,229,2,1,0,102,0,166] +0ms
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,28,68,129,0,0,18,0,104,229,2,1,0,102,0,166] +1ms
  zigbee-herdsman:adapter:zStack:unpi:parser <-- [3,15,0,0,8,24,212,10,85,0,33,0,0,214,254,28,7] +1ms
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,28,68,129,0,0,18,0,104,229,2,1,0,102,0,166,3,15,0,0,8,24,212,10,85,0,33,0,0,214,254,28,7] +0ms
  zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 28 - 2 - 4 - 129 - [0,0,18,0,104,229,2,1,0,102,0,166,3,15,0,0,8,24,212,10,85,0,33,0,0,214,254,28] - 7 +0ms
  zigbee-herdsman:adapter:zStack:znp:AREQ <-- AF - incomingMsg - {"groupid":0,"clusterid":18,"srcaddr":58728,"srcendpoint":2,"dstendpoint":1,"wasbroadcast":0,"linkquality":102,"securityuse":0,"timestamp":983974,"transseqnumber":0,"len":8,"data":{"type":"Buffer","data":[24,212,10,85,0,33,0,0]}} +49ms
  zigbee-herdsman:controller:log Received 'zcl' data '{"frame":{"Header":{"frameControl":{"frameType":0,"manufacturerSpecific":false,"direction":1,"disableDefaultResponse":true},"transactionSequenceNumber":212,"manufacturerCode":null,"commandIdentifier":10},"Payload":[{"attrId":85,"dataType":33,"attrData":0}]},"address":58728,"endpoint":2,"linkquality":102,"groupID":0}' +6s
zigbee2mqtt:info  2020-05-11 14:49:52: MQTT publish: topic 'z2m/OGSZ-S-Switch-049', payload '{"linkquality":102,"last_seen":"2020-05-11T12:49:52.914Z","battery":100,"voltage":3100,"action":"button_2_hold"}'
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [] +6ms

  zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,5,69,196,104,229,1,214] +3s
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,5,69,196,104,229,1,214] +0ms
  zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,32] +1ms
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,5,69,196,104,229,1,214,254,32] +1ms
  zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 5 - 2 - 5 - 196 - [104,229,1,214,254] - 32 +0ms
  zigbee-herdsman:adapter:zStack:znp:AREQ <-- ZDO - srcRtgInd - {"dstaddr":58728,"relaycount":1,"relaylist":[65238]} +3s
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [] +0ms
  zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,28,68,129,0,0,18,0] +45ms 
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,28,68,129,0,0,18,0] +1ms
  zigbee-herdsman:adapter:zStack:unpi:parser <-- [104,229,2,1,0,102,0,37] +0ms
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,28,68,129,0,0,18,0,104,229,2,1,0,102,0,37] +1ms
  zigbee-herdsman:adapter:zStack:unpi:parser <-- [57,18,0,0,8,24,213,10,85,0,33,255,0,214,254,28,93] +0ms
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,28,68,129,0,0,18,0,104,229,2,1,0,102,0,37,57,18,0,0,8,24,213,10,85,0,33,255,0,214,254,28,93] +1ms
  zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 28 - 2 - 4 - 129 - [0,0,18,0,104,229,2,1,0,102,0,37,57,18,0,0,8,24,213,10,85,0,33,255,0,214,254,28] - 93 +0ms
  zigbee-herdsman:adapter:zStack:znp:AREQ <-- AF - incomingMsg - {"groupid":0,"clusterid":18,"srcaddr":58728,"srcendpoint":2,"dstendpoint":1,"wasbroadcast":0,"linkquality":102,"securityuse":0,"timestamp":1194277,"transseqnumber":0,"len":8,"data":{"type":"Buffer","data":[24,213,10,85,0,33,255,0]}} +49ms
  zigbee-herdsman:controller:log Received 'zcl' data '{"frame":{"Header":{"frameControl":{"frameType":0,"manufacturerSpecific":false,"direction":1,"disableDefaultResponse":true},"transactionSequenceNumber":213,"manufacturerCode":null,"commandIdentifier":10},"Payload":[{"attrId":85,"dataType":33,"attrData":255}]},"address":58728,"endpoint":2,"linkquality":102,"groupID":0}' +3s
zigbee2mqtt:info  2020-05-11 14:49:56: MQTT publish: topic 'z2m/OGSZ-S-Switch-049', payload '{"linkquality":102,"last_seen":"2020-05-11T12:49:56.278Z","battery":100,"voltage":3100,"action":"button_2_release"}'
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [] +6ms
```

\>= 4 seconds:
```
  zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,5,69,196,104,229,1,214] +4s
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,5,69,196,104,229,1,214] +0ms
  zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,32] +1ms
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,5,69,196,104,229,1,214,254,32] +0ms
  zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 5 - 2 - 5 - 196 - [104,229,1,214,254] - 32 +0ms
[... cut ...]
  zigbee-herdsman:controller:log Received 'zcl' data '{"frame":{"Header":{"frameControl":{"frameType":0,"manufacturerSpecific":false,"direction":1,"disableDefaultResponse":true},"transactionSequenceNumber":214,"manufacturerCode":null,"commandIdentifier":10},"Payload":[{"attrId":85,"dataType":33,"attrData":0}]},"address":58728,"endpoint":2,"linkquality":99,"groupID":0}' +4s
zigbee2mqtt:info  2020-05-11 14:50:00: MQTT publish: topic 'z2m/OGSZ-S-Switch-049', payload '{"linkquality":99,"last_seen":"2020-05-11T12:50:00.078Z","battery":100,"voltage":3100,"action":"button_2_hold"}'
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [] +6ms
```

golddragon007, 2020-06-24:
Hello,
My seller confirmed that this is the normal behaviour, just I don't know why it's not in the datasheet. As far as I measured 3 sec is the maximum which you can hold, if you hold it longer there's no release, you need to write another automation to stop after 3 sec. For me, it's also a bit sad that this device has a 1-sec delay therefore technically when you press&hold the button you need to release it within 4 sec.

closed 2020-08-01 by stale[bot]

Lazy8onez, 2022-06-23:
Are any updates about this issue?
Did someone find a workaround?