Koenkk/zigbee2mqtt #3525: After added Gledopto GL-C-008 2ID support device pairing fails

jussih, 2020-05-10:
# Bug Report
## What happened
After #3436 fixes landed to zigbee2mqtt dev, updated to edge version in Home Assistant. Zigbee2mqtt revision in my edge container is `fed84cfd5e9260564e58d03047b152b6fcdc3f80`.

- Removed test device from database.db
- Removed device from Home Assistant
- Started re-pairing device
- Pairing fails with the following log:

```
zigbee2mqtt:info  2020-05-10 20:43:38: Device '0x00124b001f79d7ea' joined
zigbee2mqtt:info  2020-05-10 20:43:38: Starting interview of '0x00124b001f79d7ea'
zigbee2mqtt:info  2020-05-10 20:43:38: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"device_connected","message":{"friendly_name":"0x00124b001f79d7ea"}}'
zigbee2mqtt:info  2020-05-10 20:43:38: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"pairing","message":"interview_started","meta":{"friendly_name":"0x00124b001f79d7ea"}}'
zigbee2mqtt:debug 2020-05-10 20:43:39: Device '0x00124b001f79d7ea' announced itself
zigbee2mqtt:info  2020-05-10 20:43:39: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"device_announced","message":"announce","meta":{"friendly_name":"0x00124b001f79d7ea"}}'
zigbee2mqtt:debug 2020-05-10 20:43:39: Device '0x00124b001f79d7ea' announced itself
zigbee2mqtt:info  2020-05-10 20:43:39: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"device_announced","message":"announce","meta":{"friendly_name":"0x00124b001f79d7ea"}}'
zigbee2mqtt:debug 2020-05-10 20:43:52: Received Zigbee message from '0x00124b001f79d7ea', type 'readResponse', cluster 'genBasic', data '{"modelId":"GL-C-007"}' from endpoint 11 with groupID 0
zigbee2mqtt:warn  2020-05-10 20:43:52: Received message from unsupported device with Zigbee model 'GL-C-007'
zigbee2mqtt:warn  2020-05-10 20:43:52: Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html.
zigbee2mqtt:debug 2020-05-10 20:43:52: Received Zigbee message from '0x00124b001f79d7ea', type 'readResponse', cluster 'genBasic', data '{"manufacturerName":"GLEDOPTO"}' from endpoint 11 with groupID 0
zigbee2mqtt:warn  2020-05-10 20:43:52: Received message from unsupported device with Zigbee model 'GL-C-007'
zigbee2mqtt:warn  2020-05-10 20:43:52: Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html.
zigbee2mqtt:debug 2020-05-10 20:43:52: Received Zigbee message from '0x00124b001f79d7ea', type 'readResponse', cluster 'genBasic', data '{"powerSource":1}' from endpoint 11 with groupID 0
zigbee2mqtt:warn  2020-05-10 20:43:52: Received message from unsupported device with Zigbee model 'GL-C-007'
zigbee2mqtt:warn  2020-05-10 20:43:52: Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html.
zigbee2mqtt:debug 2020-05-10 20:43:52: Received Zigbee message from '0x00124b001f79d7ea', type 'readResponse', cluster 'genBasic', data '{"zclVersion":1}' from endpoint 11 with groupID 0
zigbee2mqtt:warn  2020-05-10 20:43:52: Received message from unsupported device with Zigbee model 'GL-C-007'
zigbee2mqtt:warn  2020-05-10 20:43:52: Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html.
zigbee2mqtt:debug 2020-05-10 20:43:52: Received Zigbee message from '0x00124b001f79d7ea', type 'readResponse', cluster 'genBasic', data '{"appVersion":1}' from endpoint 11 with groupID 0
zigbee2mqtt:warn  2020-05-10 20:43:52: Received message from unsupported device with Zigbee model 'GL-C-007'
zigbee2mqtt:warn  2020-05-10 20:43:52: Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html.
zigbee2mqtt:debug 2020-05-10 20:43:52: Received Zigbee message from '0x00124b001f79d7ea', type 'readResponse', cluster 'genBasic', data '{"stackVersion":2}' from endpoint 11 with groupID 0
zigbee2mqtt:warn  2020-05-10 20:43:52: Received message from unsupported device with Zigbee model 'GL-C-007'
zigbee2mqtt:warn  2020-05-10 20:43:52: Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html.
zigbee2mqtt:debug 2020-05-10 20:43:52: Received Zigbee message from '0x00124b001f79d7ea', type 'readResponse', cluster 'genBasic', data '{"hwVersion":1}' from endpoint 11 with groupID 0
zigbee2mqtt:warn  2020-05-10 20:43:52: Received message from unsupported device with Zigbee model 'GL-C-007'
zigbee2mqtt:warn  2020-05-10 20:43:52: Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html.
zigbee2mqtt:debug 2020-05-10 20:43:52: Received Zigbee message from '0x00124b001f79d7ea', type 'readResponse', cluster 'genBasic', data '{"dateCode":"20180920"}' from endpoint 11 with groupID 0
zigbee2mqtt:warn  2020-05-10 20:43:52: Received message from unsupported device with Zigbee model 'GL-C-007'
zigbee2mqtt:warn  2020-05-10 20:43:52: Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html.
zigbee2mqtt:debug 2020-05-10 20:43:52: Received Zigbee message from '0x00124b001f79d7ea', type 'readResponse', cluster 'genBasic', data '{"swBuildId":"2.0.2"}' from endpoint 11 with groupID 0
zigbee2mqtt:warn  2020-05-10 20:43:52: Received message from unsupported device with Zigbee model 'GL-C-007'
zigbee2mqtt:warn  2020-05-10 20:43:52: Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html.
zigbee2mqtt:info  2020-05-10 20:43:53: Successfully interviewed '0x00124b001f79d7ea', device has successfully been paired
zigbee2mqtt:warn  2020-05-10 20:43:53: Device '0x00124b001f79d7ea' with Zigbee model 'GL-C-007' is NOT supported, please follow https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html
zigbee2mqtt:info  2020-05-10 20:43:53: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"pairing","message":"interview_successful","meta":{"friendly_name":"0x00124b001f79d7ea","supported":false}}'
```

## What did you expect to happen

Device is paired as the new `GL-C-007-2ID` model.

## How to reproduce it (minimal and precise)

Try to pair a GL-C-008 2ID device using zigbee2mqtt rev `fed84cfd5e9260564e58d03047b152b6fcdc3f80` or newer.

Fabiancrg, 2020-05-11:
@Koenkk I have the same problem, upgraded to DEV branch and try to pair my Gledopto GL-C-008 (2ID), it's detected as a GL-C-007 but device is reported as not supported (same message as output above).

Here is the Zigbee-herdsman debug log and extract from database.db if this can help:

[https://pastebin.com/xZmK3Gx3](https://pastebin.com/xZmK3Gx3)

`{"id":2,"type":"Router","ieeeAddr":"0x00124b001ee6ddbb","nwkAddr":41772,"manufId":0,"manufName":"GLEDOPTO","powerSource":"Mains` (single `phase)","modelId":"GL-C-007","epList":[11,13,15],"endpoints":{"11":{"profId":49246,"epId":11,"devId":528,"inClusterList":[0,3,4,5,6,8,768],"outClusterList":[],"clusters":{"genBasic":{"attributes":{"modelId":"GL-C-007","manufacturerName":"GLEDOPTO","powerSource":1,"zclVersion":1,"appVersion":1,"stackVersion":2,"hwVersion":1,"dateCode":"20180920","swBuildId":"2.0.2"}}},"binds":[]},"13":{"profId":49246,"epId":13,"devId":57694,"inClusterList":[4096],"outClusterList":[4096],"clusters":{},"binds":[]},"15":{"profId":49246,"epId":15,"devId":544,"inClusterList":[0,3,4,5,6,8,768],"outClusterList":[],"clusters":{},"binds":[]}},"appVersion":1,"stackVersion":2,"hwVersion":1,"dateCode":"20180920","swBuildId":"2.0.2","zclVersion":1,"interviewCompleted":true,"meta":{},"lastSeen":1589197211642}`

I can see a difference in epId 15 where my devId is 544 instead of 256, maybe it's a difference between GL-C-007 and GL-C-008.

So I added this and for me it seems to be OK for me:

    {
        fingerprint: [
            {type: 'Router', manufacturerName: 'GLEDOPTO', modelID: 'GL-C-007', endpoints: [
                {ID: 11, profileID: 49246, deviceID: 528, inputClusters: [0, 3, 4, 5, 6, 8, 768], outputClusters: []},
                {ID: 13, profileID: 49246, deviceID: 57694, inputClusters: [4096], outputClusters: [4096]},
                {ID: 15, profileID: 49246, deviceID: 544, inputClusters: [0, 3, 4, 5, 6, 8, 768], outputClusters: []},
            ]},
        ],
        model: 'GL-C-008-2ID', // 2 ID controls white and color separate
        vendor: 'Gledopto',
        description: 'Zigbee LED controller RGBW (2 ID)',
        extend: gledopto.light,
        supports: 'on/off, brightness, color temperature, color',
        endpoint: (device) => {
            return {rgb: 11, white: 15};
        },
    },

jussih, 2020-05-11:
Changing the endpoint 15 deviceId to 544 in the fingerprint works for me as well.

Koenkk, 2020-05-11:
Does

```js
{
    fingerprint: [
        {type: 'Router', manufacturerName: 'GLEDOPTO', modelID: 'GL-C-007', endpoints: [
            {ID: 11, profileID: 49246, inputClusters: [0, 3, 4, 5, 6, 8, 768], outputClusters: []},
            {ID: 13, profileID: 49246, inputClusters: [4096], outputClusters: [4096]},
            {ID: 15, profileID: 49246, inputClusters: [0, 3, 4, 5, 6, 8, 768], outputClusters: []},
        ]},
    ],
    model: 'GL-C-008-2ID', // 2 ID controls white and color separate
    vendor: 'Gledopto',
    description: 'Zigbee LED controller RGBW (2 ID)',
    extend: gledopto.light,
    supports: 'on/off, brightness, color temperature, color',
    endpoint: (device) => {
        return {rgb: 11, white: 15};
    },
},
```

also work?

TheDodger, 2020-05-11:
> Does
> ...
> also work?

Confirmed.

jussih, 2020-05-11:
> Does
> ...
> also work?

It does. Requires a change to the Home Assistant integration however, since the model changed from `GL-C-007-2ID` to `'GL-C-008-2ID`.

Fabiancrg, 2020-05-11:
@Koenkk it works also without the deviceID.

Is the goal to have one device for both GL-C-007 and GL-C-008 ?
If yes, maybe we should change the model with GL-C-007-2ID/GL-C-008-2ID ?

Koenkk, 2020-05-11:
Thinking about it again, I think `deviceID` shouldn't be removed as it allows us to identify the GL-C-007 and GL-C-008 separately, which also allows separate Home Assistant discovery. Changes will be available in the latest dev branch in a few hours from now.

jussih, 2020-05-11:
@Fabiancrg I didn't realize there was a 2ID version of the GL-C-007 too. Gledopto made everything extra hard by having the GL-C-008 2ID announce itself as GL-C-007...

GL-C-007-2ID and GL-C-008-2ID should be indeed treated separately as the 008 needs color temperature controls for the white channel and 007 does not.

Koenkk, 2020-05-11:
Changes are available in latest dev now. Please let me know if things work correctly there.

Fabiancrg, 2020-05-11:
Working perfectly for me for the Z2M part*, thank you @Koenkk 

*As I am noit using HA I let @jussih confirm this.

jussih, 2020-05-11:
Seems to me it's working perfectly in Home Assistant as well. Thank you @Koenkk!

Koenkk, 2020-05-11:
Cool, then I close this, big thanks for testing and helping out!

closed 2020-05-11 by Koenkk