Koenkk/zigbee2mqtt #3528: Sylvania 72922 Interview Failed

hamsando, 2020-05-11:
# Bug Report
## What happened
Trying to pair a Sylvania 72922 (not the A variant), the device does appear in the logs, but the interview fails. I did try to pair 5+ times. (I did add a Friendly name).

## What did you expect to happen
The interview to succeed. 

## How to reproduce it (minimal and precise)
Enable permit-join, plug in device and press/hold power button for ~10 seconds to start the pairing process.
 
## Debug Info
Zigbee2mqtt version: 1.13.0
Adapter hardware: CC26X2R1
Adapter firmware version: Used uniflash 5.3.1 if that helps 

## Logs
* Log of debug herdsman output.
* MQTT publish: topic 'zigbee2mqtt/bridge/log lines removed to keep the log within the size allowance of pastebin.
https://pastebin.com/knwu2db9

Koenkk, 2020-05-11:
- Make sure to check https://www.zigbee2mqtt.io/information/FAQ.html#why-does-my-device-not-or-fail-to-pair
- Also try to pair it through touchlink: https://www.zigbee2mqtt.io/information/touchlink.html

hamsando, 2020-05-12:
I had indeed looked through those details in the FAQ - and (like a fool) had dismissed the obvious one (move the device closer to the co-ordinator) as I've successfully paired my other devices (thermostats, sensors and buttons)  "in-place" around the house - and so was over-confident. 

Moving the device next to the co-ordinator resolved the issue immediately. Thanks for reminding me to basically RTFM.

closed 2020-05-12 by hamsando